ros-navigation/navigation2 #4734: ClearEntireCostmap returns before all costmap layer include valid data

schupf2, 2024-10-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04.5 LTS
- ROS2 Version:
  - Humble binaries
- Version or commit hash:
  - 11.1.13-0jammy
- DDS implementation:
  - Cyclonedds

#### Steps to reproduce issue
When working with a costmap with more then one different layers (in our case, we use `["static_layer", "obstacle_layer", "inflation_layer"]`), we face a problem after clearing the costmap: The the `nav2_msgs::srv::ClearEntireCostmap` service returns after the costmap is cleared. However, at this point no scanner measurements are considered in the obstacle layer. Thus, the space which is not considered in the static layer is marked as free for a short time until the next scanner measurement is included in the obstacle layer.

#### Expected behavior
There should be an option to either enforce waiting on information in all layers being present  before returning from the ClearEntireCostmap service call, or some option to verify the presence of information on all layers.

#### Actual behavior
After the ClearEntireCostmap service call the costmap only shows values form that static layer and no scanner measurements are yet considered in the obstacle layer. Therefore, obstacles in front of the robot are ignored for a brief time.

SteveMacenski, 2024-10-29:
That is why there is a ‘isCurrent’ API that is checked before planning or control is allowed to make sure there is new data before the cost map is allowed to be used. Why the cost map might be in a state without valid data after a clear, it is not used until the data is again valid for all layers. Does that make sense?

schupf2, 2024-10-30:
Thank you for the quick answer - that definitely makes sense :) .
However, if one uses the the collision checker directly (and not the whole controller) - the check is not performed and one needs to get a hold of the LayeredCostmap or the Costmap2DROS in order to be able to perform the check. So maybe a service (similar to e.g., ClearEntireCostmap) that waits for isCurrent to evaluate to true, might be a nice feature.

SteveMacenski, 2024-10-31:
A collision checker though would be used inside of a controller/planner plugin, so if the controller / planner is not being utilized until data is valid yet, then there is no issue :-) There shouldn't be any calls to costmap data structures by the task server's algorithms until `isCurrent` is true

SteveMacenski, 2024-11-05:
Are you saying to use the costmap topic collision detector object? That is a good point - I think though we could adjust how the costmap publishing is done so that we don't publish a new costmap until its current again. I don't think blocking the clear costmap service is a good idea since the costmap update rate is user-defined so it could be very slow. We could not update the published costmap topics as well as blocking the algorithm's use of the costmap until `isCurrent() == true`. 

Is that what you mean / work for you?

SteveMacenski, 2024-11-06:
@schupf2 I looked over the details and it should be as easy as adding `isCurrent()` to the condition here: https://github.com/ros-navigation/navigation2/blob/main/nav2_costmap_2d/src/costmap_2d_ros.cpp#L542-L544

Would you be open to testing that out to make sure that resolves your concern and then submitting the PR? :-)

schupf2, 2024-11-08:
Amazing thank you. 
Definitely, I'll do that on Monday :)

schupf2, 2024-11-11:
I implemented the discussed change today and tested it. Unfortunately, it still doesn't solve the problem.

I discovered that the `isCurrent()` of the obstacleLayer briefly returns true after resetting the costmap. This happens due to the following two parts in the code:

- In the `ObstacleLayer::resetBuffersLastUpdated()`, the `last_updated_` of the individual observation buffers is reset to the current time. Thus, right after resetting the individual buffers will return true on `isCurrent()`, even if no data has been received.

- Furthermore, in the `ObstacleLayer::updateCosts` (precisely here:  [https://github.com/ros-navigation/navigation2/blob/fbd1d3e7964dc220c0861d5e8bdcfcc7b9ec2812/nav2_costmap_2d/plugins/obstacle_layer.cpp#L542-544](url))  `current_` is also set to true in the first update after the reset. However, this does not at all verify that data is present / that the obstacle buffers are current.

SteveMacenski, 2024-11-11:
Ah - I didn't catch that in my initial analysis. 

I believe we set that to `true` after clearing since at that point, we've called `updateMap` for the `updateBounds` and `updateCosts` to have been triggered. Whether the buffers contain new data or not, from the main costmap's perspective, the layers have performed a complete update (just without promise that new sensor data has come in in the meantime). When using the sensor timeout parameter on the observation source, if an update hasn't come in in a sufficiently long time, then we set `current = false` since there's something wrong with a sensor. If the Obstacle Layer has multiple observation sources, do we block until all have some data? 

So from that perspective, this feels correct - we did successfully update the costmap layer, it just didn't contain a promise that data was processed yet. However the nature of "current" seems insufficient for the case where we (1) want to know if we performed a valid update and (2) the costmap layer actually had data that iteration as well. If you had a 5 hz publishing sensor and a 10 hz updating costmap, not each costmap update would have a new piece of sensor data and that is an expected behavior. 

That doesn't address **our** problem here, though. I'd be open to suggestions. A couple of initial thoughts:
- We have more than just `isCurrent` to know if all layers were actually updated with new data -- though a little problematic to decypher the outputs since some layers like the StaticLayer technically have updates, but are not frequently updated that you wouldn't want to block for -- but ObstacleLayer may be. Adding an `isUpdated()` API and how we process the output of that seems tricky to get right 
- Have some delay when a clearing costmap operation happens before the costmap collision checker object uses the costmap -- though that seems very fragile / hacky and would result in less responsive behaviors that use the costmap collision checker object 
- Clearing costmap service block until data comes in, but I think that mimicks the same issues in my first thought since its hard to tell which layers "should" have updates & blocking the service call could create other data congestion issues

It seems like we'd need to make some promise about the fact that each observation source in each data processing costmap layer has not only been called to update, but that update contained new current data. And, do so in a general purpose way that accounts for the fact that some data sources come in faster / slower, regular or irregular. 

Any thoughts?

schupf2, 2024-11-12:
Thanks for your detailed answer :).

Yes, I agree that it does make sense for a costmap to be valid in general after it has undergone a complete update procedure. However, I think this makes the `isCurrent` somewhat inconsistent. After clearing the costmap it basically tells us whether the costmap has been fully updated (not taking into account if sensor data has been received as expected), while during all other times `isCurrent` evaluates to false when sensor data is not received as expected (even if the map has been updated correctly).
 
So from my perspective, also taking into account the presence of sensor data (also after a reset), seems like a valid solution. If I'm not missing something here, it might also be less likely to introduce further problems then the points you mentioned (especially since you already mentioned the problems ^^).

The changes I would suggest:

1) I don't see the necessity to reset the `last_updated_` in the observation buffer after a reset, since the buffer itself is not reset (as far as I can see). So basically, when we reset the map the buffer itself is not affected. Therefore, I would suggest to NOT call `ObservationBuffer::resetLastUpdated()` if the map is cleared. That way, the `isCurrent()` of the observation buffer would always correctly take into consideration if sensor data has been received within the `expected_update_rate`. If the `observation_keep_time` > `expected_update_rate` it would then be safe to assume that there is sensor data in the buffer if `isCurrent()` evaluates to true.

2) You said that "I believe we set that to true after clearing since at that point, we've called `updateMap` for the `updateBounds` and `updateCosts` to have been triggered". However, the `current_` is already set in `updateBounds` anyway - if there is no irregularity with the sensor, it will also be set to true right after the costmap is cleared.  Thus, we would NOT need to set `current_` to true after a reset in `updateCosts`, since it will be set in `updateBounds` anyway.

However, I noticed another thing here: If the map was cleared, the obstacle layer reports being current already after the call to `updateBounds` (assuming sensor data is present) and before the call to `updateCosts` - i.e. before the map is actually updated. Maybe we should take this into consideration as well and check for `was_reset_` in the `isCurrent()` of the ObstacleLayer.

So to sum it up. I think if we remove line [760](https://github.com/ros-navigation/navigation2/blob/f426bdaadaa1a124bea0da5f32961cef6d679ae2/nav2_costmap_2d/plugins/obstacle_layer.cpp#L760) and [544](https://github.com/ros-navigation/navigation2/blob/f426bdaadaa1a124bea0da5f32961cef6d679ae2/nav2_costmap_2d/plugins/obstacle_layer.cpp#L544) in the `ObstacleLayer`, we would obtain consistent behavior no matter if the costmap was reset. For the normal case (when sensor data is received as expected), after a reset the costmap is current again as soon as the map was updated. While if we didn't receive sensor data in the expected update rate, the costmap will not be current regardless of whether it has been cleared or not.

Combined with the change of only publishing the costmap if it's current, this would also solve my problem (at least from what I can see in a preliminary analysis) :).

schupf2, 2024-11-14:
> Doable, but starts to feel like patches on top of patches ...

You're definitely right there. I also think we should keep it as simple as is and only consider the expected_update_rate instead of the actual presence of data.

---
> I believe this could be due to a reset operation taking arbitrarily long, or reset is meant to reset all state and that's an important element of the layer's state ...

Yes, if we used to unsubscribe and resubscribe to the sensor topic the `resetBuffersLastUpdated` definitely seems to have been necessary, but as this is not done anymore I think the `resetBuffersLastUpdated` should not be needed.

In terms of "a reset operation taking arbitrarily long" - this should not matter right? For the `ObservationBuffer` we store new point clouds in the buffer when received. So if we don't unsubscribe while clearing the map, we will also obtain updates during that time - so the actual update rate should not be affected.

> So, I'm a bit cautious about a change to that.

Definitely understand that though :-).

---
>  So, if we updateMap and we have valid (although potentially empty) observation buffers, then it would always be current_ = true

As far as I can see it, we should not obtain a observation buffer that is empty and valid IF we set the `observation_persistence` and `expected_update_rate` accordingly (since the observation buffer is not cleared during a reset). That is totally fine IMO, since it's basically up to the user if they are willing to accept potentially empty observation buffers or not.

> I'd love a 3rd sanity check on that, but I'm on board.

:D

---
> current_ = false; is set in the reset along with was_reset_ = true;. So, isCurrent() is already going to return false until updateMap runs to trigger updateBounds / updateCosts. 

Yes exactly, but in the `LayeredCostmap::updateMap` the `updateBounds` is of course always called before the `updateCosts`. Thus, (in theory) after a reset `isCurrent` could be set to true before the the map is actually updated in `updateCosts` (at least for the obstacle layer), leading to a costmap with `isCurrent` being true before the obstacle layer is included in the map. 

However, I didn't notice any problems with that so far, so I guess this should not cause problems since the costmap will be updated right after.

--- 
Looking forward to @mikeferguson thoughts :).

SteveMacenski, 2024-12-02:
Sure thing, STVL / NPVL would probably be good to both be updated by this (though NPVL is so simple, it probably isn't required)

mikeferguson, 2024-12-10:
I think, for costmap historical stuff, @DLu might have more knowledge than me (I read this entire ticket but actually don't have anything of value to add)

DLu, 2024-12-13:
Greetings - 

For historical perspective, the [`isCurrent`](https://github.com/ros-planning/navigation/blob/9ad644198e132d0e950579a3bc72c29da46e60b0/costmap_2d/include/costmap_2d/layer.h#L83-L96) structure was put in place specifically so that obstacle layers that haven't gotten updates recently would cause the planning to stop. [This bit of `ObservationBuffer::isCurrent`](https://github.com/ros-planning/navigation/blob/9ad644198e132d0e950579a3bc72c29da46e60b0/costmap_2d/src/observation_buffer.cpp#L231-L244) is the *only* bit that could cause `Costmap2DROS::isCurrent` to return false (as far as I recall) in ROS 1. 

As far as I've found, the current status was [not checked before running recovery behaviors such as clearing the costmap](https://github.com/ros-planning/navigation/blob/9ad644198e132d0e950579a3bc72c29da46e60b0/move_base/src/move_base.cpp#L965). I also don't believe that [the recovery itself checked or set the current status.](https://github.com/ros-planning/navigation/blob/9ad644198e132d0e950579a3bc72c29da46e60b0/clear_costmap_recovery/src/clear_costmap_recovery.cpp#L131-L165)

> I think if we remove line [760](https://github.com/ros-navigation/navigation2/blob/f426bdaadaa1a124bea0da5f32961cef6d679ae2/nav2_costmap_2d/plugins/obstacle_layer.cpp#L760) and [...]

Line 760 was introduced in 2019 by @crdelsey in #1412. Line 544 was @SteveMacenski in 2020 with #2090. 

Ultimately, I think that there are three distinct states intertwined with this bool, which I'm going to assign arbitrary names. 
 1. `FRESH` - The costmap is newly initialized OR has been reset recently and has yet to receive new data. 
 2. `CURRENT` - The costmap has gotten some data recently within the expected update period. 
 3. `STALE` - The costmap has not gotten any data within the expected update period. 

`isCurrent` returning `false` was (in ROS 1) only an indication of the `STALE` observation data. The expansion to cover both `STALE` and `FRESH` came from #2090, by setting `current_` to false initially and when resets happen. 

I haven't yet traced through all the changes to `current_` in Nav2, but removing line 544 makes sense to me, although then we might not even need `was_reset`. I'm not sure about removing line 760, which makes sure that the costmap is not `STALE`, which might make it so that a freshly reset costmap that hasn't recieved data recently might still be marked as `CURRENT`

schupf2, 2024-12-13:
Thank you @DLu for your detailed response and providing historical insight  :)

Why do you think removing line 760 prevents the costmap from being `STALE` ? 
I would see it rather the other way around. The `resetBuffersLastUpdated()` which is called in line 760 is only setting the `last_updated_` of each observation buffer to the current time (not clearing any data from the buffer). Thus, if we don't receive any data, the costmap will still be current after a reset, since the `last_updated_` is recent enough (even if we didn't receive any data).

schupf2, 2024-12-17:
Since I already have the implementation for all my suggestions, I created a pull request (#4800) so some changes might be more visible / easier to discuss.

Looking forward to your comments on this.
In the meantime: Wish you all a merry christmas and happy new year :)

SteveMacenski, 2024-12-19:
If you could take a look at that David, I'd appreciate it

mini-1235, 2026-02-23:
I think there have been some updates related to the "current" variable in the costmap. Does that address the issue mentioned here?

SteveMacenski, 2026-02-23:
i.e. https://github.com/ros-navigation/navigation2/commit/1b9879674f167d0ee52964136e95859366430d6e

SteveMacenski, 2026-04-08:
This issue was actually addressed!