ros-navigation/navigation2 #4739: Voxel layer not working for local costmap

Cavalletta98, 2024-11-04:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Humble binaries
- Version or commit hash:
  - ros-humble-navigation2	1.1.16-1jammy.20240830.231501
    ros-humble-nav2-amcl	1.1.16-1jammy.20240830.193818
    ros-humble-nav2-behavior-tree	1.1.16-1jammy.20240830.193910
    ros-humble-nav2-behaviors	1.1.16-1jammy.20240830.205037
    ros-humble-nav2-bringup	1.1.16-1jammy.20240831.001909
    ros-humble-nav2-bt-navigator	1.1.16-1jammy.20240830.205457
    ros-humble-nav2-collision-monitor	1.1.16-1jammy.20240830.204654
    ros-humble-nav2-common	1.1.16-1jammy.20240824.181243
    ros-humble-nav2-constrained-smoother	1.1.16-1jammy.20240830.205908
    ros-humble-nav2-controller	1.1.16-1jammy.20240830.205928
    ros-humble-nav2-core	1.1.16-1jammy.20240830.204819
    ros-humble-nav2-costmap-2d	1.1.16-1jammy.20240830.202726
    ros-humble-nav2-dwb-controller	1.1.16-1jammy.20240830.214325
    ros-humble-nav2-lifecycle-manager	1.1.16-1jammy.20240830.202342
    ros-humble-nav2-map-server	1.1.16-1jammy.20240830.193909
    ros-humble-nav2-mppi-controller	1.1.16-1jammy.20240830.205055
    ros-humble-nav2-msgs	1.1.16-1jammy.20240824.181735
    ros-humble-nav2-navfn-planner	1.1.16-1jammy.20240830.205449
    ros-humble-nav2-planner	1.1.16-1jammy.20240830.205543
    ros-humble-nav2-regulated-pure-pursuit-controller	1.1.16-1jammy.20240830.205157
    ros-humble-nav2-rotation-shim-controller	1.1.16-1jammy.20240830.210830
    ros-humble-nav2-rviz-plugins	1.1.16-1jammy.20240830.230452
    ros-humble-nav2-simple-commander	1.1.16-1jammy.20240824.182428
    ros-humble-nav2-smac-planner	1.1.16-1jammy.20240830.205547
    ros-humble-nav2-smoother	1.1.16-1jammy.20240830.214913
    ros-humble-nav2-theta-star-planner	1.1.16-1jammy.20240830.205759
    ros-humble-nav2-util	1.1.16-1jammy.20240830.192949
    ros-humble-nav2-velocity-smoother	1.1.16-1jammy.20240830.204314
    ros-humble-nav2-voxel-grid	1.1.16-1jammy.20240824.181738
    ros-humble-nav2-waypoint-follower	1.1.16-1jammy.20240830.205813

- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

1. In one terminal on the robot: ros2 launch clearpath_nav2_demos nav2.launch.py use_sim_time:=false
2. In another terminal on the robot: ros2 launch clearpath_nav2_demos slam.launch.py use_sim_time:=false
3. On the host machine: ros2 launch clearpath_viz view_navigation.launch.py namespace:=j100_0538

#### Expected behavior
Build a local costmap using point clouds generated by an Ouster OS LiDAR and a RealSense D435i camera.

#### Actual behavior
Costmap is empty when only the Voxel Layer is used. However, when the Static Layer is used, it functions correctly because the Static Layer relies on SLAM data, which utilizes information from the laser scan.

#### Additional information
Attached you can find the yaml files for nav2 and slam
[params.zip](https://github.com/user-attachments/files/17617043/params.zip)

SteveMacenski, 2024-11-04:
Unfortunately, we do not currently offer support for third party robot vendors and their configurations / setups for Nav2. Please reach out to your robot vendor that setup your configurations for assistance. I can say though with near absolute certainty that the voxel layer in Humble is not broken for local costmaps. I would have received dozens of emails about that if so at the time of breakage ;-)

closed 2024-11-04 by SteveMacenski

Cavalletta98, 2024-11-04:
Thank you so much

Cavalletta98, 2024-11-15:
Hi @SteveMacenski i investigating deep and the problem is that my local costamp is working but is empty when using only  the pointcloud from relasense camera. I shared with you the configuration used for the nav2. Do you have some ideas? the same nav2 configuration is working with the simulated realsense
[nav2.zip](https://github.com/user-attachments/files/17774544/nav2.zip)