ros-navigation/navigation2 #4746: Would the obstacle layer benefit from this change?

suchetanrs, 2024-11-08:
Hey!
I was having a look at the obstacle layer and I observed a few things...

I have a 2D laser at a high frequency. In the obstacle layer's implementation, it converts every new laser reading into a pointcloud and buffers it. If a user has a high frequency laser scanner for any other reason (for better motion estimation etc) but only needs to update the local costmap at a lower frequency, isn't this conversion to a pointcloud at a high rate redundant? Can this have a potential fix?
Tagging the code where the conversion is implemented below 
https://github.com/ros-navigation/navigation2/blob/fbd1d3e7964dc220c0861d5e8bdcfcc7b9ec2812/nav2_costmap_2d/plugins/obstacle_layer.cpp#L342C1-L364C2

If I directly send a pointcloud to the obstacle layer, it has parameters to control the min range, max range and max and min heights in the pointcloud. However, there is no option to control the 2D angular range. Would it be useful to add this feature for more flexibility for the user?

SteveMacenski, 2024-11-08:
Can you define 'high rate' in your context? I've somewhat recently had a project with a 40 hz industrial 2D lidar at 0.25 deg resolution -- which is on the higher-ish end of standard for industrial rated lidars. Back of the envelope says that converts 57,600 points per second, which is approximately the same as a single 240x240 frame from a depth camera (which is usually processed at a higher resolution, at 6-20hz). So in general, I don't look at this as needing any kind of optimization considering its already pretty light weight. I've worked with some student teams in the past that are able to easily accomplish this on a Raspberry Pi (years ago, so probably a 3?) - which I view as sufficiently performant that any further improvement is really not massively value added.

With that said, I wouldn't reject contributions that improved things :-) I'm just not sure what high rate / resolution you're working with that has made this problematic. 

>  If a user has a high frequency laser scanner for any other reason (for better motion estimation etc) but only needs to update the local costmap at a lower frequency, isn't this conversion to a pointcloud at a high rate redundant?

You would lose some information, but if you're fine with that, it is certainly possible today. Rather than updating the costmap code however, I might recommend putting in a [Topic Throttler Node](https://github.com/ros-tooling/topic_tools) between your `scan` and your `costmap_scan` so that you reduce the effective rate of consumption by the costmap by reducing the frequency on the topic itself to a rate you find desirable.   

> If I directly send a pointcloud to the obstacle layer, it has parameters to control the min range, max range and max and min heights in the pointcloud. However, there is no option to control the 2D angular range. Would it be useful to add this feature for more flexibility for the user?

If you send a 3D pointcloud, there's no concept of a 2D angular range :-) I'm not entirely sure what that would be intended for in this case.

suchetanrs, 2024-11-09:
Hey @SteveMacenski 
Thanks a lot for your quick reply!

To be honest, this does not cause any problem at all! It works just fine :)
To clarify further, I am running a lidar that is similar to the one you mentioned (~ 60,000 points per second). 
I simulated the same lidar properties you stated and tried measuring the cpu load using [psutil](https://pypi.org/project/psutil/) with the interval set to 1 second. The pointcloud conversion is running on a Jetson Xavier which is clocked at ~1.3 Ghz. I receive values that vary between 8% - 11% when I tried  to measure the load. It is perfectly arguable that this load is insignificant but for a cpu that is almost fully choked, I thought this reduction could provide some breathing room :p 
 
> With that said, I wouldn't reject contributions that improved things :-) I'm just not sure what high rate / resolution you're working with that has made this problematic.

If this is a potential improvement, I will try to work on it!

---
For the second problem:

Sorry I did not provide more clarification earlier. On a lower end robot, I have a flat pointcloud that is fed into the obstacle layer. During marking onto the costmap, when I was doing some tests, I felt the need to restrict the angular range similar to how the radial range was restricted [here](https://github.com/ros-navigation/navigation2/blob/fbd1d3e7964dc220c0861d5e8bdcfcc7b9ec2812/nav2_costmap_2d/plugins/obstacle_layer.cpp#L459C1-L473C8). I did this by pruning and re-publishing the pointcloud with the points removed for now but I wondered if adding such a parameter directly to the layer would be useful.

SteveMacenski, 2024-11-11:
I think its probably best to have that as a pre-processing step for the topic that goes into the costmap layer. Its pretty routine for folks to massage the data (whether noise removal, pruning, downsampling update rate, etc) before going into the costmap. I would arguably say its recommended to do some data quality control. 

Its more straight forward to do your angular checks when you have the laserscan object since you have the direct angle of the ray available. We could compute it where you linked, but that would involve an `atan2` for each ray, which is a relatively heavy operation which could further increase compute times. I think actually implementing it inside of the costmap layer vs in a pre-processing step where you have the `LaserScan` message would be computationally more expensive overall :-)

suchetanrs, 2024-11-13:
Got it. That clears a lot. Thanks, Steve!

I am closing this issue. Will open a new one if I face further issues.

closed 2024-11-13 by suchetanrs