ros-navigation/navigation2 #4751: MPC tuning for Ackermann drive

Ram-userfriendly, 2024-11-12:
hey community, I'm trying to develop a MPC controller in NAV2 on ROS2 for Ackermann drive vehicle with radius of 4 but all the parameters available in [https://docs.nav2.org/](url) is working for very low minimum turning radius and differential,omni drive models.So help me out community with tuning the MPC . I'll attach below the parameters,system specs im using 
[https://docs.google.com/document/d/1I-2P7pUF-wTWrsQYHuJUYNPHrF5ksY_4zRxmwaPy_eU/edit?usp=sharing](url)

SteveMacenski, 2024-11-12:
Please ask on robotics stack exchange for configuration help

closed 2024-11-12 by SteveMacenski

Ram-userfriendly, 2024-11-13:
I posted the on Robotics Stack Exchange too, half a month back  but no considerable response.
https://robotics.stackexchange.com/questions/113495/mpc-controller-tuning