ros-navigation/navigation2 #4754: Unable to set goal pose if we run rviz2 later than navitation2

evshary, 2024-11-18:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 24.04
- ROS2 Version:
  - Jazzy / Rolling
- Version or commit hash:
  - f426bdaadaa1a124bea0da5f32961cef6d679ae2 (Rolling latest commit)
  - 3140a6a65c2ecbcf975dac94e3f3c64c0d9efc84 (Jazzy latest commit)
- DDS implementation:
  - CycloneDDS

#### Steps to reproduce issue

* Run the simulation first

```bash
ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False use_rviz:=False
```

* Run the rviz2 in another terminal

```bash
ros2 launch nav2_bringup rviz_launch.py
```

#### Expected behavior

The result should do nothing with the execution order.

#### Actual behavior

The rviz is stuck for a while.

[image]

After waiting some time, we can set the initial pose, but the global costmap doesn't work properly.

[image]

#### Additional information

There is no any issue if we run rviz and nav2 together.
It also works well if we run rviz first and then nav2.

Here is the log in nav2:

```
[component_container_isolated-7] [WARN] [1731909745.646348226] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[component_container_isolated-7] [INFO] [1731909745.785165110] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-7] [INFO] [1731909746.285144231] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-7] [INFO] [1731909746.785142621] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-7] [INFO] [1731909747.285143093] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-7] [INFO] [1731909747.785165097] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-7] [WARN] [1731909747.853379848] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[component_container_isolated-7] [INFO] [1731909748.285171155] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-7] [INFO] [1731909748.785144203] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-7] [INFO] [1731909749.285145701] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-7] [INFO] [1731909749.785140808] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-7] [WARN] [1731909749.878329922] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[component_container_isolated-7] [INFO] [1731909750.285153676] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-7] [INFO] [1731909750.785139863] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-7] [INFO] [1731909751.285171461] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-7] [INFO] [1731909751.785236796] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-7] [WARN] [1731909752.085459342] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[component_container_isolated-7] [INFO] [1731909752.285106163] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-7] [INFO] [1731909752.785160362] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-7] [ERROR] [1731909752.785244117] [global_costmap.global_costmap]: Failed to activate global_costmap because transform from base_link to map did not become available before timeout
[component_container_isolated-7] [ERROR] [1731909752.785606599] [lifecycle_manager_navigation]: Failed to change state for node: planner_server
[component_container_isolated-7] [ERROR] [1731909752.785636720] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
[component_container_isolated-7] [WARN] [1731909754.293492285] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[component_container_isolated-7] [WARN] [1731909756.501404703] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[component_container_isolated-7] [WARN] [1731909758.708340139] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[component_container_isolated-7] [WARN] [1731909760.735512443] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[component_container_isolated-7] [WARN] [1731909762.943117665] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[component_container_isolated-7] [WARN] [1731909765.150435302] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[component_container_isolated-7] [WARN] [1731909767.357452515] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[component_container_isolated-7] [WARN] [1731909769.565269254] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[component_container_isolated-7] [WARN] [1731909771.591201935] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[component_container_isolated-7] [WARN] [1731909773.798548704] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[component_container_isolated-7] [INFO] [1731909774.666120162] [amcl]: initialPoseReceived
[component_container_isolated-7] [WARN] [1731909774.666267098] [amcl]: Failed to transform initial pose in time (Lookup would require extrapolation into the future.  Requested time 82.692000 but the latest data is at time 82.656000, when looking up transform from frame [base_footprint] to frame [odom])
[component_container_isolated-7] [INFO] [1731909774.666293410] [amcl]: Setting pose (82.692000): -2.011 -0.542 0.020
```

I guess this log might be the key

```
[component_container_isolated-7] [ERROR] [1731909752.785244117] [global_costmap.global_costmap]: Failed to activate global_costmap because transform from base_link to map did not become available before timeout
[component_container_isolated-7] [ERROR] [1731909752.785606599] [lifecycle_manager_navigation]: Failed to change state for node: planner_server
[component_container_isolated-7] [ERROR] [1731909752.785636720] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
```

SteveMacenski, 2024-11-18:
```
[component_container_isolated-7] [ERROR] [1731909752.785244117] [global_costmap.global_costmap]: Failed to activate global_costmap because transform from base_link to map did not become available before timeout
[component_container_isolated-7] [ERROR] [1731909752.785606599] [lifecycle_manager_navigation]: Failed to change state for node: planner_server
[component_container_isolated-7] [ERROR] [1731909752.785636720] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
```

Rviz has no relationship with Nav2 in terms of functional performance. However, as the [Getting Started Guide](https://docs.nav2.org/getting_started/index.html) mentions, you need to give the robot its initial pose on startup so that you can complete the map->odom->base_link transformation tree. You do that in tutorial / testing using rviz, though in production its based on the known robot's last pose. If you didn't have rviz launch, then you probably didn't set this initial pose, and thus you see the global costmap timing out.

tl;dr: you need to set the initial pose. You can see this done in all Nav2 system tests, Nav2 python simple commander demos, and in other tutorial/educational demonstrations of Nav2 on GitHub :-)

evshary, 2024-11-19:
Hi @SteveMacenski,
Thanks for your answer. However, I'm not sure if I get your point. I can set the initial pose after launching the rivz, but global costmap is still timed out. I know rviz is just a way to display and does nothing to do with nav2, so I can't understand why launching rviz later doesn't work in this case.

SteveMacenski, 2024-11-19:
> After waiting some time, we can set the initial pose, but the global costmap doesn't work properly.

I guess, the following snippet tells a clear story to me, but I don't discount the opportunity that I'm an idiot on occasion :laughing: 

```
[component_container_isolated-7] [INFO] [1731909752.785160362] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-7] [ERROR] [1731909752.785244117] [global_costmap.global_costmap]: Failed to activate global_costmap because transform from base_link to map did not become available before timeout
[component_container_isolated-7] [ERROR] [1731909752.785606599] [lifecycle_manager_navigation]: Failed to change state for node: planner_server
[component_container_isolated-7] [ERROR] [1731909752.785636720] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.

...

[component_container_isolated-7] [WARN] [1731909754.293492285] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
```

This is telling me that the map frame isn't available -- ergo the initial pose wasn't set. Once we hit a set deadline, if we haven't got a valid TF tree, the global costmap node refuses to activate properly and returns a failure. **That is new** in Jazzy, previously it would retry forever, but it was pointed out that this was a bad idea. 

Then, AMCL gives us logs that it cannot do anything because initial pose wasn't set. 

All roads seem to lead back to the initial pose wasn't set :upside_down_face:

SteveMacenski, 2024-11-19:
What is the logging from rviz / nav2 when the rviz display starts to respond again? Clearly something timed out and I want to know what it is / what it is related to. It seems like perhaps the same global costmap lifecycle transition.

What happens if you remove each of the Nav2 panel plugins, one at a time, and launch between them? Anyone in in particular have the issue removed if not present (again, I expect to be related to the initialization - so I would think the main Nav2 panel if any).

evshary, 2024-11-21:
> What is the logging from rviz / nav2 when the rviz display starts to respond again? Clearly something timed out and I want to know what it is / what it is related to. It seems like perhaps the same global costmap lifecycle transition.

After rviz display starts to respond again:

rviz2
```
[rviz2-1] [INFO] [1732157523.046762030] [rviz]: Message Filter dropping message: frame 'base_scan' at time 5.802 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1732157523.110928382] [rviz]: Message Filter dropping message: frame 'odom' at time 5.940 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1732157523.239344329] [rviz]: Message Filter dropping message: frame 'base_scan' at time 60.201 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1732157523.302777705] [rviz]: Message Filter dropping message: frame 'odom' at time 60.264 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1732157523.431236205] [rviz]: Message Filter dropping message: frame 'base_scan' at time 60.402 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1732157523.495733757] [rviz]: Message Filter dropping message: frame 'odom' at time 60.444 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1732157523.655407298] [rviz]: Message Filter dropping message: frame 'base_scan' at time 60.600 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1732157523.687198787] [rviz]: Message Filter dropping message: frame 'odom' at time 60.660 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1732157523.847229009] [rviz]: Message Filter dropping message: frame 'base_scan' at time 60.801 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1732157523.911115823] [rviz]: Message Filter dropping message: frame 'odom' at time 60.840 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1732157524.039079667] [rviz]: Message Filter dropping message: frame 'base_scan' at time 61.002 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1732157524.102697059] [rviz]: Message Filter dropping message: frame 'odom' at time 61.056 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1732157524.263133692] [rviz]: Message Filter dropping message: frame 'base_scan' at time 61.200 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1732157524.295430103] [rviz]: Message Filter dropping message: frame 'odom' at time 61.236 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1732157524.455102883] [rviz]: Message Filter dropping message: frame 'base_scan' at time 61.401 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1732157524.487568538] [rviz]: Message Filter dropping message: frame 'odom' at time 61.452 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1732157524.646650880] [rviz]: Message Filter dropping message: frame 'base_scan' at time 61.602 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1732157524.711744093] [rviz]: Message Filter dropping message: frame 'odom' at time 61.632 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1732157524.839211244] [rviz]: Message Filter dropping message: frame 'base_scan' at time 61.800 for reason 'discarding message because the queue is full'
...
```

nav2
```
[component_container_isolated-7] [INFO] [1732157521.039148365] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-7] [WARN] [1732157521.241269141] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[component_container_isolated-7] [INFO] [1732157521.539235593] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-7] [INFO] [1732157522.039177701] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-7] [ERROR] [1732157522.039220781] [global_costmap.global_costmap]: Failed to activate global_costmap because transform from base_link to map did not become available before timeout
[component_container_isolated-7] [ERROR] [1732157522.039616016] [lifecycle_manager_navigation]: Failed to change state for node: planner_server
[component_container_isolated-7] [ERROR] [1732157522.039642736] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
[component_container_isolated-7] [WARN] [1732157523.448310789] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[component_container_isolated-7] [WARN] [1732157525.655386476] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[component_container_isolated-7] [WARN] [1732157527.863406841] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[... cut ...]
> What happens if you remove each of the Nav2 panel plugins, one at a time, and launch between them? Anyone in in particular have the issue removed if not present (again, I expect to be related to the initialization - so I would think the main Nav2 panel if any).

OK, I think the issue might come from this in rviz. It works perfectly if I remove Selector in rviz file.

```
  - Class: nav2_rviz_plugins/Selector
    Name: Selector
```

Here is the rviz file that works on my side.
[available_rviz.zip](https://github.com/user-attachments/files/17839237/available_rviz.zip)

evshary, 2024-11-25:
Hi @SteveMacenski I'm not sure if you think this is a bug or not. Let me know if I should reopen the issue. Thank you

SteveMacenski, 2024-11-25:
I agree this is a bug -- CC @padhupradheep as the author of this newer panel. I suspect it has to do with https://github.com/ros-navigation/navigation2/blob/main/nav2_rviz_plugins/src/selector.cpp#L154-L168

padhupradheep, 2024-11-25:
Thanks for reporting @evshary 

I'll take a look at it soon.

SteveMacenski, 2024-11-25:
I know you've got a few things on your queue @padhupradheep for Nav2 -- I'd put this on the bottom, the other 2 are higher priority. This is something we should fix for sure, but isn't functionally breaking Nav2 uses in deployed applications, so I would deprioritize this over those

padhupradheep, 2024-11-27:
Makes sense to me. @evshary you want to give a swing at it?

evshary, 2024-11-27:
Hi @padhupradheep Thanks for looking into it. I could give it a try, but I've been blocked with other issues currently. Therefore, don't wait for me if this is urgent.

closed 2025-01-29 by SteveMacenski