ros-navigation/navigation2 #4756: Feature Request:Make the isPathValid Service Configurable with Parameters

BCKSELFDRIVEWORLD, 2024-11-19:
## Feature request

#### Feature description
Currently, the isPathValid service considers a path valid as long as the costs are not lethal. However, in many scenarios, users might need more flexibility in determining what constitutes a valid path. For instance, there might be cases where paths with costs above a certain threshold (but below lethal cost) should also be considered invalid.

To address this, I propose introducing a parameter that allows users to specify the threshold cost for path validity. This would make the service more adaptable to different use cases and improve its overall utility.

#### Proposed Feature:

- Add a parameter to the isPathValid service that allows users to define the maximum allowable cost for a path to be considered valid.

SteveMacenski, 2024-11-19:
Seems reasonable to have a max cost parameter or in the service request itself. I think the service request having a field for the max, valid cost would be great & pretty easy contribution. @BCKSELFDRIVEWORLD can you submit a PR?

Youlun-Peng, 2025-01-24:
Hey, @SteveMacenski ! Is there any progress here? Can I contribute to address this issue? :D

BCKSELFDRIVEWORLD, 2025-01-24:
İ will sent pr this week

mini-1235, 2025-01-25:
> i will sent pr this week

Sorry, I just saw this reply after I made the PR 🥲

BCKSELFDRIVEWORLD, 2025-01-25:
If you haven’t seen it, no problem. Congratulations! I’m sorry for keeping you @SteveMacenski waiting this long; I was busy and forgot. :/

CesarBear, 2025-01-28:
@SteveMacenski I am open to contributing something else. Is there any issue suitable for my first trial help?

SteveMacenski, 2025-01-28:
@CesarBear please check out tickets with help wanted, enhancement, or good first issue!

* https://github.com/ros-navigation/navigation2/issues?q=is%3Aissue%20state%3Aopen%20label%3A%22help%20wanted%22
* https://github.com/ros-navigation/navigation2/issues?q=is%3Aissue%20state%3Aopen%20label%3A%22good%20first%20issue%22
* https://github.com/ros-navigation/navigation2/issues?q=is%3Aissue%20state%3Aopen%20label%3Aenhancement

In particular, one of these might be good!:
* https://github.com/ros-navigation/navigation2/issues/4403
* https://github.com/ros-navigation/navigation2/issues/4695
* https://github.com/ros-navigation/navigation2/issues/4794
* https://github.com/ros-navigation/navigation2/issues/4769
* https://github.com/ros-navigation/navigation2/issues/4757
* https://github.com/ros-navigation/navigation2/issues/4846

closed 2025-01-29 by SteveMacenski