ros-navigation/navigation2 #4759: Nav2 path planner issues

BrainBoxTayo, 2024-11-21:
**Required Info:**

- Operating System:
  - Ubuntu 24.04
- ROS2 Version:
  - Jazzy binaries

#### Expected behavior
The robot is meant to move to the desired location
#### Actual behavior
The robot does *nothing*; these are the log files
___

```
[rviz2-17] [INFO] [1732224052.795018475] [rviz2]: Setting goal pose: Frame:map, Position(1.38254, -2.50791, 0), Orientation(0, 0, -0.804285, 0.594243) = Angle: -1.86894
[bt_navigator-22] [INFO] [1732224052.802246292] [bt_navigator]: Begin navigating from current location (-0.00, -0.00) to (1.38, -2.51)
[controller_server-18] [INFO] [1732224052.831471224] [controller_server]: Received a goal, begin computing control effort.
[controller_server-18] [WARN] [1732224052.831619991] [controller_server]: No goal checker was specified in parameter 'current_goal_checker'. Server will use only plugin loaded general_goal_checker . This warning will appear once.
[controller_server-18] [WARN] [1732224052.831641991] [controller_server]: No progress checker was specified in parameter 'current_progress_checker'. Server will use only plugin loaded progress_checker . This warning will appear once.
[controller_server-18] [WARN] [1732224053.134075888] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 37.0370 Hz.
[controller_server-18] [WARN] [1732224053.192542308] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 33.3333 Hz.
[controller_server-18] [WARN] [1732224053.262441328] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 25.0000 Hz.
[controller_server-18] [WARN] [1732224053.329031935] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 30.3030 Hz.
[controller_server-18] [WARN] [1732224053.367981937] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 62.5000 Hz.
[controller_server-18] [WARN] [1732224053.390512937] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 100.0000 Hz.
[controller_server-18] [WARN] [1732224053.483967700] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 45.4545 Hz.
[controller_server-18] [WARN] [1732224053.591123368] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 31.2500 Hz.
[controller_server-18] [WARN] [1732224053.837647469] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 31.2500 Hz.
[controller_server-18] [WARN] [1732224053.882163992] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 52.6316 Hz.
[controller_server-18] [INFO] [1732224053.882380157] [controller_server]: Passing new path to controller.
[controller_server-18] [WARN] [1732224053.934579177] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 31.2500 Hz.
[controller_server-18] [WARN] [1732224053.993899712] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 28.5714 Hz.
[controller_server-18] [WARN] [1732224054.041685048] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 47.6190 Hz.
[controller_server-18] [WARN] [1732224054.090370876] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 35.7143 Hz.
[controller_server-18] [WARN] [1732224054.537343387] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 35.7143 Hz.
[controller_server-18] [INFO] [1732224054.883022464] [controller_server]: Passing new path to controller.
[controller_server-18] [WARN] [1732224055.541958201] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 30.3030 Hz.
[controller_server-18] [WARN] [1732224055.589491963] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 37.0370 Hz.
[controller_server-18] [WARN] [1732224055.932063858] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 34.4828 Hz.
[controller_server-18] [INFO] [1732224055.932154235] [controller_server]: Passing new path to controller.
[controller_server-18] [WARN] [1732224056.134384424] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 38.4615 Hz.
[controller_server-18] [INFO] [1732224056.932500337] [controller_server]: Passing new path to controller.
[controller_server-18] [WARN] [1732224057.439105356] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 33.3333 Hz.
[controller_server-18] [WARN] [1732224057.483830018] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 43.4783 Hz.
[controller_server-18] [WARN] [1732224057.741355192] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 35.7143 Hz.
[controller_server-18] [INFO] [1732224057.982144344] [controller_server]: Passing new path to controller.
[controller_server-18] [WARN] [1732224058.435693373] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 38.4615 Hz.
[controller_server-18] [INFO] [1732224059.032194389] [controller_server]: Passing new path to controller.
[controller_server-18] [WARN] [1732224059.082658299] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 40.0000 Hz.
[controller_server-18] [WARN] [1732224059.183879602] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 34.4828 Hz.
[controller_server-18] [WARN] [1732224059.287310471] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 40.0000 Hz.
[controller_server-18] [WARN] [1732224059.632438054] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 31.2500 Hz.
[controller_server-18] [WARN] [1732224059.684076801] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 34.4828 Hz.
[controller_server-18] [INFO] [1732224060.032362077] [controller_server]: Passing new path to controller.
[controller_server-18] [WARN] [1732224060.140030561] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 25.6410 Hz.
[controller_server-18] [WARN] [1732224060.184843440] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 40.0000 Hz.
[controller_server-18] [WARN] [1732224061.031863213] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 35.7143 Hz.
[controller_server-18] [INFO] [1732224061.082081348] [controller_server]: Passing new path to controller.
[controller_server-18] [WARN] [1732224061.335453038] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 31.2500 Hz.
[controller_server-18] [WARN] [1732224061.383205000] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 34.4828 Hz.
[controller_server-18] [WARN] [1732224061.450022267] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 26.3158 Hz.
[controller_server-18] [WARN] [1732224061.984948525] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 37.0370 Hz.
[controller_server-18] [WARN] [1732224062.083627536] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 45.4545 Hz.
[controller_server-18] [INFO] [1732224062.083747666] [controller_server]: Passing new path to controller.
[controller_server-18] [WARN] [1732224062.136554690] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 47.6190 Hz.
[controller_server-18] [WARN] [1732224062.211730474] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 32.2581 Hz.
[controller_server-18] [WARN] [1732224062.264889089] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 45.4545 Hz.
[controller_server-18] [WARN] [1732224062.306051813] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 66.6667 Hz.
[... cut ...]
[controller_server-18] [WARN] [1732224070.339921881] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 23.8095 Hz.
[controller_server-18] [INFO] [1732224070.340033421] [controller_server]: Cancellation was successful. Stopping the robot.
[controller_server-18] [WARN] [1732224070.340054374] [controller_server]: [follow_path] [ActionServer] Client requested to cancel the goal. Cancelling.
[controller_server-18] [INFO] [1732224070.360008931] [controller_server]: Optimizer reset

```

SteveMacenski, 2024-11-21:
Please provide more information about your setup, configuration files, launch files, etc

SteveMacenski, 2024-11-22:
Do you see the issue with our default bringups in `nav2_bringup` or can replicate it with your parameters? 

You also don't show any navigation use, only sim/rviz and slam. It doesn't appear that you have any planner/controller or other navigation systems enabled. Please provide **full details** that we  can work from to try to identify an issue. 

Though, what I'm likely to ask is to check if `nav2_bringup` works for you, then basically change one thing at a time until you find your issue.

BrainBoxTayo, 2024-11-22:
Hi Steve, thanks for the help so far.
I am using nav2_bringup, it's in my bumperbot_bringup.launch.py

```python
# Nav2 Bringup Launch File
            IncludeLaunchDescription(
                os.path.join(get_package_share_directory("nav2_bringup"),
                             "launch", "navigation_launch.py"),
                launch_arguments={"use_sim_time": "True"}.items()
            )
```

I tried modifying the controller frequency in the controller server. I changed it to a 100 and the error just outputs something along the lines of:

Control loop missed its desired rate of 100.0000 Hz. Current loop rate is 111.1871 Hz.

SteveMacenski, 2024-11-23:
I'll be honest, I don't have the information here in this ticket I can use to really say much of anything other than 'start with `nav2_bringup` with out defaults and try to repeat it --> change things one at a time until you find your breakage'. Once you can point to something that breaks the behavior, I can see what's happening in Nav2 or point out that its a setup problem on your side. 

You're getting some ... unique logging and the details of the issue aren't sufficient for me to really point in any particular direction / place

alifazeli00, 2024-11-23:
In addition to the Nav2 package for navigation in ROS 2, what other packages can be used? Are there any references or resources available to explore and compare these packages and their features?

BrainBoxTayo, 2024-11-23:
@SteveMacenski I feel like an idiot, the issue was nav2 was publishing to cmd_vel so I just remapped it to a twist_mux topic. Thank you for the help!

closed 2024-11-23 by BrainBoxTayo