ros-navigation/navigation2 #4769: Add precommit to Nav2

tonynajjar, 2024-11-28:
## Feature request

#### Feature description

Fixing linting issues after CI runs is always a pain. Giving the option for contributors to use pre-commit to fix all linting issues before pushing would lead to a smoother workflow, not to mention the saving of some CI time

SteveMacenski, 2024-12-02:
We have the CI linting-only jobs that should be accelerating that by checking the linters separate of the build / test stages.  https://github.com/ros-navigation/navigation2/blob/main/.github/workflows/lint.yml I believe this already does what you intend?

SteveMacenski, 2024-12-02:
You should be seeing those job results in the GitHub GUI in the same way as a precommit job https://github.com/ros-navigation/navigation2/pull/4761

tonynajjar, 2024-12-02:
Yes it's already great that the linting job runs before the long build and testing jobs but I meant to bring it one step closer to the developer (optionally). So what I'm proposing is a pre-commit config file (or similar) that will basically run the same thing as this linting job but locally before commiting (e.g. pre-commit setups a virtual environment). The user would need to run `pre-commit install` on their local Nav2 repo to "opt-in".
This would help fix linting issues before even committing which would save some time

SteveMacenski, 2024-12-02:
We could do that, yes. However I do wonder if it is necessary with good process. You should be running `colcon test` on the packages that you modify (at least) and as part of that, the linters are run. Running the tests are important before opening a PR to begin with to make sure you don't break existing behavior.

With all that said, if you added a precommit option, I'd merge it.

Nils-ChristianIseke, 2025-02-08:
I can work on this. I would suggest using what is done in ros2_control as a starting point: https://github.com/ros-controls/ros2_control/blob/master/.pre-commit-config.yaml.

SteveMacenski, 2025-02-10:
Adding in codespell might be nice too - figure that's another linting adjacent thing!

closed 2025-03-24 by SteveMacenski