ros-navigation/navigation2 #4770: Fail in loading parameters for controller_server when adding bringup_launch.py into another launch

dvdmc, 2024-11-28:
## Bug report

**Required Info:**

- Operating System: Ubuntu 22.04
  - 
- ROS2 Version: Humble
  - 
- Version or commit hash: ros-humble-navigation2	1.1.16-1jammy.20240830.231501
  - 
- DDS implementation: [Eclipse Cyclone DDS](https://cyclonedds.io/)
  - 

#### Steps to reproduce issue

Including the bringup_launch.py into another launch file (for example, in a supposed tb_nav package) and simply map all the arguments using no namespace.

#### Expected behavior
Nav2 is able to do the bringup

#### Actual behavior
I get errors in controller_server:

```
[lifecycle_manager-5] [ERROR] [1732801697.158933701] [lifecycle_manager_localization]: Failed to change state for node: map_server
[lifecycle_manager-5] [ERROR] [1732801697.158979996] [lifecycle_manager_localization]: Failed to bring up all requested nodes. Aborting bringup.
[bt_navigator-10] [INFO] [1732801697.165182434] [bt_navigator]: 
[bt_navigator-10] 	bt_navigator lifecycle node launched. 
[bt_navigator-10] 	Waiting on external lifecycle transitions to activate
[bt_navigator-10] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[bt_navigator-10] [INFO] [1732801697.165360277] [bt_navigator]: Creating
[controller_server-6] [INFO] [1732801697.171776012] [controller_server]: Creating controller server
[planner_server-8] [INFO] [1732801697.232687539] [global_costmap.global_costmap]: 
[planner_server-8] 	global_costmap lifecycle node launched. 
[planner_server-8] 	Waiting on external lifecycle transitions to activate
[planner_server-8] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-8] [INFO] [1732801697.247297223] [global_costmap.global_costmap]: Creating Costmap
[controller_server-6] [INFO] [1732801697.252753698] [local_costmap.local_costmap]: 
[controller_server-6] 	local_costmap lifecycle node launched. 
[controller_server-6] 	Waiting on external lifecycle transitions to activate
[controller_server-6] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[velocity_smoother-12] [INFO] [1732801697.265705738] [velocity_smoother]: 
[velocity_smoother-12] 	velocity_smoother lifecycle node launched. 
[velocity_smoother-12] 	Waiting on external lifecycle transitions to activate
[velocity_smoother-12] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[lifecycle_manager-13] [INFO] [1732801697.306324939] [lifecycle_manager_navigation]: Creating
[lifecycle_manager-13] [INFO] [1732801697.312174904] [lifecycle_manager_navigation]: Creating and initializing lifecycle service clients
[waypoint_follower-11] [INFO] [1732801697.313905654] [waypoint_follower]: 
[waypoint_follower-11] 	waypoint_follower lifecycle node launched. 
[waypoint_follower-11] 	Waiting on external lifecycle transitions to activate
[waypoint_follower-11] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[waypoint_follower-11] [INFO] [1732801697.325595632] [waypoint_follower]: Creating
[controller_server-6] [INFO] [1732801697.331784491] [local_costmap.local_costmap]: Creating Costmap
[behavior_server-9] [INFO] [1732801697.400330003] [behavior_server]: 
[behavior_server-9] 	behavior_server lifecycle node launched. 
[behavior_server-9] 	Waiting on external lifecycle transitions to activate
[behavior_server-9] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[lifecycle_manager-13] [INFO] [1732801697.461628676] [lifecycle_manager_navigation]: Starting managed nodes bringup...
[lifecycle_manager-13] [INFO] [1732801697.461703820] [lifecycle_manager_navigation]: Configuring controller_server
[controller_server-6] [INFO] [1732801697.462576347] [controller_server]: Configuring controller interface
[controller_server-6] [INFO] [1732801697.462819862] [controller_server]: getting goal checker plugins..
[controller_server-6] [INFO] [1732801697.463074770] [controller_server]: Controller frequency set to 20.0000Hz
[controller_server-6] [INFO] [1732801697.463123176] [local_costmap.local_costmap]: Configuring
[controller_server-6] [INFO] [1732801697.474313005] [local_costmap.local_costmap]: Using plugin "static_layer"
[controller_server-6] [INFO] [1732801697.485957028] [local_costmap.local_costmap]: Subscribing to the map topic (/map) with transient local durability
[ERROR] [kobuki_ros_node-1]: process has died [pid 672, exit code -6, cmd '/root/turtlebot2_ws/install/kobuki_node/lib/kobuki_node/kobuki_ros_node --ros-args --params-file /root/catkin_ws/install/tb_unizar_nav/share/tb_unizar_nav/params/kobuki_params.yaml'].
[controller_server-6] [INFO] [1732801697.487589588] [local_costmap.local_costmap]: Initialized plugin "static_layer"
[controller_server-6] [INFO] [1732801697.487635720] [local_costmap.local_costmap]: Using plugin "obstacle_layer"
[controller_server-6] [INFO] [1732801697.490855754] [local_costmap.local_costmap]: Subscribed to Topics: 
[controller_server-6] [INFO] [1732801697.490970440] [local_costmap.local_costmap]: Initialized plugin "obstacle_layer"
[... 175 lines cut ...]
        }.items(),
    )

    return LaunchDescription([
        # Declare launch arguments
        declarenamespace_cmd,
        declare_use_namespace_cmd,
        declare_slam_cmd,
        declare_map_yaml_cmd,
        declare_use_localization_cmd,
        declare_use_sim_time_cmd,
        declare_nav_params_file_cmd,
        declare_kobuki_params_file_cmd,        
        declare_autostart_cmd,
        declare_use_composition_cmd,
        declare_use_respawn_cmd,
        declare_log_level_cmd,
        declare_use_robot_state_pub_cmd,
        # Include the kobuki group
        declare_kobuki_ros_node,
        start_robot_state_publisher_cmd,
        # Include the bringup launch file
        bringup_launch,
    ])
``

padhupradheep, 2024-11-29:
Hey, are you sure, you are passing the right parameter file? I see that controller server complaining about missing critics and so on.

dvdmc, 2024-11-29:
Hi, this is the parameter file loaded by `default_nav_params_file`, is there any way to confirm it is being loaded?

```
bt_navigator:
  ros__parameters:
    global_frame: map
    robot_base_frame: base_link
    odom_topic: /odom
    bt_loop_duration: 10
    default_server_timeout: 20
    wait_for_service_timeout: 1000
    action_server_result_timeout: 900.0
    navigators: ["navigate_to_pose", "navigate_through_poses"]
    navigate_to_pose:
      plugin: "nav2_bt_navigator::NavigateToPoseNavigator"
    navigate_through_poses:
      plugin: "nav2_bt_navigator::NavigateThroughPosesNavigator"
    # 'default_nav_through_poses_bt_xml' and 'default_nav_to_pose_bt_xml' are use defaults:
    # nav2_bt_navigator::navigate_to_pose_w_replanning_and_recovery.xml
    # nav2_bt_navigator::navigate_through_poses_w_replanning_and_recovery.xml
    # They can be set here or via a RewrittenYaml remap from a parent launch file to Nav2.

    # plugin_lib_names is used to add custom BT plugins to the executor (vector of strings).
    # Built-in plugins are added automatically
    # plugin_lib_names: []

    error_code_names:
      - compute_path_error_code
      - follow_path_error_code

controller_server:
  ros__parameters:
    controller_frequency: 20.0
    costmap_update_timeout: 0.30
    min_x_velocity_threshold: 0.001
    min_y_velocity_threshold: 0.5
    min_theta_velocity_threshold: 0.001
    failure_tolerance: 0.3
    progress_checker_plugins: ["progress_checker"]
    goal_checker_plugins: ["general_goal_checker"] # "precise_goal_checker"
    controller_plugins: ["FollowPath"]
    use_realtime_priority: false

    # Progress checker parameters
    progress_checker:
      plugin: "nav2_controller::SimpleProgressChecker"
      required_movement_radius: 0.5
      movement_time_allowance: 10.0
    # Goal checker parameters
    #precise_goal_checker:
    #  plugin: "nav2_controller/SimpleGoalChecker"
    #  xy_goal_tolerance: 0.25
    #  yaw_goal_tolerance: 0.25
    #  stateful: True
    general_goal_checker:
      stateful: True
      plugin: "nav2_controller::SimpleGoalChecker"
      xy_goal_tolerance: 0.25
      yaw_goal_tolerance: 0.25
    FollowPath:
      plugin: "nav2_mppi_controller::MPPIController"
      time_steps: 56
      model_dt: 0.05
      batch_size: 2000
      ax_max: 3.0
      ax_min: -3.0
      ay_max: 3.0
      az_max: 3.5
      vx_std: 0.2
      vy_std: 0.2
[... 254 lines cut ...]
    FootprintApproach:
      type: "polygon"
      action_type: "approach"
      footprint_topic: "/local_costmap/published_footprint"
      time_before_collision: 1.2
      simulation_time_step: 0.1
      min_points: 6
      visualize: False
      enabled: True
    observation_sources: ["scan"]
    scan:
      type: "scan"
      topic: "scan"
      min_height: 0.15
      max_height: 2.0
      enabled: True

loopback_simulator:
  ros__parameters:
    base_frame_id: "base_footprint"
    odom_frame_id: "odom"
    map_frame_id: "map"
    scan_frame_id: "base_scan"  # tb4_loopback_simulator.launch.py remaps to 'rplidar_link'
    update_duration: 0.02
```

SteveMacenski, 2024-12-02:
> [controller_server-6] [ERROR] [1732801697.533337914] [controller_server]: Couldn't load critics! Caught exception: No critics defined for FollowPath

This is a classic error I see when the params file isn't being passed into the nodes, so there are no parameters given (and DWB as the default in Humble doesn't have any default critics, so it loudly complains). Please look through your paths and make sure that (1) your yaml file is valid, (2) it is located where you have it set so it is properly loadable, and (3) that your launch file is passing this into the bringup correctly. You can see examples of how to do this in the other 2 TB3/TB4 launch files in `main`'s `nav2_bringup`. 

Unfortunately, we don't have developer cycles to support every user with their configuration issues on an individual basis :(

closed 2024-12-02 by SteveMacenski

dvdmc, 2024-12-02:
I tested today using the launch file from `nav2_bringup` directly:

The following works:
```
ros2 launch nav2_bringup bringup_launch.py use_localization:=false use_namespace:=false use_composite:=false map:=/root/catkin_ws/src/tb_nav/maps/dummy.yaml
```

However, if I add a different param file using tabs to check that the route is correct:
```
ros2 launch nav2_bringup bringup_launch.py use_localization:=false use_namespace:=false use_composite:=false map:=/root/catkin_ws/src/tb_nav/maps/dummy.yaml params_file:=/root/catkin_ws/src/tb_nav/params/nav2_params.yaml
```
The errors appear. So it seems that it isn't a problem with my specific launchfile...