ros-navigation/navigation2 #4773: nav2_waypoint_follower build failure due to inability to locate yaml-cpp::yaml-cpp

Mechazo11, 2024-11-29:
Solution proposed in PR #4772 

Hi,
I was attempting to build the ```jazzy``` version of `navigation2` software stack in a source-built ROS 2 Jazzy workspace in Ubuntu 22.04 today. However, I kept getting an linking error specifically inability to find ```yaml-cpp::yaml-cpp``` target defined in the [robot_localization](https://github.com/Mechazo11/robot_localization) package. Note, even though the hyperlink goes to my fork of robot_localization, I made no changes to it. Cannot remember why I forked robot_localization in the first place but the solution I proposed in PR #4772 uses my fork of robot_localization.

**System Information**
OS: Ubuntu 22.04 (also applicable to Ubuntu 24.04)
ROS 2 Version: ROS 2 Jazzy
Version: ```jazzy``` branch
DDS Implementation: CycloneDDS

#### Steps to reproduce issue
Assuming you have access to a base ROS 2 Jazzy workspace (binary or built from source), please build my moveit2 workspace first and git clone and build the ```main``` branch of `navigation2` stack to reproduce the error (given below)
```
<source your global ROS 2 workspace here first>
cd ~
git clone https://github.com/Mechazo11/moveit2_nav2_jazzy_ws.git
cd moveit2_nav2_jazzy_ws
vcs import src < moveit2_nav2_jazzy.repos --recursive
rosdep install -r --from-paths src --rosdistro jazzy -i -y
colcon build --packages-up-to moveit_planners_ompl --packages-ignore stomp stomp_moveit moveit_planners_stomp --cmake-args -DCMAKE_BUILD_TYPE=Release
source ./install/setup.bash
git clone https://github.com/ros-navigation/navigation2.git
colcon build --packages-ignore stomp stomp_moveit moveit_planners_stomp --cmake-args -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS="-w"
```

Please note without the ```-w``` flag, ```nav2_amcl``` raises a header depreciation warning which the compiles counts as an error stopping the build. This is another potential error that may need another PR to fix.

#### Expected behavior
```nav2_waypoint_follower``` builds successfully.

#### Actual behavior
```nav2_waypoint_follower``` fails with the following error.

```console
--- stderr: nav2_waypoint_follower                                                             
CMake Error at /home/tigerwife/moveit2_nav2_jazzy_ws/install/robot_localization/share/robot_localization/cmake/robot_localizationExport.cmake:61 (set_target_properties):
  The link interface of target "robot_localization::rl_lib" contains:

    yaml-cpp::yaml-cpp

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /home/tigerwife/moveit2_nav2_jazzy_ws/install/robot_localization/share/robot_localization/cmake/ament_cmake_export_targets-extras.cmake:9 (include)
  /home/tigerwife/moveit2_nav2_jazzy_ws/install/robot_localization/share/robot_localization/cmake/robot_localizationConfig.cmake:41 (include)
  CMakeLists.txt:20 (find_package)
```

jncfa, 2024-11-29:
I dont think it make much sense adding a find_package call in nav2_waypoint_follower if the package isnt directly used...

You should instead add the missing package to the ament_export_dependencies call, this should also fix your issue: https://github.com/Mechazo11/robot_localization/blob/ros2/CMakeLists.txt#L352-L371

SteveMacenski, 2024-12-02:
> However, I kept getting an linking error specifically inability to find yaml-cpp::yaml-cpp target defined in the [robot_localization](https://github.com/Mechazo11/robot_localization) package.

Use the appropriate `jazzy-devel` branch, not `ros2` which targets rolling. This is all tested extensively in Jazzy / Rolling, so we do know this works and our CI continues to turn over and function as expected in both distributions. It seems like you're accidentally mixing and matching branches from different distributions which have different styles of CMake. This is also what @jncfa points out :-)

closed 2024-12-02 by SteveMacenski

Mechazo11, 2024-12-02:
@SteveMacenski  acknowledged, I will try out this suggestion.