ros-navigation/navigation2 #4776: RecoveryNode should return RUNNING on first child's failure and second child's success

tonynajjar, 2024-12-03:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 24
- ROS2 Version:
  - jazzy
- Version or commit hash:
  - main

#### Steps to reproduce issue

I created a quick dirty test, I can share it later

#### Expected behavior

1- If the first child of the RecoveryNode fails, I would expect the second child to be ticked after the specified `bt_loop_duration` (aka at the next tick).

2- If the second child of the RecoveryNode succeeds, I would expect the first child to be (re)ticked after the specified `bt_loop_duration` (aka at the next tick).

#### Actual behavior

In both cases 1 and 2, the next child is ticked immediately (at the next iteration of the internal while loop) 

#### Additional information

The solution would be to simply make the RecoveryNode return RUNNING [here](https://github.com/ros-navigation/navigation2/blob/3ee384df622ed964680452dcd53da6d0e9a15514/nav2_behavior_tree/plugins/control/recovery_node.cpp#L67) and [here](https://github.com/ros-navigation/navigation2/blob/3ee384df622ed964680452dcd53da6d0e9a15514/nav2_behavior_tree/plugins/control/recovery_node.cpp#L101)

SteveMacenski, 2024-12-03:
I don't disagree - will look at PR :-)

tonynajjar, 2024-12-05:
I'll reopen the issue until the last comment is addressed just so we don't forget about it

SteveMacenski, 2024-12-06:
@tonynajjar should we consider reverting the PR changes then?

tonynajjar, 2024-12-10:
> @tonynajjar should we consider reverting the PR changes then?

Yeah let's do it until we have a design consensus

SteveMacenski, 2024-12-10:
I'll ping Davide in another thread to get his take

SteveMacenski, 2024-12-11:
Davide says that this is fine and expected. @tonynajjar should we close this out? I agree that it doesn't make things fully introspectable if looking at the tree output states, but it is in line with the library's use and other built-in nodes. We could change them here in Nav2, but it doesn't seem like updates to changing BT.CPP's to do that as well is wanted

tonynajjar, 2024-12-11:
Some more information on why this is fine and expected would have been helpful since this is a subjective design question.
This design decision has the significant drawback of making the BT "less reactive", especially with BT nodes with a retry behavior -> since the control node does not return before all the retries are done, other parts of the tree that may want to preempt the retrying are not executed. It essentially acts similar to a blocking function call in a high-rate loop.

jplapp, 2024-12-16:
We keep stumbling on this issue - a simple example on how this is non-intuitive is this:
`<Timeout><RetryUntilSuccessful num_attempts="-1"><AlwaysFailure/></RetryUntilSuccessful></Timeout>` 

Which will _never time out_ because Retry has its "own" loop. Unfortunately[ this seems to be a design decision](https://github.com/BehaviorTree/BehaviorTree.CPP/pull/723).

facontidavide, 2024-12-16:
I will be happy to modify **Retry** as I did for **Sequence**

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/include/behaviortree_cpp/controls/sequence_node.h#L37

Will that help?

SteveMacenski, 2025-03-26:
@facontidavide I think it would be good to have that async option on all of the control nodes (and possibly not just 'async' but the default always-on behavior) 

Closing this ticket - it looks like there's not a concensus to be made here at the moment. I generally agree that there should only be 1 node ticked per BT tick so that the flow is fully introspect-able, but right now BT.CPP's control flow nodes don't do this (though some have the option) so I don't think we should go crazy on Nav2's custom BT control flow nodes as those are only part of the source. This seems like an 'all or nothing' topic since there's mixes of Nav2 and BT.CPP provided control nodes in most folks' trees. 

If at some point BT.CPP has this globally and/or someone would like to tackle that, I'd be happy to have that reflected in Nav2's nodes as well! I can see the argument for reactivity to immediately trigger the next behavior, however.

closed 2025-03-26 by SteveMacenski

facontidavide, 2025-03-27:
I will be happy to discuss this.
Unfortunately, changing the defaultbehavior of all the nodes is a breaking change that would motivate a version 5.0.
I am not against it, since I have other ABI breaking changs in mind that I am looking for.