ros-navigation/navigation2 #4778: Goals get randomly aborted

patham9, 2024-12-03:
Nav2 is not reliable and needs to be re-invoked multiple times before it manages to get to a goal location.
Specifically, it aborts the goal randomly. Using the in-built Turtlebot4 simulation I would have expected a reliability that matches the ROS1 navigation stack.

**Required Info:**

- Operating System:
  - Ubuntu 24
- ROS2 Version:
  - ROS2 Jazzy
- Version or commit hash: ros-jazzy-navigation2	1.3.2-1noble.20240922.123501
- DDS implementation: Fast-RTPS

#### Steps to reproduce issue

Normal Turtlebot 4 bringup.

#### Expected behavior

No random aborts.

#### Actual behavior

Random goal abortion

SteveMacenski, 2024-12-03:
Please provide logs, a situation description, and preferably something we can reproduce. Currently, this description doesn't provide what's needed to be actionable. I have no such issues in my Jazzy docker container with Nav2 with the current binaries. We have many system tests that run reliably and hundreds of companies using Nav2 in deployed applications, so I believe it is generally reliable :wink: 

But, if we can find where the disconnect is for your application, then we can work to resolve it

patham9, 2024-12-03:
Thank you for your quick response! Yes that's what I also guess due to its widespread use, surprisingly it just didn't work reliably for me personally yet. Maybe there is a violated timing assumption, as when the operating system is also running compute-heavy machine learning models, delay spikes can sometimes occur.
Just to confirm, your docker also runs binary version 1.3.2-1noble.20240922.123501?
Which logs would you prefer to see?

SteveMacenski, 2024-12-04:
> Maybe there is a violated timing assumption, as when the operating system is also running compute-heavy machine learning models, delay spikes can sometimes occur.

If those spikes also trash your networking bandwidth that could explain alot. That would then be less to do with nav2 and more to do with the resource allocation in your system. But what do your logs say -- just dump me the logs from your Nav2 stdout terminal (or isolate it to just give me the Nav2 bringup parts of it). I want to see "why" its failing, whether its network related, algorithmically not working well, tuning, etc. 

>  Just to confirm, your docker also runs binary version 1.3.2-1noble.20240922.123501?

I see `1.3.3-1noble.20241115.195529`

- `sudo docker run -it --net=host --privileged -v .:/root/jazzy_ws --volume="${XAUTHORITY}:/root/.Xauthority" --env="DISPLAY=$DISPLAY" -v="/tmp/.gazebo/:/root/.gazebo/" -v /tmp/.X11-unix:/tmp/.X11-unix:rw --shm-size=1000mb osrf/ros:jazzy-desktop-full`
- apt update
- rosdep init
- rosdep update
- apt install ros-jazzy-nav2-bringup ros-jazzy-navigation2

Then run the normal demos, all seems fine to me and the system tests.

SteveMacenski, 2024-12-10:
@patham9 any update?

patham9, 2024-12-11:
I apologize that it took me so long, thank you for the reminder.
I hope the event log is useful: https://gist.github.com/patham9/276120eaf098bb2eeec274cff01a056f
particularly the first error seems relevant: "Failed to create a plan from potential when a legal potential was found. This shouldn't happen.".
Please let me know if this is helpful, I am happy to help investigating the issue further as I now found it also appears for a colleague of mine.

SteveMacenski, 2024-12-14:
That is very odd. I know you commented on the ticket with the same error (https://github.com/ros-navigation/navigation2/issues/4655). I've never seen this happen personally in all my years and until that was filed, no one's said they've had a problem either in 10+ years. 

The only 2 change I see made anywhere near there is (1) adding in cancel goal checkers and terminating if the planning time is exceeded. Are you seeing this happen when the planning time you set in your config file is exceeded by the planner? I could tentatively see the error state that it introduces not being properly handled in some case that we can plug the hole on. Or (2) we removed checking if the start cell is occupied, so it could be that its not getting a path because you're in occupied space -- are you seeing this happen in that case?

Please also provide your config file

patham9, 2025-01-15:
Google search reveals further cases, not always was a response given, here a recent one: https://robotics.stackexchange.com/questions/109980/failed-to-create-a-plan-from-potential-when-a-legal-potential-was-found-this-sh

Case 2 from 4 years ago: https://answers.ros.org/question/350510/

I dived into the codebase in the meanwhile and I think I have identified the bug which makes the "that shouldn't happen" printout a reachable code path with the default turtlebot4 simulation parameters.

SteveMacenski, 2025-03-14:
@patham9 any update on this? It sounds like you may have found the issue / solution for here and https://github.com/ros-navigation/navigation2/issues/4655

vahapt, 2025-03-24:
@SteveMacenski I'm having the same problem with a real robot passing through a narrow exit (that's the only exit from the room). Due to sensor errors (lidar specifically) global_costmap/obstacle_layer is updated and thus walls slightly move back and forth. Since the exit is too tight, this error causes the exit to be seen as blocked.

When I remove obstacle_layer from global_costmap totally and navigate only with static_layer & inflation_layer everything works without any problems (except if there's a real obstacle :) )

I've also tried to
reduce -> global_costmap/inflation_layer/inflation_radius=0.5
keep -> local_costmap/inflation_layer/inflation_radius=0.7
This helped but I don't like it.

This behavior seems to be normal as we want the planner to avoid obstacles. But even if there are marked obstacles I'd prefer it draw the path (ignoring obstacle layer) if it's the only possible path instead of aborting the whole navigation process.

Detail data:
- Robot footprint: "[ [0.50, 0.40], [0.50, -0.40], [-0.50, -0.40], [-0.50, 0.40] ]" (100cm length, 80cm width)
- Pivot is at the center
- Exit width: 100cm
- Differential drive

patham9, 2025-05-01:
@SteveMacenski 
The issue seems related to parameter choice, but I notice there are two separate issues which are slightly different:

- @vehapt's issue: the given obstacles do not seem to allow for a safe solution with the existing sensory noise / accuracy of localization, without altering inflation radius (which I guess isn't safe as it could lead to collisions in practice if set too low), in this case maybe more accurate sensors could be necessary.

- Goal is aborting with the "This shouldn't happen." message: In my case this happens when the robot is navigating areas of the map that have not been mapped yet. When the map is not yet very stable the localization can jump a bit, and so do the points I set as goal. These can then be a bit closer to obstacles than I had set them to be initially, which seems to cause the issue in combination with the inflation parameters I had used. I am still not 100% sure about this issue though, this is just a summary of what I observed so far.

padhupradheep, 2025-05-08:
@vahapt: did you solve your issue?
@patham9: would it make sense to close the issue for now? Since you say: 

>These can then be a bit closer to obstacles than I had set them to be initially, which seems to cause the issue in combination with the inflation parameters I had used. 

there is not much we can update in the stack.

patham9, 2025-05-09:
@padhupradheep I agree, I think it can be closed for now, as different parameters seem to resolve it.

closed 2025-05-09 by SteveMacenski