ros-navigation/navigation2 #4794: Unpopular feature request: add costmap source to collision monitor

tonynajjar, 2024-12-12:
## Feature request

#### Feature description

I know the collision monitor bypasses the costmap on purpose for faster response but for my application I care less about faster response and more about leveraging the persistence of some of the costmap layers because I lack a 360 degrees perception coverage.
So I'm proposing a CM source that does collision-checking against a costmap it subscribes to.
What do you think?

#### Implementation considerations

SteveMacenski, 2024-12-12:
I'd merge a PR as long as the documentation in the readme and website warned against doing it for the reasons you know about (but that you _can_ if needed)

tonynajjar, 2025-06-04:
I gave it a shot here: https://github.com/angsa-robotics/navigation2/pull/7
So now it's an even simpler first issue for someone to integrate this into Nav2 `main` and to battle-test it against more scenarios (e.g. rolling/non-rolling costmaps)

Lotusymt, 2025-09-24:
Hi all — I’d like to take this issue.

Progress:
  Implemented a CostmapSource for Collision Monitor, adapted from @tonynajjar’s prototype (angsa-robotics#7).

Added unit tests (gtest):
  Lethal cell → exactly one point at expected center.
  Stale messages are ignored (timeout respected).
  Cost threshold logic (254 vs 253) works.
  Frame transform (map→base_link) is applied correctly.

Questions:
 1. For tests: are unit tests sufficient, or would you prefer that I also add a small rosbag-driven launch test (deterministic, 10–15s bag) to exercise the data flow end-to-end in CI?
 2. Any specific distro targets I should verify locally (Humble/Iron/Rolling)?

Next steps:
  1. Add whichever integration test you prefer (or skip if unit tests are enough).
  2. Update README/website to note trade-offs (latency vs persistence).
  3. Add a brief CHANGELOG.rst entry and ensure linters pass.
  
Thanks—please let me know if I’m missing anything important for this PR.

SteveMacenski, 2025-09-24:
On distros, typically we add things to the main branch which targets rolling but can be compiled against jazzy and kilted. If the changes don’t break ABI, then we can backport!

For testing, unit and integration tests would be necessary. Right now we don’t have any rosbag-based testing in Nav2 so I would actually be very interested in your thoughts about a thoughtful design pattern that we could leverage with an example from this PR for the future - if you’re up for it!

Lotusymt, 2025-09-26:
Thanks! Targeting main (Rolling) with ABI stability for Jazzy/Kilted. I’ll add unit + integration tests. I’d love to explore a rosbag-aided integration example and will share thoughts in a short outline by this weekend.

Lotusymt, 2025-10-16:
Hi Steve — thanks for the detailed explanation, that really helped clarify the design rationale!
I’ve started drafting the dataset-based test following the Robot Localization example you mentioned. I’m narrowing v1 to CollisionMonitor(costmap source) with a small dataset fixture. I plan to include a minimal test harness (launch + a tiny publisher for /cmd_vel_in) to play the bag and evaluate CM’s output (/cmd_vel) over time.

For v1, I’ll use simple trajectory/metrics checks rather than a single snapshot, e.g.:

   1. Time-to-stop after obstacle appears
   2. Hold-stop % while obstacle persists
   3. Time-to-resume after obstacle clears
   4. No-false-stop in clean segments
   5. Max overshoot during stop

Would this direction sound good to you? If so, I’ll proceed and open a Draft PR once it’s runnable to get your feedback before tightening thresholds and polishing the structure. Thanks again!