ros-navigation/navigation2 #4802: The first-time failure in the IsBatteryLow condition check

Tacha-S, 2024-12-18:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 24.04
- ROS2 Version:
  - Jazzy binaries
- Version or commit hash:
  - ros-jazzy-navigation2   1.3.3-1noble.20241115.195529
- DDS implementation:
  - Cyclone DDS

When performing the check with the IsBatteryLow condition for the first time, it almost always results in FAILURE because a subscriber is created and then spin_some is executed shortly thereafter.

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

SteveMacenski, 2024-12-18:
What is your suggested solution? This seems like a reasonable choice to me and the design pattern we do in multiple BT nodes. I can see how it takes some time to 'warm up' to get the first message since its just starting, but I don't know how actionable a change would be. 

If your application is sensitive to that, you could use a service call instead of a topic to obtain the battery information or possibly tick and have a delay action node to tick again to wait for a message on startup

Tacha-S, 2024-12-18:
Constructing a behavior tree while considering whether the nodes will function correctly on the first run can easily become a source of bugs and make the behavior tree itself difficult to read.

Even if sufficient time is allocated for warmup, subscribers won't be created unless the nodes are executed at least once. Therefore, ensuring sufficient time between launching the node and triggering the behavior tree execution is completely ineffective.

How about calling `initialize()` once in the constructor?

SteveMacenski, 2024-12-19:
Some thoughts before responding to your items:

I'm noticing that the [BT Action Nodes](https://github.com/ros-navigation/navigation2/blob/main/nav2_behavior_tree/plugins/action/compute_path_to_pose_action.cpp#L23-L29) and [BT Service Nodes](https://github.com/ros-navigation/navigation2/blob/main/nav2_behavior_tree/plugins/action/clear_costmap_service.cpp#L23-L28) construct their service/action clients in the constructor of those nodes. 

We could very easily just create all ROS interfaces for BT nodes (like `isBatteryLow`) in the constructor as well. The only downside of this is that the topic themselves are parameters. The other parameters being dynamically adjustable in `initialize()` make sense. 

So, we could create the ROS interfaces, with the current input port status, in the constructor and also leave the creation in `initialize()` when the topic has changed only. We do that very similarly in the `isBatteryLow` BT node. I think we'd just want to branch out a new `createROSInterfaces()` method that is called in both the constructor and initialize, so we don't duplicate code. 

I think that would solve the issue 

---

> How about calling initialize() once in the constructor?

We likely cannot, since they get parameters that need to be gotten in the tick due to problems discussed in https://github.com/ros-navigation/navigation2/pull/3988 -- this is actually how we _used to_ handle it. Perhaps @maksymdidukh can also chime in his thoughts.  But also see above. 

>  Alternatively, using wait_for_message() to retrieve the message might be a good idea, but it would likely be rejected as it wouldn't work on distros where this functionality is not implemented.

We don't need to backport everywhere if we cannot, that could be a viable solution. Waiting for message would need to block for no more than the tick rate, the same way we do for services and actions that return the `RUNNING` state while we're waiting for a service/action timeout and respecting the tick rate for reactive behavior ([example](https://github.com/ros-navigation/navigation2/blob/main/nav2_behavior_tree/include/nav2_behavior_tree/bt_service_node.hpp#L187-L217)). 

Though, waiting for a message for, lets say 250ms, every time we retick a tree would be blocking all other productive action frequently. I'm not sure exactly this is the best way. 

> Constructing a behavior tree while considering whether the nodes will function correctly on the first run can easily become a source of bugs and make the behavior tree itself difficult to read.

This could just pedantic due to the nature of the `isBatteryLow` condition and the related logic around autonomous docking -- but I don't see that much of a problem. If you have a BT branch that is under a reactive control node, it'll be checked frequently for state changes. Waiting a couple of ticks to get the message to realize that its low seems pretty minimal. If this was another node, I could see this being higher-impact, which is also why its interesting to discuss this since the pattern would change for many BT nodes if we changed this one.

SteveMacenski, 2025-01-14:
@Tacha-S thoughts / updates?

Tacha-S, 2025-01-14:
While studying the tick rate of behavior trees, I forgot to reply.
Upon reconsideration, the following suggestion seems very promising.

> So, we could create the ROS interfaces, with the current input port status, in the constructor and also leave the creation in initialize() when the topic has changed only in tick(). We do that very similarly in the isBatteryLow BT node. I think we'd just want to branch out a new createROSInterfaces() method that is called in both the constructor and initialize, so we don't duplicate code.

SteveMacenski, 2025-01-14:
Is this something you'd be open to contributing? :smile:

Tacha-S, 2025-01-15:
Of course, I will create a PR with the proposed changes.

closed 2025-03-10 by SteveMacenski