ros-navigation/navigation2 #4814: WHY the humble version of navigation2 is missing controller_exceptions.hpp in the nav2_core/inludes folder ?

G-Apple1, 2024-12-23:
## Bug report

**Required Info:**

- Operating System: ubuntu22.04
  - 
- ROS2 Version: humble
  - 
- Version or commit hash: humble
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```
The humble version of navigation2 is missing controller_exceptions.hpp in the nav2_core/inlude folder. Why ?
```
[image]

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

mini-1235, 2024-12-23:
The latest humble release is 1.1.17, which does not have controller exception implemented, you should use the correct branch code

SteveMacenski, 2024-12-23:
Precisely. Changes are made over the evolution of the project and not all changes can be backported to older distributions using best-practices ABI/API stability policy requirement s

closed 2024-12-23 by SteveMacenski