ros-navigation/navigation2 #4815: Correct writing of plugin.xml

mini-1235, 2024-12-23:
Hi, according to the latest navigation2 document, the file contains the following information:

- library path: Plugin’s library name and it’s location.
- class name: Name of the class.
- class type: Type of class.
- base class: Name of the base class.
- description: Description of the plugin.

However, I have noticed that in the jazzy and rolling branch have removed most of the name attribute, e.g.
```
<class type="nav2_theta_star_planner::ThetaStarPlanner" base_class_type="nav2_core::GlobalPlanner">
        <description>An implementation of the Theta* Algorithm</description>
</class>
```
except for [constrained-smoother](https://docs.nav2.org/configuration/packages/configuring-constrained-smoother.html), [rviz-plugins](https://github.com/ros-navigation/navigation2/blob/main/nav2_rviz_plugins/plugins_description.xml), and [drive-on-heading](https://github.com/ros-navigation/navigation2/blob/main/nav2_behaviors/behavior_plugin.xml). I wonder if it is necessary to use the name attribute in the latest version of navigation2, when we are writing our own plugins

and also I think there is an error in the [navigation2_tutorial](https://github.com/ros-navigation/navigation2_tutorials/blob/rolling/nav2_sms_behavior/behavior_plugin.xml), where the xml does not match with the nav2 document for the rolling and jazzy branch

mini-1235, 2025-01-02:
After further digging, I believe the `name` attribute now works as a lookup name in pluginlib,
```
Pre-Groovy versions of pluginlib required specifying a "lookup name" for exported classes in both the plugin description file and export macro. This lookup name acted as an alias for the true class name -- the true class name was not used in the user facing interface. The reason why this lookup alias was used instead of the real name was due to a technical limitation in older versions.

One can now use the real name of a class instead of the lookup name. However, if users want to still use the lookup name, they can add it in their plugin description file. 
```
and most of the ros2 pluginlib tutorial ignored this attribute.

Should we remove all the name attributes in the latest version(as we are using the "::" syntax to replace "/" in the newer version) and update the documentation to avoid ambiguity?

SteveMacenski, 2025-01-02:
> according to the latest navigation2 document

Can you point me to this so I can look?

> I wonder if it is necessary to use the name attribute in the latest version of navigation2, when we are writing our own plugins

The name attribute is optional, when not set, it uses the `type`. This allows you to set arbitrary names, that we just set using a `/` but could also be `foo_bar`. We removed it recently to standardize all plugins to use the `::` nomenclature instead of mixed some plugins using that and others using `/`. 

> Should we remove all the name attributes in the latest version(as we are using the "::" syntax to replace "/" in the newer version) 

There's no 'should', but you absolutely could! 

> and update the documentation to avoid ambiguity?

Please point me to that :-)

mini-1235, 2025-01-02:
> The name attribute is optional, when not set, it uses the type. This allows you to set arbitrary names, that we just set using a / but could also be foo_bar. We removed it recently to standardize all plugins to use the :: nomenclature instead of mixed some plugins using that and others using /.

Thanks for clarifying that 

It was the [writing new behavior plugin](https://docs.nav2.org/plugin_tutorials/docs/writing_new_behavior_plugin.html) guide, where it says 
> Next step would be to create plugin’s description file in the root directory of the package. For example, behavior_plugin.xml file in our tutorial package. This file contains following information
- library path: Plugin’s library name and it’s location.
- class name: Name of the class.
- class type: Type of class.
- base class: Name of the base class.
- description: Description of the plugin.
```
<library path="nav2_sms_behavior_plugin">
  <class type="nav2_sms_behavior::SendSms" base_class_type="nav2_core::Behavior">
    <description>This is an example plugin which produces an SMS text message recovery.</description>
  </class>
</library>
```
It mentions the class name but give an example without it, which kind of make me confused at first. For me, it would be better if we either remove the name part or have more description for those attributes:)

What makes me more confusing is actually the [navigation tutorial](https://github.com/ros-navigation/navigation2_tutorials), where is still have 
```
<class name="nav2_sms_behavior/SendSms" type="nav2_sms_behavior::SendSms" base_class_type="nav2_core::Behavior">
```
on the rolling and jazzy branch.
May I open a PR for the tutorial repo to stay consistent with the navigation2 docs, and also make some small updates on the nav2 docs?

SteveMacenski, 2025-01-02:
>  For me, it would be better if we either remove the name part or have more description for those attributes:)

I'd be happy to merge a PR that either does that or adds in a note that if `name` isn't set, the name will be the `type`. 

> What makes me more confusing is actually the [navigation tutorial](https://github.com/ros-navigation/navigation2_tutorials), where is still have

Ah, I would appreciate a PR to fix those :-) Admittedly the tutorials aren't top-of-mind as we're making updates and changes. That is just an oversight

SteveMacenski, 2025-01-03:
I think with your PRs this can be closed!

closed 2025-01-03 by SteveMacenski