ros-navigation/navigation2 #4816: Build failed with latest ros2 rolling

chanhhoang99, 2024-12-24:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 24.04.1 LTS
- ROS2 Version:
  - rolling ridley 
- Version or commit hash:
  - 771eca41bb7b6ad26856c434dd79e48a399c4601
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
```
Build ros2 from source: https://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html

Then follow instruction to build nav2 https://docs.nav2.org/development_guides/build_docs/index.html
```

#### Expected behavior
Build should be passed
#### Actual behavior
Received failure log:
```
In file included from /home/server/nav2_ws/src/navigation2/nav2_util/src/robot_utils.cpp:21:
/home/server/ros2_ws/install/tf2/include/tf2/tf2/convert.h:39:7: error: #warning CONVERT_HEADER_DEPERCATION [-Werror=cpp]
   39 |     # warning CONVERT_HEADER_DEPERCATION
      |       ^~~~~~~
In file included from /home/server/nav2_ws/src/navigation2/nav2_util/include/nav2_util/robot_utils.hpp:27,
                 from /home/server/nav2_ws/src/navigation2/nav2_util/src/robot_utils.cpp:22:
/home/server/ros2_ws/install/tf2/include/tf2/tf2/time.h:36:7: error: #warning TIME_HEADER_DEPERCATION [-Werror=cpp]
   36 |     # warning TIME_HEADER_DEPERCATION
      |       ^~~~~~~
In file included from /home/server/nav2_ws/src/navigation2/nav2_util/include/nav2_util/robot_utils.hpp:28:
/home/server/ros2_ws/install/tf2/include/tf2/tf2/transform_datatypes.h:39:7: error: #warning TRANSFORM_DATATYPES_HEADER_DEPERCATION [-Werror=cpp]
   39 |     # warning TRANSFORM_DATATYPES_HEADER_DEPERCATION
      |       ^~~~~~~
In file included from /home/server/nav2_ws/src/navigation2/nav2_util/src/costmap.cpp:18:
/home/server/ros2_ws/install/tf2/include/tf2/tf2/LinearMath/Quaternion.h:23:7: error: #warning QUATERNION_HEADER_DEPERCATION [-Werror=cpp]
   23 |     # warning QUATERNION_HEADER_DEPERCATION
      |       ^~~~~~~
cc1plus: all warnings being treated as errors
```

#### Additional information
There is change from upstream of ros2/tf2 headers, should we use -Wno-error=cpp to ignore this error ?

SteveMacenski, 2025-01-02:
A few updates were made before the holidays in TF -- we have an open PR for updating to resolve the API change here https://github.com/ros-navigation/navigation2/pull/4732. I retriggered CI to check if we're all building again and will merge once ready!

SteveMacenski, 2025-01-02:
@chanhhoang99 are you building geometry2 rolling from source? I'm not seeing it in the binaries yet and there hasn't been a rolling release since the depreciation warnings were added.

chanhhoang99, 2025-01-03:
Hi @SteveMacenski ,
Yes, I built geometry2 from source.

SteveMacenski, 2025-01-03:
@chanhhoang99 we should have this resolved sometime this month for source builds, for now you could use the tagged version of rolling's geometry_msgs instead of the latest on the main branch (`0.40.0`). That would fix your source build version (unless you critically needed something that has been added in the last ~6-8 weeks). 

We could also update Nav2's CI to have geometry2 built from source as well and merge this in, but that would break `main` for all other users on `rolling`'s binaries. I'd wager there are more binary-rolling users than source-rolling users, so I want to make sure I'm considerate of other developer's needs.

SteveMacenski, 2025-01-03:
L.O.L. for me

The rolling release just happened 20 hours ago. I kick started that PR job and likely should be able to merge this within an hour to resolve the problem :laughing:

SteveMacenski, 2025-01-04:
Merged! Thanks for the report

closed 2025-01-04 by SteveMacenski

AdarshKaran, 2025-01-23:
Hi, I am experiencing a similar issue with the following setup:

**System Details**:
- **OS**: Ubuntu 24.04.1 LTS
- **ROS 2 Distro**: Rolling
- **DDS Implementation**: Fast DDS (default)

### **Installation Steps**:
1. Installed ROS 2 Rolling from binaries (via Debian packages) as given in [ROS 2 Rolling Installation Guide](https://docs.ros.org/en/rolling/Installation/Ubuntu-Install-Debs.html). This has been upgraded to the latest version.

2. Followed the Nav2 build instructions from [Nav2 Development Guide](https://docs.nav2.org/development_guides/build_docs/index.html).

---

### **Issue**:
After cloning the "main" branch of the repo as per Step 2. above, I ran the following command:
```bash
rosdep install -y \
  --from-paths ./src \
  --ignore-src
colcon build \
  --symlink-install
```
All the dependencies were installed but I got the following error. 
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
nav2_bringup: Cannot locate rosdep definition for [slam_toolbox]
```
But, according to the Nav2 build docs this is expected and can be ignored. After this, I encountered the following error log and the build failed:
```
Starting >>> nav2_common
Starting >>> nav_2d_msgs
Starting >>> nav2_loopback_sim
Finished <<< nav2_common [0.32s]                                   
Starting >>> nav2_msgs
Starting >>> nav2_voxel_grid
Starting >>> opennav_docking_core
Finished <<< nav_2d_msgs [0.33s]
Starting >>> dwb_msgs
Finished <<< opennav_docking_core [0.09s]                        
Finished <<< nav2_voxel_grid [0.28s]                             
--- stderr: nav2_loopback_sim                                    
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:363: UserWarning: Unbuilt egg for pytest-repeat [unknown version] (/usr/lib/python3/dist-packages)
  self.local_index = Environment(self.shadow_path + sys.path)
---
Finished <<< nav2_loopback_sim [0.83s]
--- stderr: nav2_msgs                                             
Traceback (most recent call last):
  File "/opt/ros/rolling/lib/rosidl_generator_type_description/rosidl_generator_type_description", line 50, in <module>
    sys.exit(main())
             ^^^^^^
  File "/opt/ros/rolling/lib/rosidl_generator_type_description/rosidl_generator_type_description", line 46, in main
    generate_type_hash(args.generator_arguments_file)
  File "/opt/ros/rolling/lib/python3.12/site-packages/rosidl_generator_type_description/__init__.py", line 162, in generate_type_hash
    with include_path.open('r') as include_file:
         ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/pathlib.py", line 1015, in open
    return io.open(self, mode, buffering, encoding, errors, newline)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/opt/ros/rolling/share/geometry_msgs/msg/PoseStampedArray.json'
gmake[2]: *** [CMakeFiles/nav2_msgs__rosidl_generator_type_description.dir/build.make:153: rosidl_generator_type_description/nav2_msgs/msg/CollisionMonitorState.json] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:215: CMakeFiles/nav2_msgs__rosidl_generator_type_description.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< nav2_msgs [2.47s, exited with code 2]
Aborted  <<< dwb_msgs [3.61s]                                

Summary: 5 packages finished [4.10s]
  1 package failed: nav2_msgs
  1 package aborted: dwb_msgs
  2 packages had stderr output: nav2_loopback_sim nav2_msgs

```

Specifically, this line
```
FileNotFoundError: [Errno 2] No such file or directory: '/opt/ros/rolling/share/geometry_msgs/msg/PoseStampedArray.json'
```

Any guidance would be appreciated! Thank You!

chanhhoang99, 2025-01-23:
@AdarshKaran ,
Please try installing ros2 from source then source ros2 setup.bash file
Below is mine ros2 build workspace, you should see missing file

server@server-sim-1:~/ros2_ws/install$ find . -name PoseStam*
./geometry_msgs/share/geometry_msgs/msg/PoseStamped.idl
./geometry_msgs/share/geometry_msgs/msg/PoseStampedArray.idl
./geometry_msgs/share/geometry_msgs/msg/PoseStamped.json
./geometry_msgs/share/geometry_msgs/msg/PoseStampedArray.json
./geometry_msgs/share/geometry_msgs/msg/PoseStampedArray.msg
./geometry_msgs/share/geometry_msgs/msg/PoseStamped.msg

SteveMacenski, 2025-01-24:
Right now, you need to build geometry_msgs from source for Nav2 rolling. We made a big update that involved a new message definition that was added to geometry_msgs but is not yet released into binaries. 

Clean your workspace's build/install spaces; clone  geometry_msgs rolling branch; build -- you should be good to go. On the next rolling sync, this goes away!

AdarshKaran, 2025-01-29:
> Right now, you need to build geometry_msgs from source for Nav2 rolling. We made a big update that involved a new message definition that was added to geometry_msgs but is not yet released into [...]

Thank You so much @SteveMacenski . It works fine now! I built the geometry_msgs package from source and it worked.
Thank You @chanhhoang99 for your help too.