ros-navigation/navigation2 #4818: parameter "debug_visualizations" of SmacHybridPlanner is not working

YoungRicky7, 2024-12-25:
- Operating System:
  ubuntu 22.04
- ROS2 Version:
  humble
- Version or commit hash:
 10aa732f50052c8f92b9591a2e985b5c66cf7f26
- DDS implementation:
  Fast-RTPS

#### Steps to reproduce issue
when I ran the planner with the parameter "debug_visualizations" set as true,there was no topic named as "/expansion" which was described in the blob [https://github.com/ros-navigation/navigation2/tree/main/nav2_smac_planner]

```

#### Expected behavior
as what is showed in the blob
![image](https://github.com/user-attachments/assets/99d86efd-ff92-436b-a211-51b16858b257)

#### Actual behavior
I used ros2 topic and I got:
![image](https://github.com/user-attachments/assets/3ad0dd1d-3164-4820-8ea1-d75ba9f8721c)
####Additional info
```
youngricky@youngricky-Hasee:~$ ros2 topic list
/amcl/transition_event
/amcl_pose
/behavior_server/transition_event
/behavior_tree_log
/bond
/bt_navigator/transition_event
/clicked_point
/clock
/cmd_vel
/cmd_vel_nav
/cmd_vel_teleop
/controller_server/transition_event
/diagnostics
/downsampled_costmap
/downsampled_costmap_updates
/global_costmap/costmap
/global_costmap/costmap_raw
/global_costmap/costmap_updates
/global_costmap/footprint
/global_costmap/global_costmap/transition_event
/global_costmap/published_footprint
/global_costmap/voxel_marked_cloud
/goal_pose
/imu
/initialpose
/joint_states
/joy
/local_costmap/clearing_endpoints
/local_costmap/costmap
/local_costmap/costmap_raw
/local_costmap/costmap_updates
/local_costmap/footprint
/local_costmap/local_costmap/transition_event
/local_costmap/published_footprint
/local_costmap/voxel_grid
/local_costmap/voxel_marked_cloud
/local_plan
/map
/map_server/transition_event
/map_updates
/mobile_base/sensors/bumper_pointcloud
/neor_mini/cmd_ackermann
/neor_mini/pose
/odom
/odom/ground_truth
/parameter_events
/particle_cloud
/performance_metrics
/plan
/plan_smoothed
/planner_server/transition_event
/preempt_teleop
/robot_description
/rosout
/scan
/smoother_server/transition_event
/speed_limit
/tf
/tf_static
/trajectories
/transformed_global_plan
/unsmoothed_plan
/velocity_smoother/transition_event
/waypoint_follower/transition_event
/waypoints
```
planner's yaml
```
planner_server:
  ros__parameters:
    planner_plugins: ["GridBased"]
    use_sim_time: True

    GridBased:
      plugin: "nav2_smac_planner/SmacPlannerHybrid" # In Iron and older versions, "/" was used instead of "::"
      debug_visualizations: True         # For Hybrid nodes: Whether to publish expansions on the /expansions topic as an array of poses (the orientation has no meaning) and the path's footprints on the /planned_footprints topic. WARNING: heavy to compute and to display, for debug only as it degrades the performance.
      downsample_costmap: false           # whether or not to downsample the map
      downsampling_factor: 1              # multiplier for the resolution of the costmap layer (e.g. 2 on a 5cm costmap would be 10cm)
      tolerance: 0.25                     # dist-to-goal heuristic cost (distance) for valid tolerance endpoints if exact goal cannot be found.
      allow_unknown: true                 # allow traveling in unknown space
      max_iterations: 100000000           # maximum total iterations to search for before failing (in case unreachable), set to -1 to disable
      max_on_approach_iterations: 1000    # Maximum number of iterations after within tolerances to continue to try to find exact solution
      max_planning_time: 5.0              # max time in s for planner to plan, smooth
      motion_model_for_search: "DUBIN"    # Hybrid-A* Dubin, Redds-Shepp
      angle_quantization_bins: 72        # Number of angle bins for search
      analytic_expansion_ratio: 3.5       # The ratio to attempt analytic expansions during search for final approach.
      analytic_expansion_max_length: 3.0  # For Hybrid/Lattice nodes: The maximum length of the analytic expansion to be considered valid to prevent unsafe shortcutting
      analytic_expansion_max_cost: 200.0  # The maximum single cost for any part of an analytic expansion to contain and be valid, except when necessary on approach to goal
      analytic_expansion_max_cost_override: false  #  Whether or not to override the maximum cost setting if within critical distance to goal (ie probably required)
      minimum_turning_radius: 0.5         # minimum turning radius in m of path / vehicle
      reverse_penalty: 2.0                # Penalty to apply if motion is reversing, must be => 1
      change_penalty: 0.0                 # Penalty to apply if motion is changing directions (L to R), must be >= 0
      non_straight_penalty: 1.2           # Penalty to apply if motion is non-straight, must be => 1
      cost_penalty: 2.0                   # Penalty to apply to higher cost areas when adding into the obstacle map dynamic programming distance expansion heuristic. This drives the robot more towards the center of passages. A value between 1.3 - 3.5 is reasonable.
      retrospective_penalty: 0.015
      lookup_table_size: 20.0             # Size of the dubin/reeds-sheep distance window to cache, in meters.
      cache_obstacle_heuristic: false     # Cache the obstacle map dynamic programming distance expansion heuristic between subsequent replannings of the same goal location. Dramatically speeds up replanning performance (40x) if costmap is largely static.    
      use_quadratic_cost_penalty: False
      downsample_obstacle_heuristic: True
      allow_primitive_interpolation: False
      smooth_path: True                   # If true, does a simple and quick smoothing post-processing to the path

      smoother:
        max_iterations: 1000
        w_smooth: 0.3
        w_data: 0.2
        tolerance: 1.0e-10
        do_refinement: true
        refinement_num: 2
```

SteveMacenski, 2024-12-28:
The parameter guide targets the main branch, but parameter was renamed and some debug visualizations were added last year. Check on the hybrid A* cpp file for the appropriate “debug” param name. Expansions I believe are supported, though footprints of the planned paths I think are in Jazzy and newer only.

YoungRicky7, 2024-12-28:
Thank you for your reply, I'll check it out
 

YoungRicky
***@***.***

YoungRicky7, 2024-12-28:
> The parameter guide targets the main branch, but parameter was renamed and some debug visualizations were added last year. Check on the hybrid A* cpp file for the appropriate “debug” param name in the Humble branch. Expansions I believe are supported in humble, though footprints of the planned paths I think are in Jazzy and newer only.

I checked the source code and did not find the “debug” param or the topic "/expansion" that is published，so I add the relevant source code in Jazzy branch to make it work.

SteveMacenski, 2025-01-02:
Got it - it looks like we added that in Iron (https://github.com/ros-navigation/navigation2/blob/iron/nav2_smac_planner/src/smac_planner_hybrid.cpp#L139). Sorry I couldn't point that to you directly before, I was on the go. To get that batteries-included, you need to use Iron or newer (Iron, Jazzy, main).

but, it looks like you found what you need - sorry for the delay due to the holidays and travel!

closed 2025-01-02 by SteveMacenski