ros-navigation/navigation2 #4826: Robot immediately enters recovery when goal is set and aborts

Flamethr0wer, 2025-01-05:
Hello. I am trying to run Nav2 with my differential drive robot but it is behaving in a very strange manner, apparently never even trying to reach the goal. I do not know the cause but modifying the parameters barely seems to make any difference.

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble binaries
- Version or commit hash:
  - I installed nav2 and nav2_bringup through apt
         apt binaries: ros-humble-navigation2	1.1.17-1jammy.20241128.124843
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
Well this is a real robot, but the setup is very similar to Josh Newans' [articubot_one](https://github.com/joshnewans/articubot_one), as I had been following his tutorials, just with 4 wheels and in open-loop. The Nav2 parameters are as follows:
```
amcl:
  ros__parameters:
    use_sim_time: False
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.8
    beam_skip_threshold: 0.3
    do_beamskip: true
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.1
    update_min_d: 0.05
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    set_initial_pose: true
    initial_pose:
      x: 0.616
      y: -2.717
      z: 0.0    # Typically 0.0 for a 2D plane
      yaw: 0.0  # Replace with your desired initial orientation in radians
    scan_topic: scan

bt_navigator:
  ros__parameters:
    use_sim_time: False
    global_frame: map
    robot_base_frame: base_link
    odom_topic: /odom
    bt_loop_duration: 10
    default_server_timeout: 20
[... 367 lines cut ...]
    open_loop: true 

    wheels_per_side: 2
    wheel_separation_multiplier: 5.0
    # left_wheel_radius_multiplier: 0.1
    # right_wheel_radius_multiplier: 0.1

    odom_frame_id: odom
    # pose_covariance_diagonal: [0.2, 0.2, 1000000.0, 1000000.0, 1000000.0, 0.05]
    # twist_covariance_diagonal: [0.2, 0.2, 1000000.0, 1000000.0, 1000000.0, 0.05]
    enable_odom_tf: true

    # cmd_vel_timeout: x
    # publish_limited_velocity: x
    velocity_rolling_window_size: 10
```

Then just launch robot_launch.py (the main launch file with ros2_control), localization_launch.py and navigation_launch.py from nav2_bringup and set a goal pose.

#### Expected behavior
The robot should navigate to the goal pose and reach the goal, with AMCL correcting odometry miscalculations.

#### Actual behavior
As soon as a goal is set, the robot does some weird, fluctuating spinning (allegedly a spin) and then a backup, before aborting completely and sometimes restarting the nav2 stack. AMCL also does not do much to locate the robot, in the meantime. Here is a video showing this:
https://github.com/user-attachments/assets/8fd7edfe-e7bf-490e-831a-b7549ec9adf0

BCKSELFDRIVEWORLD, 2025-01-05:
I think the CPU might be overloaded. Could you check the CPU usage while you’re experiencing this issue?

Flamethr0wer, 2025-01-05:
@BCKSELFDRIVEWORLD I used htop and it says it does go into 90-100% at specific points, although I am using a Raspberry Pi 4B with 8GB of RAM which I think should be enough...

SteveMacenski, 2025-01-06:
90-100% is alot and means that things will be missing their deadlines. I can't say 100% for sure its only caused by that, but if you're able to test with another computer, you should be able to tell that yourself if you don't see those issues one another more powerful computer. I think it is likely that @BCKSELFDRIVEWORLD's comment is probably correct though. Computers don't behave well when at full utilization, things are being lost or threads starved if at 100%.

Flamethr0wer, 2025-01-06:
@SteveMacenski I don't have another computer I can test this on, but seeing the messages, it's indeed likely the recovery is being executed due to missed deadlines. Although that doesn't really explain why, while the robot in Rviz saw itself as just turning left, the real robot performed erratic back-and-forth spinning instead (that's the reason for so much drift in the LaserScan data). Also, is there any way to increase the deadlines or reduce the load on the CPU enough for it to run correctly?

SteveMacenski, 2025-01-06:
Probably also other compute related issues with messages being dropped to the robot base or other issues related to nondeterministic behavior due to the compute resources being choked. I can't offer much suggestion here other than to use a more powerful computer, look at your most expense processes and select CPU-appropriate alternatives or optimizations given your constrained system, and perhaps try to use optimized DDS parameters / vendor / composition to try to reduce the overhead due to the networking stack.

Flamethr0wer, 2025-01-06:
@SteveMacenski Are there any parameters which could be modified/reduced to adapt Nav2 for a slower computer while retaining enough capabilities? My environment won't be very dynamic, so maybe there are some overly heavy processes I can slow down or change?

SteveMacenski, 2025-01-06:
That's a much more involved question than I can give an easy answer to. It depends on what trade offs you're willing to make and what your robot needs to accomplish. This is more of a system integration question than I can provide generic advice for -- you'd have to analyze your system and performance and determine that balance for yourself :-) 

If you have further questions, I'd suggest Robotics Stack Exchange. I'm closing this issue since the root cause is pretty confidently known as a user integration challenge rather than an issue with Nav2

closed 2025-01-06 by SteveMacenski

Flamethr0wer, 2025-01-06:
Okay, I'll have a look into that and have added a question on SE (https://robotics.stackexchange.com/questions/114197/how-to-decrease-nav2-cpu-usage in case anyone wants to contribute)

millejade, 2025-10-24:
Hi @SteveMacenski!

First, thanks for this excellent package.
I have a similar issue but for me, I'm using 12 cores - each of them only reaches 10%-15% (I'm using htop).

This is the issue that I am encountering:
```
[bt_navigator-20] [WARN] [1761212781.599806920] [BehaviorTreeEngine]: Behavior Tree tick rate 100.00 was exceeded!
[bt_navigator-20] [WARN] [1761212781.609964140] [bt_navigator_navigate_to_pose_rclcpp_node]: Timed out while waiting for action server to acknowledge goal request for compute_path_to_pose
[planner_server-18] [INFO] [1761212781.611218931] [global_costmap.global_costmap]: Message Filter dropping message: frame 'laser' at time 1761212781.287 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[bt_navigator-20] [WARN] [1761212781.620306620] [BehaviorTreeEngine]: Behavior Tree tick rate 100.00 was exceeded!
[bt_navigator-20] [WARN] [1761212781.630433247] [bt_navigator_navigate_to_pose_rclcpp_node]: Node timed out while executing service call to global_costmap/clear_entirely_global_costmap.
[controller_server-16] [INFO] [1761212781.639955209] [controller_server]: Goal was canceled. Stopping the robot.
[controller_server-16] [WARN] [1761212781.639999031] [controller_server]: [follow_path] [ActionServer] Client requested to cancel the goal. Cancelling.
[bt_navigator-20] [ERROR] [1761212781.640371342] [bt_navigator]: Goal failed
[bt_navigator-20] [WARN] [1761212781.640395800] [bt_navigator]: [navigate_to_pose] [ActionServer] Aborting handle.
[planner_server-18] [INFO] [1761212781.660266169] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[planner_server-18] [INFO] [1761212781.960402726] [global_costmap.global_costmap]: Message Filter dropping message: frame 'laser' at time 1761212781.587 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[planner_server-18] [INFO] [1761212781.987280006] [global_costmap.global_costmap]: Message Filter dropping message: frame 'laser' at time 1761212781.687 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[planner_server-18] [WARN] [1761212782.332538684] [planner_server]: Planner loop missed its desired rate of 20.0000 Hz. Current loop rate is 1.4878 Hz
```

I used the default values of Nav2 params - except the `robot_radius`, which should reflect the robot, and `inflation_radius` and `cost_scaling_factor`.
A similar issue occurred where the robot stops moving, rotating - I guess it does some recovery - when in fact the global plan is cleared from obstacles and the robot is also out of obstacles (I can visualize it using the local and global costmaps and well-versed on using `inflation_radius` and `cost_scaling_factor`.

What actually happens is:
- I set the goal, and the global plan seems okay.
- Sometimes, the robot starts rotating even if it is supposed to move forward (I'm not sure if it is doing recovery, but I encountered it inconsistently).
-  While navigating, most of the time, it moves continuously, then suddenly, it will stop and do some rotation (which is kinda weird for as based on costmaps, both global and local, the robot is almost clear from obstacles. It will get stuck at that state until I set a new goal.

Some checks that I did:
- `odom` -> `base_footprint` is being published by the `robot_localization` package - when using Mapping (SLAM Toolbox), it seems it doesn't have much drift (I'm using industry-grade IMU and Motion Controller).
- I was also able to map properly.

Some assumptions:
- I'm using third-party LiDAR, and it already has a ROS2 driver, but the scan rate is up to 10Hz only. Not sure if this will affect but the `/scan` topic is pretty responsive during Mapping or when I only run this node.
- AMCL: I'm not sure if there are delays on my RViz since I'm using LAN through wifi to connect my machines, but the main point is, there would be times that `map` -> `odom` is completely off. (This is when I am using Localization and Navigation)
- I noticed some great improvement when I changed the planner from DWB to Regulated Pure Pursuit. In DWB, it also stops and rotates, but the movement is like spaghetti. In Regulated PurePursuit, the movement is more reasonable with respect to the Global Planner, which is NavFn.

I am searching for the same issue, that's why I commented here - but let me know if you need further information and if I need to raise a new issue.

Current setup:
- ROS Humble
- Installed Nav2 last Jan/Feb 2024 (and never updated it)
- RMW Implementation: Default to FastDDS.
- Nav2 Params: Similar to default, with some adjustments mentioned above.

Thank you very much in advance.

millejade, 2025-12-07:
Hi!

I was able to have a workaround. It actually works fine on a very controlled environment (like there's no other wifi network that can interfere).

It also works if all nodes are running on the same machine (which is located on the robot).

But since my setup requires multiple machines connected to LAN, I need an external machine for RViz2 to send goals. I encountered that issue in an environment where there's a lot of interference.

Flamethr0wer, 2025-12-07:
@millejade So you were able to solve the immediate recovery when setting a goal? What's the workaround?

millejade, 2025-12-07:
The default parameters works okay!

I think the problem is the network I am using to connect Multiple Machines.

I already discussed the work around, please read it carefully. (Though I'm not sure if that would also apply to your issue)