ros-navigation/navigation2 #4827: MPPI controller plugin build assumes AVX2 and FMA instructions are available

rickarmstrong, 2025-01-06:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu24.04.1 LTS
- ROS2 Version:
  -  Jazzy binaries installed with `sudo apt install ros-jazzy-navigation2`
- Version or commit hash:
  - ros-jazzy-navigation2	1.3.4-1noble.20241228.015128
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

```
docker run -it --net=host --privileged --volume="${XAUTHORITY}:/root/.Xauthority" --env="DISPLAY=$DISPLAY" -v="/tmp/.gazebo/:/root/.gazebo/" -v /tmp/.X11-unix:/tmp/.X11-unix:rw --shm-size=1000mb osrf/ros:jazzy-desktop-full
apt update
apt upgrade
apt install ros-jazzy-navigation2 -y
apt install ros-jazzy-nav2-bringup -y
apt install ros-jazzy-nav2-minimal-tb* -y
source /opt/ros/<ros2-distro>/setup.bash
ros2 launch nav2_bringup tb4_simulation_launch.py headless:=False
```

#### Expected behavior
A running Turtlebot 4 simulation running Nav 2 in Gazebo.
#### Actual behavior
A running Gazebo sim with a Turtlebot 4 in a warehouse, but none of the Nav2 stuff running and RViz is frozen.
#### Additional information
I'm running on a pretty old Intel i5-based computer.
More details and an answer here: https://robotics.stackexchange.com/questions/114131/trouble-getting-nav2-getting-started-example-to-run
TL;DR: running under `gdb` reveals that `controller_server` croaks with 
```
Thread 1 "controller_serv" received signal SIGILL, Illegal instruction.
0x00007fffef45da59 in mppi::critics::ConstraintCritic::initialize() ()
   from /opt/overlay_ws/install/nav2_mppi_controller/lib/libmppi_critics.so
```
Workaround is to build from source, remove compilation flags that enable AVX2 and FMA, or switch to a different controller plugin.

rickarmstrong, 2025-01-06:
BTW, I don't know that there's anything to do here. I think the right thing is to add a caveat to the docs; I'm happy to do that if it's desired.

SteveMacenski, 2025-01-06:
Sure thing on the docs. Any modern processor should have AVX2 so its not functionally been much of an issue (even low power ARM). 

By the way, there’s an Eigen based version in #4621  that you should try out that replaces xtensor and explicit AVX flags. Does that work fine for you (and likely be quite a speed up)?

rickarmstrong, 2025-01-07:
I'll give the Eigen-based fork a spin tonight.

rickarmstrong, 2025-01-07:
I was unable to build the `eigen_mppi` branch of Ayush1285's fork . I see that `nav2_mppi_controller/CMakeLists.txt` is doing `add_compile_options(-mfma)` in that branch so it won't run on my crusty old computer anyhow, with out turning that off. 
In the near future, I'll be running the nav on a different, less-crusty machine and I imagine the Eigen-based plugin will be merged and I'll give it a go.

SteveMacenski, 2025-01-07:
Ah, I think your SOL on MPPI then, we need some vectorization operations to make scoring 60+ length trajectories with 10+ critics with 2000 batches at 30 hz (36M major operations/sec) :laughing: I think all somewhat modern machines have AVX2 and MFMA, so this hasn't been an issue we've run into before. Honestly speaking, if your machine is so old it doesn't have these, it probably can't handle MPPI anyway even if it did have AVX2/MFMA available due to the sheer compute requirements needed for MPC. 

MFMA I looked up has been around since 2013, I think we can safely call this is an edgecase that I don't think requires explicit documentation, but I will make a note in the README shortly.

SteveMacenski, 2025-01-07:
https://github.com/ros-navigation/navigation2/blob/main/nav2_mppi_controller/README.md#mfma-and-avx2-optimizations

closed 2025-01-07 by SteveMacenski

rickarmstrong, 2025-01-07:
Hey, at least now I have a second legit reason to upgrade my dev box (beyond "runs Starfield") :)

SteveMacenski, 2025-01-07:
P.S.

I'm looking to have a new desktop box for development soon, I've been looking at these 2, which I used for a project with AMD and I was shocked at the power for the cost, and still at a form factor & power level that could be used on an AMR. It cut down compiling time from my new Intel-based i7 thinkpad by 40%.   
- https://store.minisforum.com/products/minisforum-um790-pro?variant=43865372786933&currency=USD&utm_medium=product_sync&utm_source=google&utm_content=sag_organic&utm_campaign=sag_organic&gad_source=1
- https://store.minisforum.com/products/minisforum-um890pro?_pos=8&_fid=70ebfb945&_ss=c&variant=45607692796149