ros-navigation/navigation2 #4833: Navigation working on simulation but not on real robot

apo1234545, 2025-01-08:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 24.04 
- ROS2 Version:
  - Jazzy
- Version or commit hash:
  - c8e8b23

#### Issue
I am using NAV2 Framework on ROS2 Jazzy. I installed the default version using the installation instruction on  the website. Using the parameters i have below, everything works perfect on the Gazebo simulation on my laptop (Lenovo ThinkPad 16GB RAM), but the same parameters (with fewer particles for AMCL) seems to produce error on the real robot (Raspberry Pi 5 8GB).

I use NavFn A* Planner and MPPI Controller on a differential drive robot

```

amcl:
  ros__parameters:
    alpha1: 0.5
    alpha2: 0.5
    alpha3: 0.5
    alpha4: 0.5
    #alpha5: 0.2 #for Omni models only
    base_frame_id: "base_footprint_link"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 1.75
    laser_max_range: 8.0
    laser_min_range: 0.5
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2500
    min_particles: 1000
    odom_frame_id: "odom"
    pf_err: 0.01
    pf_z: 0.99
    recovery_alpha_fast: 0.1
    recovery_alpha_slow: 0.001
    resample_interval: 1
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.3
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.1
    update_min_d: 0.1
    z_hit: 0.8
    z_max: 0.05
    z_rand: 0.35
    z_short: 0.05
    scan_topic: scan
    set_initial_pose: false
    initial_pose:
      x: 0.0
      y: 0.0
      z: 0.0
      yaw: 0.0

bt_navigator:
  ros__parameters:
    global_frame: map
    robot_base_frame: base_link
    odom_topic: /odom
    bt_loop_duration: 10
[... 419 lines cut ...]
A few times is working and gets to the goal.
Errors:
```
[bt_navigator-5] [WARN] [1736250730.138830672] [bt_navigator_navigate_to_pose_rclcpp_node]: Timed out while waiting for action server to acknowledge goal request for compute_path_to_pose
[controller_server-1] [INFO] [1736250730.172082709] [controller_server]: Cancellation was successful. Stopping the robot.
[controller_server-1] [INFO] [1736250730.172181505] [controller_server]: [follow_path] [ActionServer] Client requested to cancel the goal. Cancelling.
[bt_navigator-5] [ERROR] [1736250730.192228227] [bt_navigator_navigate_to_pose_rclcpp_node]: Failed to get result for follow_path in node halt!
[controller_server-1] [INFO] [1736250730.192322116] [controller_server]: Optimizer reset
[bt_navigator-5] [WARN] [1736250730.198756764] [bt_navigator]: [navigate_to_pose] [ActionServer] Aborting handle.
[bt_navigator-5] [ERROR] [1736250730.198915042] [bt_navigator]: Goal failed
```

```
[bt_navigator-5] [WARN] [1736250958.744983480] [bt_navigator]: [navigate_to_pose] [ActionServer] Aborting handle.
[bt_navigator-5] [ERROR] [1736250958.745188406] [bt_navigator]: Goal failed
[controller_server-1] [WARN] [1736250958.759738628] [controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is 8.4485 Hz.
[controller_server-1] [INFO] [1736250958.759827998] [controller_server]: Cancellation was successful. Stopping the robot.
[controller_server-1] [INFO] [1736250958.759847110] [controller_server]: [follow_path] [ActionServer] Client requested to cancel the goal. Cancelling.
[controller_server-1] [INFO] [1736250958.779499165] [controller_server]: Optimizer reset
```
#### Additional information
Robot control and SLAM works okay. AMCL works okay too. The problems starts when i try to navigate to a pose.

Any feedback would be highly appreciated.
Thank you!!

SteveMacenski, 2025-01-08:
We don't generally offer user setup help (since you make the point that it works fine for you on another machine) -- you may need to tune the system to work well on the RPi given its compute constraints. I'd ask some motivating questions though:
- What's your CPU/memory usage? If its spiking above 90-100%, then you're likely dropping messages and threads are being starved. In that case, its no wonder you're seeing errors like this
- Are you using composition nodes? For something like this, you want to squeeze out performance as much as possible
- I haven't tried to test MPPI on an RPi, but I'm guessing it is using up all of your CPU. Unless you need to the power of MPPI, I'd recommend checking out one of the other lower overhead algorithms. [This paper](https://arxiv.org/pdf/2307.15236) should provide you some details. Have you tried to profile compute use and see where you hot spot are?
- It seems like you might be asking too much of your local costmap given your computer. Reduce that update rate. 
-

apo1234545, 2025-01-10:
> We don't generally offer user setup help (since you make the point that it works fine for you on another machine) -- you may need to tune the system to work well on the RPi given its compute [...]

Thank you. You help was very helful.
I lowered the number of AMCL particles, lowered local costmap frequency, lowered controller frequency (while increasing model_dt), lowered batch_size and works better.
I am still having some issues though.
I will continue working on it next week and i will provide the updated parameters as reference for anyone insterested in using NAV2 on Raspberry Pi 5.

SteveMacenski, 2025-01-10:
Keeping it under around ~85% would be a good benchmark. ~80% is usually where I like it.

Sounds like you've got a handle on it. I'd still recommend maybe not using MPPI and instead RPP if you're that resource constrained!

closed 2025-01-10 by SteveMacenski