ros-navigation/navigation2 #4835: use_composition causing failure to launch

Flamethr0wer, 2025-01-08:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble binaries
- Version or commit hash:
  - apt binaries: ros-humble-navigation2	1.1.17-1jammy.20241128.124843
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
Simply run `ros2 launch nav2_bringup navigation_launch.py use_composition:=True`

#### Expected behavior
Nav2 should launch using the node composition.

#### Actual behavior
The launching process gets stuck at
```
[INFO] [launch]: All log files can be found below /home/flamethrower/.ros/log/2025-01-08-19-04-40-981949-robot-5858
[INFO] [launch]: Default logging verbosity is set to INFO
```
with -d:
```
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[INFO] [launch]: All log files can be found below /home/flamethrower/.ros/log/2025-01-08-19-09-10-832912-robot-5877
[INFO] [launch]: Default logging verbosity is set to DEBUG
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0xffffae463160>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0xffffae463160>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0xffffae462c20>'
Executing <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py:228> result=None created at /opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py:318> took 1.834 seconds
```

SteveMacenski, 2025-01-08:
Did you actually launch a component container? You're bypassing [`bringup_launch.py` which does this](https://github.com/ros-navigation/navigation2/blob/main/nav2_bringup/launch/bringup_launch.py#L129-L138) by launching `navigation_launch.py` directly. That is what creates the container that then Navigation and Localization (and others potentially in your application) can then load their nodes into. 

Running `navigation_launch.py` on the commandline is not expected to work by itself for these reasons. I'd recommend checking out our[ Getting Started Guide](https://docs.nav2.org/getting_started/index.html).

closed 2025-01-08 by SteveMacenski

Flamethr0wer, 2025-01-09:
Oh my bad thank you I added a container and now it works