ros-navigation/navigation2 #4837: Feature Request: Service Introspection

ottojo, 2025-01-09:
## Feature request
#### Feature description
It would be useful to be able to use [Service Introspection](https://docs.ros.org/en/jazzy/Tutorials/Demos/Service-Introspection.html) introduced in Jazzy with nav2 nodes (such as the costmap server). This requires some configuration for both the client and server, which is currently not enabled in nav2.

#### Implementation considerations
It has to be considered whether unconditionally enabling service introspection is a good idea. The tutorial in the ROS docs show how to introduce a parameter to enable or disable service introspection, it is unclear to me if it's expected to only turn this feature on during debugging, and if there are downsides to always enabling it.

SteveMacenski, 2025-01-09:
The client side is easy, I think most all uses of service clients use our `nav2_utils` service client wrapper, so one spot should hit most if not all of them (but worth doing a quick grep to make sure of that). 

The server side will be slightly more involved since I don't think we wrap anything there. I'm not sure if the right answer is to:
- create a `nav2_util` service server wrapper like the client
- create a `nav2_util` helper object that deals with these mechanics, getting the parameters, and setting it up that given a `node` and the server
- Better yet, contribute back to `rclcpp` by modifying `create_service` to handle this logic so that anyone creating a server can have this handled implicitly when a standardized parameter is set. I think you could do that with `create_client` too :wink:  
- Something else?

How do you think we should approach it? 

I'd say leave it default off, but let users flip it on with a parameter.

SteveMacenski, 2025-01-23:
@ottojo what do you think?

mini-1235, 2025-01-25:
I can work on this if no one is on it :)

SteveMacenski, 2025-01-27:
@mini-1235 please!

mini-1235, 2025-02-03:
> I'd say leave it default off, but let users flip it on with a parameter.
Where do you think we should put the parameter to flip this on, I haven't found a suitable place in a `nav2_params.yaml`, also, do you think it is necessary to support dynamic configuration of this parameter

closed 2025-04-02 by SteveMacenski