ros-navigation/navigation2 #4842: condition distance traveled always returns failure

guidout, 2025-01-10:
Since when the behavior tree is first loaded, all nodes are in the IDLE state before they are ticked for the first time, doesn't this if statement always return IDLE and therefore the tick method always return FAILURE?

https://github.com/ros-navigation/navigation2/blob/507b3659e6c622d9339b61b53810320b600526a2/nav2_behavior_tree/plugins/condition/distance_traveled_condition.cpp#L46

I also asked the question here but didn't get answers:
https://robotics.stackexchange.com/questions/114059/nav2-distance-traveled-condition

SteveMacenski, 2025-01-10:
Please do not cross post.

Have you actually tried to use it and see that it doesn't work or just speculating looking at the software? We have a unit test that seems to point to it working [1]

[1] https://github.com/ros-navigation/navigation2/blob/main/nav2_behavior_tree/test/plugins/condition/test_distance_traveled.cpp

gritelli, 2025-01-14:
Yes, I did test it and the condition always returns IDLE.
This is the BT I used for the test:
```
<root main_tree_to_execute="MainTree">
  <BehaviorTree ID="MainTree">

    <RateController hz="1.0">
      <ReactiveFallback name="CheckDistanceTravelled">
        <DistanceTraveled distance="2.0" global_frame="map" robot_base_frame="base_link"
        />

        <PipelineSequence>
          <GoalUpdatedController>
            <RecoveryNode number_of_retries="1" name="RecoveryComputePathToPose">
              <ComputePathToPose goal="{goal}" path="{path}"
                planner_id="GridBased_SmacHybrid_production_line" />
              <ClearEntireCostmap name="ClearGlobalCostmap-Context"
                service_name="global_costmap/clear_entirely_global_costmap" />
            </RecoveryNode>
          </GoalUpdatedController>
          <RecoveryNode number_of_retries="1" name="RecoveryFollowPath">
            <FollowPath path="{path}" controller_id="FollowPath_TEB_prod_line"
              goal_checker_id="general_goal_checker" />
            <ClearEntireCostmap name="ClearLocalCostmap-Context"
              service_name="local_costmap/clear_entirely_local_costmap" />
          </RecoveryNode>
        </PipelineSequence>
      </ReactiveFallback>
    </RateController>
  </BehaviorTree>
</root>
```
I added a print out in the `distance_traveled_condition.cpp`:
```
BT::NodeStatus DistanceTraveledCondition::tick()
{
    std::cout << "Distance Traveled Condition Status: " << status()
              << std::endl;

    if (status() == BT::NodeStatus::IDLE)
...
```

It always prints IDLE.

Maybe my behavior tree does not implement this condition correctly?

guidout, 2025-01-14:
Yes, I did test it and the condition always returns IDLE.
This is the BT I used for the test:
```
<root main_tree_to_execute="MainTree">
  <BehaviorTree ID="MainTree">

    <RateController hz="1.0">
      <ReactiveFallback name="CheckDistanceTravelled">
        <DistanceTraveled distance="2.0" global_frame="map" robot_base_frame="base_link"
        />

        <PipelineSequence>
          <GoalUpdatedController>
            <RecoveryNode number_of_retries="1" name="RecoveryComputePathToPose">
              <ComputePathToPose goal="{goal}" path="{path}"
                planner_id="GridBased_SmacHybrid_production_line" />
              <ClearEntireCostmap name="ClearGlobalCostmap-Context"
                service_name="global_costmap/clear_entirely_global_costmap" />
            </RecoveryNode>
          </GoalUpdatedController>
          <RecoveryNode number_of_retries="1" name="RecoveryFollowPath">
            <FollowPath path="{path}" controller_id="FollowPath_TEB_prod_line"
              goal_checker_id="general_goal_checker" />
            <ClearEntireCostmap name="ClearLocalCostmap-Context"
              service_name="local_costmap/clear_entirely_local_costmap" />
          </RecoveryNode>
        </PipelineSequence>
      </ReactiveFallback>
    </RateController>
  </BehaviorTree>
</root>
```
I added a print out in the `distance_traveled_condition.cpp`:
```
BT::NodeStatus DistanceTraveledCondition::tick()
{
    std::cout << "Distance Traveled Condition Status: " << status()
              << std::endl;

    if (status() == BT::NodeStatus::IDLE)
...
```

It always prints IDLE.

Maybe my behavior tree does not implement this condition correctly?

SteveMacenski, 2025-01-14:
Can you check the test and see an error? Unfortunately, we're a bit resource constrained right now, so I do need developers to do a bit of research to help suss out their issues, if they exist. I don't have the bandwidth right now to dig into each reported issue from scratch :-) 

I see that the [main version of this node](https://github.com/ros-navigation/navigation2/blob/main/nav2_behavior_tree/plugins/condition/distance_traveled_condition.cpp) looks different from the link that you provided based on some hash. What branch is that from and have you tested any changes as addressing the behavior (and/or looked at `main` for potential updates)? It could be that your branch is no longer supported

guidout, 2025-01-14:
That hash should be the latest commit from the Humble branch.
I'll do some digging and report back. 
I understand your resource constraints, totally. Thanks for replying, I'll report back as I find out more.

p0rys, 2025-01-30:
@guidout Is there any news yet? I've stumbled across the same problem and would like to avoid digging through the BehaviorTreeCPP code if someone else has already done it. 

From my observation, the problem seems to be that the condition always returns SUCCESS or FAILURE and then starts again with IDLE on the next call since it was already “completed”.

The UnitTest seems to work, as it reacts slightly differently to the library. It calls two ticks in succession without switching to the IDLE state in between

guidout, 2025-02-03:
@p0rys you are exactly right. When the `resetChildren` gets called this condition (like any other) will reset to IDLE therefore the tick method of this plugin never returns anything different than IDLE and then the pose is reset causing the distance to always be zero and the condition never satisfied. 

My fix was to move the `nav2_util::getCurrentPose` in the constructor and wrap in a while loop (I'll have to add some sort of timeout so at some point I do get an error if the transform is not available). Then I just comment out the whole `if (status() == BT::NodeStatus::IDLE)` statement.

I haven't had the time to do a deep dive into BehaviorTreeCPP so I think there could be a more elegant solution.

I also think many conditions might suffer from a similar issue.

shivaang12, 2025-02-08:
@guidout @p0rys It would be really helpful if you could write a test case that clearly exposes this issue, as the current unit test isn’t detecting it.

adivardi, 2025-02-11:
I was meaning to open an issue for this, but I think it is related.

Currently, any node using `if (!BT::isStatusActive(status()))` to initialize a variable/flag cannot be used under `ReactiveSequence` alongside another long-running task.
As @p0rys mentioned, if any other child returns Running or Failure, the sequence halts all children, setting them to IDLE. On the next tick, they re-initialize.
This applies to `rateController`, `DistanceController`, `DistanceTraveledCondition`, `GoalUpdatedCondition` and a bunch more.

In a way, it sort of correct, as `ReactiveSequence` should not have 2 long nodes inside it, and all these "compare current state to a previous state" nodes are long-running in a way.
It is however quite obscure and non-intuitive, and limits the nodes usefulness (since `ReactiveSequence` is such a basic node).

Detecting when a node should reset is tricky. Without resetting, they may retain values from a previous task (see https://github.com/ros-navigation/navigation2/pull/4744 and https://github.com/ros-navigation/navigation2/pull/3988#issuecomment-2443930344).

There is one solution suggestion from Steve [in this old message](https://github.com/ros-navigation/navigation2/pull/2591#issuecomment-976009401) to add `RunID` variable. Another approach could be re-initializing all nodes when the BT navigator receives a new task or switches navigators.

SteveMacenski, 2025-02-11:
>  if any other child returns Running or Failure, the sequence halts all children, setting them to IDLE. On the next tick, they re-initialize.

I believe that is only on `Failure` and that is expected behavior. That's why there's a `GloballyUpdatedGoalCondition` option analog to the `GoalUpdatedCondition` to retain some state globally. This is not a bug :-)

adivardi, 2025-02-12:
In a `Sequence`, it is only on failure. I agree that it is not a bug and with everything you wrote.

In a `ReactiveSequence` and a `ReactiveFallback` it is also when another child returns `Running`  This is also not really a bug, but it is non-intuitive that a `DistanceTraveledCondition` is a long-running node and that any node decorated by for example a DistanceController becomes a long running node and cannot be with another long running node.

I believe this is the issue in this ticket.
Looking at the tree the author posted, every time the `PipelineSequence` returns running (for example, if `FollowPath` is working well and returns running, the `DistanceTraveled` will be halted and reset to IDLE. So in the next tick iterations `if (!BT::isStatusActive(status()))` is True -> will return Failure. And so on every time `PipelineSequence` returns Running.

I had the same issue with a RateContrller inside a ReactiveSequence last week 😅 
This is btw also true for all the other nodes that check `!BT::isStatusActive(status())`. But most of them just re-read the inputs so it's not really a problem.

References:
Reactive Sequence: https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/src/controls/reactive_sequence.cpp#L51
Reactive Fallback: https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/src/controls/reactive_fallback.cpp#L51
Both lead to either `resetChildren()` or `haltChild` from    https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/src/control_node.cpp#L38

SteveMacenski, 2025-02-13:
> In BehaviorTreeCPP, a sequence (in my case ReactiveSequence) calls the method “resetChildren” after receiving “FAILURE” from a child, which in turn resets the status of all children to IDLE. This is why the DistanceTraveled and TimeExpired conditions always return Failure as they only ever execute the “initialization block” and never actually check the respective condition.

This is expected in my opinion. These are typically used to wrap long-running operations like living in the main navigation pipeline. For example [1] nearly all of our BT nodes use the `PipelineSequence` to wrap the navigation task of planning->control->recovery, in various formats and styles. Thus the child `RateController`, `DistanceController`, etc is not reset because the `PipelineSequence` is active and these BT nodes are consistently a member of the primary running branch. That is, until a system-level recovery is performed in the `Sequence` on L30, whereas it goes out of scope and is reset. 

This seems like good, desirable behavior for this application that they were developed for. With that said, `PipelineSequence` is a Nav2-specific node and perhaps we should use something more standard. I was around the team that developed these Nav2-specific BT Control nodes, though I wasn't involved in that part as I was busy with porting other parts of the system. Its a real point of conversation if we want to keep those, move to something else, or if the behavior needs to be adjusted due to these customize nodes. Things 'work' now, but doesn't mean things can't be _better_.

[1] https://github.com/ros-navigation/navigation2/blob/main/nav2_bt_navigator/behavior_trees/navigate_to_pose_w_replanning_and_recovery.xml#L13

> That's why there's a GloballyUpdatedGoalCondition option analog to the GoalUpdatedCondition to retain some state globally. This is not a bug :-)

Maybe having some globally-kept version of those BT nodes would be of value? Or, what suggestions do you propose 😄

adivardi, 2025-02-20:
In general I agree - it also sort of make sense that a `rate_controller` and similar are a long-running task (since they refer to a historical state). It is too bad that they cannot be used for something like reactive conditions easily, but I also don't see an easy way to solve that.

At the minimum, I think it would be good to add a note to the `@brief` of these classes, so it is clear and not a surprise. I can make such PR if you think it makes sense.

Regarding a more "real" fix, I think the issue with the `Global` versions is that they don't reset between tasks. Correct me if I am wrong, but I think this is why the `!BT::isStatusActive(status())` cases were added.
I still like your suggestion of adding a `runID` variable and resetting the `globalCondition` when the change. I think that could be the cleanest way to reset them, and could even be implemented in the same node (add an additional `global` flag that enable/disable the local resets).
Another approach could be to reset all nodes from the navigator when a new task is requested, but I can't think of a clean way to do that.

SteveMacenski, 2025-03-10:
> At the minimum, I think it would be good to add a note to the @brief of these classes, so it is clear and not a surprise. I can make such PR if you think it makes sense.

Please (sorry for the delay; vacation + illness) do! 

>  I think the issue with the Global versions is that they don't reset between tasks

This is true, except when `always_reload_bt_xml` is set to true so that each navigation request uses a fresh, reloaded tree instead of reusing the previous tree when the trees are the same XML file. If we don't want to reset on `halt()` when a node's scope is finished, then we also don't know when a new tree request is given. But, there is a way around it via that reload parameter.

> I still like your suggestion of adding a runID variable and resetting the globalCondition when the change

That would work too! I would aruge that this could be a contribution back to BT.CPP itself to insert a special `runID` blackboard variable as part of the BT factory. [Or add it here](https://github.com/ros-navigation/navigation2/blob/e3c41beb08cd1939d8eed378768d2744bf47904c/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_server_impl.hpp#L247-L289) when we create a new tree or reuse the old tree. Then, in the BT nodes that are "global", we reset when that changes. That would need some documentation so that folks are aware about how that all works. 

What do you think? I think there are 2 distinct things: one is documenting the current nodes about how they are intended to be used and work. Then another for adding in potential global nodes using this runID concept.

adivardi, 2025-03-14:
The sounds like a good plan.
I will try to work on documenting the current nodes soon.
For the other change, I am also interested to work on that, just not sure on availability in the near future.

By the way, I was not aware of `always_reload_bt_xml`, that is really useful!
I noticed that in the original PR you recommended to not using it resetting nodes between tasks. Any reason you don't like this option?

SteveMacenski, 2025-03-14:
It just takes some extra time to reconstruct the tree, so there's latency in the initial starting of execution. This is especially true if there are ROS interfaces (srv, msg, action) to remote machines in which need to be initialized that might take a couple of seconds to setup if talking to cloud servers -- just for example. 

More a preference for expediency!

SteveMacenski, 2025-03-24:
@adivardi any update on the docs piece?

adivardi, 2025-04-01:
I opened a PR now #5035 (sorry for the delay, I was on vacation)

Also opened an issue for the `RunID` feature: #5036

SteveMacenski, 2025-04-01:
Great - closing as completed! Will follow up in that ticket / PR