ros-navigation/navigation2 #4844: nav2_mppi_controller source code build error

YoungRicky7, 2025-01-13:
- Operating System:
  ubuntu 22.04
- ROS2 Version:
  humble 

#### Steps to reproduce issue
 When I used colcon build to build the source code of mppi, I gor error:
`youngricky@youngricky-Hasee:~/graduation_project_ws$ colcon build
Starting >>> nav2_smac_planner
Starting >>> ackermancar_description
Starting >>> ackermancar_navigation2
Starting >>> nav2_mppi_controller
Finished <<< ackermancar_description [0.09s]                                                                                                                                                       
Finished <<< ackermancar_navigation2 [0.13s]
Finished <<< nav2_smac_planner [0.25s]                                                                         
Starting >>> planner_playground
Finished <<< planner_playground [0.18s]                                                                      
--- stderr: nav2_mppi_controller                             
In file included from /usr/include/xtensor/xaccumulator.hpp:18:0,
                 from /usr/include/xtensor/xmath.hpp:26,
                 from /usr/include/xtensor/xcontainer.hpp:25,
                 from /usr/include/xtensor/xtensor.hpp:20,
                 from /home/youngricky/graduation_project_ws/src/nav2_mppi_controller/include/nav2_mppi_controller/tools/noise_generator.hpp:24,
                 from /home/youngricky/graduation_project_ws/src/nav2_mppi_controller/src/noise_generator.cpp:15:
/usr/include/xtensor/xexpression.hpp: In instantiation of ‘struct xt::detail::is_crtp_base_of_impl<xt::xexpression, xt::has_simd_interface<xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 2, (xt::layout_type)1, xt::xtensor_expression_tag>, float> >’:
/usr/include/xtl/xtype_traits.hpp:323:12:   required from ‘struct xtl::disjunction<xt::detail::is_crtp_base_of_impl<xt::xexpression, xt::has_simd_interface<xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 2, (xt::layout_type)1, xt::xtensor_expression_tag>, float> >, xt::detail::is_crtp_base_of_impl<xt::xexpression, xt::has_simd_interface<xt::xfunction<xt::detail::plus, const xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 1, (xt::layout_type)1, xt::xtensor_expression_tag>&, const xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 2, (xt::layout_type)1, xt::xtensor_expression_tag>&>, float> > >’
/usr/include/xtensor/xoperation.hpp:212:90:   required by substitution of ‘template<class F, class ... E> using xfunction_type_t = typename std::enable_if<xtl::disjunction<xt::is_xexpression<typename std::decay<_Args>::type>...>::value, xt::detail::xfunction_type<F, E ...> >::type::type [with F = xt::detail::logical_and; E = {xt::has_simd_interface<xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 2, (xt::layout_type)1, xt::xtensor_expression_tag>, float>, xt::has_simd_interface<xt::xfunction<xt::detail::plus, const xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 1, (xt::layout_type)1, xt::xtensor_expression_tag>&, const xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 2, (xt::layout_type)1, xt::xtensor_expression_tag>&>, float>}]’
/usr/include/xtensor/xoperation.hpp:375:17:   required by substitution of ‘template<class E1, class E2> xt::detail::xfunction_type_t<xt::detail::logical_and, E1, E2> xt::operator&&(E1&&, E2&&) [with E1 = xt::has_simd_interface<xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 2, (xt::layout_type)1, xt::xtensor_expression_tag>, float>; E2 = xt::has_simd_interface<xt::xfunction<xt::detail::plus, const xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 1, (xt::layout_type)1, xt::xtensor_expression_tag>&, const xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 2, (xt::layout_type)1, xt::xtensor_expression_tag>&>, float>]’
/usr/include/xtensor/xassign.hpp:345:104:   required from ‘static constexpr bool xt::xassign_traits<E1, E2>::simd_interface() [with E1 = xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 2, (xt::layout_type)1, xt::xtensor_expression_tag>; E2 = xt::xfunction<xt::detail::plus, const xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 1, (xt::layout_type)1, xt::xtensor_expression_tag>&, const xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 2, (xt::layout_type)1, xt::xtensor_expression_tag>&>]’
/usr/include/xtensor/xassign.hpp:353:106:   required from ‘static constexpr bool xt::xassign_traits<E1, E2>::simd_assign() [with E1 = xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 2, (xt::layout_type)1, xt::xtensor_expression_tag>; E2 = xt::xfunction<xt::detail::plus, const xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 1, (xt::layout_type)1, xt::xtensor_expression_tag>&, const xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 2, (xt::layout_type)1, xt::xtensor_expression_tag>&>]’
/usr/include/xtensor/xassign.hpp:378:57:   [ skipping 4 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]
/usr/include/xtl/xmeta_utils.hpp:603:29:   required from ‘decltype(auto) xtl::mpl::static_if(const TF&, const FF&) [with bool cond = false; TF = xt::assign_xexpression(xt::xexpression<D>&, const xt::xexpression<E2>&) [with E1 = xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 2, (xt::layout_type)1, xt::xtensor_expression_tag>; E2 = xt::xfunction<xt::detail::plus, const xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 1, (xt::layout_type)1, xt::xtensor_expression_tag>&, const xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 2, (xt::layout_type)1, xt::xtensor_expression_tag>&>]::<lambda(auto:64)>; FF = xt::assign_xexpression(xt::xexpression<D>&, const xt::xexpression<E2>&) [with E1 = xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 2, (xt::layout_type)1, xt::xtensor_expression_tag>; E2 = xt::xfunction<xt::detail::plus, const xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 1, (xt::layout_type)1, xt::xtensor_expression_tag>&, const xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 2, (xt::layout_type)1, xt::xtensor_expression_tag>&>]::<lambda(auto:65)>]’
/usr/include/xtensor/xassign.hpp:198:58:   required from ‘void xt::assign_xexpression(xt::xexpression<D>&, const xt::xexpression<E2>&) [with E1 = xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 2, (xt::layout_type)1, xt::xtensor_expression_tag>; E2 = xt::xfunction<xt::detail::plus, const xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 1, (xt::layout_type)1, xt::xtensor_expression_tag>&, const xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 2, (xt::layout_type)1, xt::xtensor_expression_tag>&>]’
/usr/include/xtensor/xsemantic.hpp:623:31:   required from ‘xt::xcontainer_semantic<D>::derived_type& xt::xcontainer_semantic<D>::assign_xexpression(const xt::xexpression<E>&) [with E = xt::xfunction<xt::detail::plus, const xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 1, (xt::layout_type)1, xt::xtensor_expression_tag>&, const xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 2, (xt::layout_type)1, xt::xtensor_expression_tag>&>; D = xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 2, (xt::layout_type)1, xt::xtensor_expression_tag>; xt::xcontainer_semantic<D>::derived_type = xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 2, (xt::layout_type)1, xt::xtensor_expression_tag>]’
/usr/include/xtensor/xsemantic.hpp:489:57:   required from ‘xt::xsemantic_base<D>::derived_type& xt::xsemantic_base<D>::assign(const xt::xexpression<E>&) [with E = xt::xfunction<xt::detail::plus, const xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 1, (xt::layout_type)1, xt::xtensor_expression_tag>&, const xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 2, (xt::layout_type)1, xt::xtensor_expression_tag>&>; D = xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 2, (xt::layout_type)1, xt::xtens [... cut ...] sor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 1, (xt::layout_type)1, xt::xtensor_expression_tag>&, int>]’
/home/youngricky/graduation_project_ws/src/nav2_mppi_controller/src/optimizer.cpp:208:38:   required from here
/usr/include/xtensor/xexpression.hpp:168:88: internal compiler error: 段错误
         xtl::disjunction< std::is_base_of<B<E>, F<E>>, std::is_base_of<B<F<E>>, F<E>>> {};
                                                                                        ^
`

SteveMacenski, 2025-01-13:
Do you have xtensor, xsimd, etc installed properly? How did you install them? What hardware are you running on, and does that have the capability of running avx2/mfma optimizations (basically anything modern since 2013, minus some raspberry pis)

This seems to be working in CI, my docker images, and in our nightly builds, so I don't think the issue here is likely the software.

YoungRicky7, 2025-01-14:
it was the version of g++ that installed in my PC, it was too old, I updated it to g++-11 and no errors showed up

closed 2025-01-14 by YoungRicky7