ros-navigation/navigation2 #4845: Setting minimum_turning_radius to 0 for SmacPlannerHybrid causes error

tonynajjar, 2025-01-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 24
- ROS2 Version:
  - Jazzy

#### Steps to reproduce issue

```
planner_server:
  ros__parameters:
    expected_planner_frequency: 20.0
    planner_plugins: ["GridBased"]
    GridBased:
      plugin: "nav2_smac_planner::SmacPlannerHybrid"
      tolerance: 0.2
      downsample_costmap: false
      downsampling_factor: 1
      allow_unknown: true
      max_iterations: 1000000
      max_on_approach_iterations: 1000 
      max_planning_time: 1.5 
      motion_model_for_search: "REEDS_SHEPP" 
      angle_quantization_bins: 64 
      analytic_expansion_ratio: 3.5
      minimum_turning_radius: 0.1
      reverse_penalty: 10.0 
      change_penalty: 0.15 
      non_straight_penalty: 1.50
      cost_penalty: 0.1
      lookup_table_size: 10.0
      cache_obstacle_heuristic: False
      debug_visualizations: false
      smoother:
        max_iterations: 1000
        w_smooth: 0.3
        w_data: 0.2
        tolerance: 1.0e-10

```

#### Expected behavior

#### Actual behavior

I get GridBased plugin failed to plan from (10.38, 0.41) to (10.35, -3.65): "Start occupied".

#### Additional information

----

SteveMacenski, 2025-01-13:
That is not a valid configuration, there must be a non-zero value for the minimum turning radius.

SteveMacenski, 2025-01-13:
I just updated the docs to mention that as well

SteveMacenski, 2025-01-23:
I think this can be closed?

closed 2025-01-27 by SteveMacenski

SteveMacenski, 2025-08-27:
> I get paths that rotates-in place towards the goal, goes straight until the goal, rotates in place to match goal orientation

If you want such rigid behavior, why not set that in the behavior tree. RotateToHeading to the relative goal heading, plan a path as a straight line, then rotate to final heading with the same Spin node. 

I assume you want to use this in more than trivial straight line applications though, so:

> rotates in place to match goal orientation

In Smac you can use the any goal orientations new feature to do this so that it plans to any orientation to the goal pose and then the controller can kick in to rotate to the specific heading required. See the `goal_heading_mode` parameter. That'll get you the end of the path behavior you want.

The start of path is a little trickier. The rotate in place / costmap cost penalties I assume will do some of the lifting there. It sounds more like you want an infeasible planner like NavFn and have a 'rotate to heading' spin to start tracking. Or possibly plan twice. Once to get the relative heading you want to travel (NavFn if doesn't need to be feasible, or Smac if it does), rotate to that heading, then plan again from that heading to get the more straight line approach. Conveniently, since you compute the first plan and then spin, you have the entire time while you're spinning to compute the next path (since you know the position and the newly-found orientation) so you shouldn't need to delay execution at all. 

We could also make spinning in place at the start a "Free" operation (if I don't already? that sounds familiar to me) by parameter so that the Lattice can rotate in place for the first plan freely so you can find the optimal path. The only problem I see with that is that you really only want that for the first plan while stopped. Once you're moving, you don't want that since you'd constantly be stopping and rotating and continuing for minor changes. But, this can be something in the behavior tree as a blackboard variable pretty easily I suspect.