ros-navigation/navigation2 #4847: SmootherServer's interface smoothPlan is not called while running nav2

YoungRicky7, 2025-01-14:
**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - humble

I was running a simulation with gazebo and nav2,  trying to get experment data of the smoother(time cost length etc.)

#### Expected behavior
the code that I added should be printed in the terminator and there should  be a smoothed path that being published in the "/plan_smoothed" topic
[image]
#### Actual behavior
It's not printed as expected and nothing is in the topic
#### Additional information
everything else is working just fine(planner,controller)
[image]
the goal is already set and the robot is running,but no path showed in the /plan_smoothed topic:
[image]
the smoother is also already activated:
[image]

nav2_yaml:
`amcl:
  ros__parameters:
    use_sim_time: True
    alpha1: 0.0
    alpha2: 0.0
    alpha3: 0.0
    alpha4: 0.0
    alpha5: 0.0
    base_frame_id: "base_link"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 12.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan

bt_navigator:
  ros__parameters:
    use_sim_time: True
    global_frame: map
    robot_base_frame: base_link
    odom_topic: /odom
[... 353 lines cut ...]
    robot_base_frame: base_link
    transform_tolerance: 0.1
    use_sim_time: true
    simulate_ahead_time: 2.0
    max_rotational_vel: 1.0
    min_rotational_vel: 0.4
    rotational_acc_lim: 3.2

robot_state_publisher:
  ros__parameters:
    use_sim_time: True

waypoint_follower:
  ros__parameters:
    
    use_sim_time: True
    loop_rate: 20
    stop_on_failure: false
    waypoint_task_executor_plugin: "wait_at_waypoint"
    wait_at_waypoint:
      plugin: "nav2_waypoint_follower::WaitAtWaypoint"
      enabled: True
      waypoint_pause_duration: 20

`

SteveMacenski, 2025-01-14:
How are you calling the smoother server? I don't see a BT XML or a script that is calling it. If you need some help in that, check out our tutorial I wrote on it: https://docs.nav2.org/tutorials/docs/adding_smoother.html

YoungRicky7, 2025-01-15:
> How are you calling the smoother server? I don't see a BT XML or a script that is calling it. If you need some help in that, check out our tutorial I wrote on it: https://docs.nav2.org/tutorials/docs/adding_smoother.html

Thank you! I'll take a look at it, and one dumb question, how do I find the nav2 docs of the version that I use?(for me it's humble),cuz it seems that nav2.org's docs is for the latest version and I happened to ran into some errors when I was following the tutorials.

SteveMacenski, 2025-01-15:
We only currently support docs for the current version -- though I can't find any reference to `nav2_ceres_costaware_smoother`, I don't believe it was ever called that in a released distribution (and definitely not humble, I'm looking at humble's source branch). That tutorial is just showing an example (but maybe I can change it to a *real* algorithm as an example -- changing that now).

YoungRicky7, 2025-01-15:
> We only currently support docs for the current version -- though I can't find any reference to `nav2_ceres_costaware_smoother`, I don't believe it was ever called that in a released distribution (and definitely not humble, I'm looking at humble's source branch). That tutorial is just showing an example (but maybe I can change it to a _real_ algorithm as an example -- changing that now).
> 
> I'd recommend using `nav2_smoother::SavitzkyGolaySmoother ` for testing.

Thank you for your reply, I'll give it a try.

closed 2025-01-15 by YoungRicky7