ros-navigation/navigation2 #4865: [Graceful Controller] Additional Controller Techniques - Experimentation

SteveMacenski, 2025-01-18:
Right now the Graceful Controller computes a smooth spiral curve from the control law from the robot's current state to a pose on the path to track. The path pose is based on a lookahead distance that is shortened up if there is a collision in the trajectory's spiral curve on the way. Getting out some ideas / thoughts so we can collaborate on this as a community instead of staying in some notes document 

There are quite a few other ways, I think that this control law could be utilized that are worth experimenting with for improvements or different behaviors:
- [ ] Interpolate the path point for the distance requested by the lookahead, to smooth out the trajectories from frame to frame. We do this in the RPP controller, so this should be a relatively straight forward improvement that improves the controller as-is
- [ ] Consider committing to a given trajectory for a period of time before computing a new spiral. Each iteration, we compute a new spiral and perhaps its worth committing to the current spiral for a couple of seconds (except if collision or large change in cost) so we commit to actions more fully, before then computing a new curve
- [ ] Consider sampling a set of spiral trajectories with different spiral parameters instead of them as static parameters. Sample within some range of values and score them for the best trajectory
- [ ] Consider the above, with relaxation on the exact goal pose so we compute (collision valid) spirals towards or along the path, but no so exact. Further, we could sample based on some time duration into the future from the robot's starting pose + spiral parameters. Part of the scoring could be how close / near the path or it is (so not even considering the path's poses in the spiral constraints) 
- [ ] Perhaps rather than randomly sampling (like RRT) or sampling in fixed intervals (like DWA) of the the spiral parameters, attempt to use convex optimization on some cost functions to select the best trajectory and goal [Unconstrained Model Predictive Control for Robot Navigation under
Uncertainty](https://assets.amazon.science/61/6a/49cbec1749e9a41410593433295d/unconstrained-model-predictive-control-for-robot-navigation-under-uncertainty.pdf) [Robot Navigation with MPEPC in Dynamic and Uncertain
Environments: From Theory to Practice](https://d1wqtxts1xzle7.cloudfront.net/109264024/Park-iros-workshop-12-libre.pdf?1703022077=&response-content-disposition=inline%3B+filename%3DRobot_Navigation_with_MPEPC_in_Dynamic_a.pdf&Expires=1737166463&Signature=cMPtBnMp8UFxhypQHG3-15g17Ob0f4lXeHF6Vw5j0lZ9MHgbtgscH4lejMuY2KhpWMKsFYpQIvfZg-r~-O4aI1rzZ5kzJww9FcochA0D9JNIW2FA8H4ZPo33obguBmFaodEuFEURScGY9VgUXjqS4GEVDZTeXSJ1JVv8Oe3DMDn42CVXOOJsbuPvYpWCBi6P16Dnu9THaNYdhVcKcKo6o6QwR2gklhcoO8Sh~hs4xgRSkGwwwgCzgVs2pAdYol1fgF3h7kDm7z7BQ5YPkX109aBCvi5h6FSC64~1gipfwPhGTdzgWr7NXlQKlVN9kxm-zGVkZzX6H5U~oeMkkFhkIA__&Key-Pair-Id=APKAJLOHF5GGSLRBV4ZA) [Robot Navigation with Model Predictive Equilibrium Point Control](https://citeseerx.ist.psu.edu/document?repid=rep1&type=pdf&doi=121486810a5a5dc5b06befcddf0ef46463a767e3) [dissertation](https://deepblue.lib.umich.edu/bitstream/handle/2027.42/120760/jongjinp_1.pdf?sequence=1)

Some things we can score: 
* Path smoothness
* Collision
* How well it follows the path // gets close to the desired intermediate goal pose on the path
* Adjust scoring weighting based on distance from obstacle (follow goal better when far from objects; avoid objects better when close to them)

---

- [ ] Consider the aforementioned, but in multiple timesteps. If we sample / optimize for a trajectory that is some time in the future, then we can do so again at that time to compute a new trajectory to hand it off to. For example: we compute trajectories that are 1s into the future via sampling or optimization. Then we do so again for another 1s and so forth to compute more sophisticated trajectories. This is somewhat similar to MPPI, but more coarsely and the trajectories themselves are known to be smooth and feasibly connect to each other. This would be quite computational and not sure its an improvement over MPPI, so perhaps this is better a thought experiment to pull ideas from for simpler modifications. I think Graceful as a simple, smooth trajectory planner is where we should keep it -- rather than being highly reactive or dealing with multiple time-step issues, where MPC/MPPI shine best.  https://www.cs.cmu.edu/~iwan/papers/vfhstar.pdf
- [ ]

mikeferguson, 2025-01-18:
A couple notes on the above ideas:
  
 * I've actually never tried "interpolating path points" - but my gut says that would probably work pretty well
 * I'm not sure about the "committing to a spiral" - my sense is that isn't causing any real issues (especially if you add some interpolation stuff). This is especially true if you have the continuous replanning turned on (since your whole plan disappears and gets replaced).
 * For sampling different spirals, you might only need to sample across `k1` parameter - to basically change the curvature (maybe also `k2`, but I feel like they are inter-related enough you might only have to vary one to get a pretty good set of spirals). That said, I think any sort of exhaustive search will get computationally nasty and negate some of the use cases for a lightweight controller.

Some other notes that I had sitting around:

 * Biggest bang for our buck would probably be to get rid of the `slowdown radius` and do the forward simulation using proper acceleration limits. This same thing could be done with docking - where it might actually be a very big improvement (especially if people know their acceleration limits well, they can be certain the robot won't hit the dock real hard)
 * I have a note in the previous PR that `The collision checking in the iterative loop assumes you are starting at max speed (not at your current speed). This is probably not that big of a deal, but if you're going very slowly in a tight area, the controller might wobble a bit more`. I think this is actually easier to do with the nav2 controller API than what existing in nav1 where we had to separately subscribe to odom and whatnot.

A few remaining features that exist in the older graceful controller, but not the nav2:

 * Footprint expansion - as you go faster, your footprint is enlarged to account for the uncertainty in how well your controls will actually work. This is especially helpful if you have an environment which is quite varied in how open space it is (like, when you want to go 1.5m/s in wide open hallways but automatically slow down when entering a cubicle farm that is quite a bit less open). This can of course be achieved through speed limit maps, but that's a lot more manual to setup per installation.

VArt3m, 2025-01-20:
Great ideas above, good to see improvements to Graceful controller which is my favorite so far.
Tacking on some possible improvements:

* Not just interpolating but extrapolating (extending) original paths near endpoint (goal). As far as I understand when the robot approaches the goal, the lookahead distance effectively shrinks because there are no further points. This leads to change of behavior of the controller (in my experience, for the worse) - it becomes much eager to rotate so if there's any CTE it will stop at a very awkward angle near the goal. In some cases final rotation in place can fix this, though this is undesirable in the first place, but some robots have Ackermann kinematics. I did improve this behavior in my project by adding a continuous path pre-processor extending path past original goalpoint (coupled with my own goal checker).
* Somewhat similar kind of pre-processing with cusps (forward-backward path reverse points) - in my project I prune the path beyond nearest cusp and extend trajectory like outlined above. When the cusp point is reached, the path beyond current cusp is fed to the controller. This helps to follow intended trajectory more precisely, allows for gradual slowdowns, and prevents the robot from "shortcutting" the trajectory - very good for three-point turns and Reeds-Shepp trajectories.
* Pruning the path to limit path direction change relative to the current point in path. I haven't tried this _yet_, but if we have - for example - a path with 180° turn with radius significantly smaller than lookahead then the robot will likely make a significant "shortcut" because the controller will find the next lookahead point father than the turn is. Pruning the path to the point that deviates for example no more than 90° or 60° in its orientation than current closest point on path would help to follow such turns more exactly.

mikeferguson, 2025-01-20:
> Not just interpolating but extrapolating (extending) original paths near endpoint (goal). As far as I understand when the robot approaches the goal, the lookahead distance effectively shrinks because there are no further points.

We actually do this in the docking stuff.. but for some reason I've never actually experimented with doing it for general navigation.

SteveMacenski, 2025-01-22:
> I'm not sure about the "committing to a spiral" - my sense is that isn't causing any real issues

From my extremely superficial testing of the controller, this seemed to cause some uncommitted action/wobbling, but also I was testing when the orientations weren't being computed correctly, so this may be moot now. I could easily also see with path point interpolation this being discarded too. Its worth toying with, but I agree with you, it could unnecessary / not particularly useful. 

 > For sampling different spirals, you might only need to sample across k1 parameter - to basically change the curvature 

This is a great insight, thanks!

> I think any sort of exhaustive search will get computationally nasty and negate some of the use cases for a lightweight controller.

I don't think this has to be only a lightweight controller, I think we can offer a heavier-weight option that leverages the smooth control law to do more sampling or optimization-based solutions. Kind of like the Smac Planner sharing a common-A*, the Graceful Controller package could be a set of Controllers based on the shared spiral smooth control law. The current controller would always remain, but have some other additional options. 

> Get rid of the slowdown radius
> Footprint expansion 

Agreed and added both! 

> The collision checking in the iterative loop assumes you are starting at max speed (not at your current speed). This is probably not that big of a deal, but if you're going very slowly in a tight [...]

Is the action item to use the current velocity instead? Yeah, that would be very easy, I think we even provide that Twist to the controller when it computes its velocity. That's a `good-first-issue` kind of contribution too (I'm looking for more of those right now too).

SteveMacenski, 2025-01-22:
> Not just interpolating but extrapolating (extending) original paths near endpoint (goal). As far as I understand when the robot approaches the goal, the lookahead distance effectively shrinks [...]

@VArt3m see the RPP controller, we actually do that already for these exact same reasons. I'd love if you contributed that here!

> Somewhat similar kind of pre-processing with cusps (forward-backward path reverse points) - in my project I prune the path beyond nearest cusp and extend trajectory like outlined above. When the [...]

@VArt3m see the RPP controller, we actually do that already for these exact same reasons. I'd love if you contributed that here! (wow, copy paste 😆 )

> Pruning the path to limit path direction change relative to the current point in path. I haven't tried this yet, but if we have - for example - a path with 180° turn with radius significantly [...]

This is worth some testing. I would counter by saying that this controller really doesn't follow paths very precisely since it creates its own curve of its own design based on its own selection of a lookahead point on the path. I'm not sure this is the best controller algorithm if you want accurate path tracking in turns. However, if the behavior of that 180 turn is _really bad_, then I agree we should make some special arrangements. We could look at the orientations of the path points and use those as part of the calculation of the distance along the path (not just X,Y, but X,Y,Yaw distances with some yaw scale). I think that would probably work smashingly and also reduce the lookahead when doing normal turns to be more conservative as well. Either way, we'd make this a parameterized option so folks could turn it off/on as they saw fit. 

@VArt3m open to some contributions? These sound like they're all pretty analog to other things we have, so that should make the implementation and review much more straight forward.

SakshayMahna, 2025-02-17:
Hi, is this still under progress?
Can I work on this?

SteveMacenski, 2025-03-07:
Currently no one is working on this, but a few people have expressed interest recently! If you'd like to work on part of this, I'd love the help and just mention here what you plan to do and a timeline so that others can know what is taken and when it could be expected for them to build upon!

SakshayMahna, 2025-03-08:
Sure,
I'll start with this:

> * [ ]  Interpolate the path point for the distance requested by the lookahead, to smooth out the trajectories from frame to frame. We do this in the RPP controller, so this should be a relatively straight forward improvement that improves the controller as-is. Other RPP-based bells-and-whistles improvements?

I can provide some updates on this by next week.

SakshayMahna, 2025-03-12:
The current implementation of the Graceful Controller involves finding a point on the given trajectory that is within a look ahead threshold. It uses that point to construct a local trajectory and follows through it.

In my implementation, I tried to interpolate a point that falls within a lookahead distance on the given trajectory. And as well extrapolate a point not on the trajectory when it is about to reach the goal (Just like it is done in RPP). However, there is one slight thing to take care, Graceful Controller makes use of the position and orientation of the robot (RPP only uses position), therefore we need to interpolate/extrapolate an orientation as well. For now I took it as the angle from the interpolated point to the point that is closest to interpolated point on the trajectory.

Following are the results:

Current Implementation:
https://github.com/user-attachments/assets/6e1a501e-9456-4b16-8e79-4c4456d5a412

Proposed Implementation:
https://github.com/user-attachments/assets/62dadc00-eb49-4cae-b7ea-dda80409db7a

I had two questions:
- What exact improvements should we be expecting from this?
- Any ideas how we can exactly monitor/observe them?

SteveMacenski, 2025-03-12:
Can you highlight the key differences you'd like us to note in the videos representing changes you made (just for clarity's sake)? I see that it appears to be smoother frame-to-frame when computing the spiral curves - which is very nice indeed! Does that come mostly from the interpolation? 

 > What exact improvements should we be expecting from this?

I think that's up to some testing and engineering analysis to decide. If this is in response to my first improvement bullet in this ticket, I think this should create much smoother trajectories from frame-to-frame since there is much less jumping due to discretization. This was an important step for RPP for high speed vehicles, but given how sensitive Graceful is to changes in position and orientation, I would think it would be doubly as important here. 

In the first video, you showed it short cutting the corner path - do you still see that? The second video didn't show a similar turn.

> Any ideas how we can exactly monitor/observe them?

This visualization of the trajectories and the robot's actual motion thereby should be evident enough for this task, I think. It should be clear that (a) the trajectories look much more smooth and continuous w.r.t. the previous iterations rather than jumping around. (b) Additionally, if you ignore the trajectories and plot only the robot base frame, I would hope that you could see the robot's behavior itself is much more smooth as a result (RQT plot with the X/Y/Theta pose might also help there).

Let me know when you're ready for a code review! I'd love to get this in :-)

SakshayMahna, 2025-03-15:
I conducted a few more tests, and have some observations.

1. Visually I don't see much difference between the performance. Both the controllers (current and new) seem to behave almost the same.
2. The difference can be observed in the plots of linear and angular velocities. Especially the linear velocity seems quite smoother because of this change. Plots shown below:
3. For the end path interpolation (extrapolation), it seems the original Graceful Controller implementation is better than the changed one. Visually I observe that in the original implementation, the robot slows down when it is about to reach the end of the trajectory. But because of the extrapolation I don't see that happening for the new implementation.

Linear Velocity Plots:

**Original Implementation**
[image]

**New Implementation**
[image]

*Note that the waypoints/trajectory provided in the start and towards the end are same. (In between the goal poses provided are different).*

However, the jitters in the original implementation velocity are quite evident as compared to the new implementation.

Angular Velocity Plots:

**Original Implementation**
[image]

**New Implementation**
[image]

*Note that the waypoints/trajectory provided in the start and towards the end are same. (In between the goal poses provided are different).*

Nothing major to focus here, but overall the angular velocity for new implementation seems slightly smoother compared to the current implementation.

If all seems fine with this improvement I can start with the PR and code review then.

SteveMacenski, 2025-03-21:
> The difference can be observed in the plots of linear and angular velocities. Especially the linear velocity seems quite smoother because of this change

Oh very nice, that is not only smoother (especially in angular jerk), but we spend more time moving at our maximum velocity, meaning we're also executing tasks faster!

> For the end path interpolation (extrapolation), it seems the original Graceful Controller implementation is better than the changed one. Visually I observe that in the original implementation, the [...]

I think it would be good to open 2 PRs. One that is the interpolation while tracking the path and another that does the end-of-path interpolation. We can have some folks review the end of path interpolation one separately that I think would be interested in that work and testing it out for their applications and wouldn't block the path-tracking part of this. We could also spot some errors about why its not slowing down to help resolve as well so it could be included :-) 

Please open the PRs! I can mark this off on the TODO list for the graceful controller then 😄 I think the next point on committing to a given spiral would be the next good experiment to test out. I'm less certain that this will be a meaningful improvement (or an improvement), but I think it is worthwhile to investigate!

SakshayMahna, 2025-03-22:
I have opened a PR. I kept the path interpolation of goal as a parameter. It can be turned on and off through that (similar to RPP).

Meanwhile through the review, I'll start working on the commiting to given spiral.

SakshayMahna, 2025-03-29:
Here my observations on committing to spiral.

https://github.com/user-attachments/assets/451e8bba-5935-4e3d-8269-4534f799f06c
This video shows committing to 5 timesteps for the spiral. Overall, the robot seems a little paced in it's approach towards the target. But sometimes (slightly captured at the start of the video) the robot goes in a zig zag pattern because it has committed to a certain path, thereby taking some time to return to the original planned path.

I also recorded certain graphs:

## Linear Velocity Plots

### Committing 1 timestep
[image]

### Committing 5 timestep
[image]

### Committing 10 timestep
[image]

## Angular Velocity Plots

### Committing 1 timestep
[image]

### Committing 5 timestep
[image]

### Committing 10 timestep
[image]

Overall the plots look smoother, but the zig zag behavior seems problematic to me.

Let me know what you think.

Also side note, the PR #5003 passes all the checks now. I'll make any required changes once reviewed.

SakshayMahna, 2025-04-05:
In the current implementation, when we are close to the goal, we calculate the minimum of the calculated velocity (according to control law) and a ratio of maximum velocity based on `distance_left / slowdown radius`.

The aim is to get this velocity (when close to goal) using acceleration.

In my implementation, I calculated the velocity based on maximum deceleration (provided by user) and the distance left: `sqrt(2 * distance_left * max_deceleration)`, and took the minimum of this value with calculated velocity (according to control law).

Correct me if I am wrong or there is a problem in my understanding / implementation.

However, here are the results for the same:

### Acceleration Limit Video
https://github.com/user-attachments/assets/1f7ac2ea-7538-4e49-a8ab-4413d8c2358b

### Slowdown Radius Video
https://github.com/user-attachments/assets/6e1a501e-9456-4b16-8e79-4c4456d5a412

### Acceleration Limit Velocity Plot
[image]

### Slowdown Radius Velocity Plot
[image]

In the slowdown radius implementation there exists a small window when robot is moving at a constant velocity and then velocity starts to decrease. Compared with acceleration limit implementation, the robot directly starts to reduce it's velocity.

Visibly, for slowdown radius implementation, we can visibily see the robot slow down when near goal. For acceleration limit implementation, the robot comes to a stop in a flow.

SteveMacenski, 2025-04-08:
> Correct me if I am wrong or there is a problem in my understanding / implementation.

This is correct! Can I see the implementations? It could depend on how you compute the distance to use (path integral distance? L2 distance?).

It is possible that the accelerations set are high enough that it effectively doesn't need to slow down until very close instead of the larger radius? Since this is only used at the end of the path, this shouldn't be tuned for the max allowable acceleration profile for normal navigation, it should be more constrained for this stopping behavior.

SakshayMahna, 2025-04-11:
Here is the implementation (Line 70):
https://github.com/SakshayMahna/navigation2/blob/1c8ea91a041e88ce816c4a8a2becb4c320db780f/nav2_graceful_controller/src/smooth_control_law.cpp#L70

I used the earlier present `EgocentricPolarCoordinates` implementation. What I understand it gives the direct L2 distance from current pose to target pose. I can try with path integral distance, that may make a difference.

From the overall code, this `calculateRegularVelocity` function is called at each iteration of the simulation, with a different target pose. However, when the robot is about to reach it's destination, the target pose should keep constant, and the stopping behavior should start when this starts to happen.

I see that it would require some additional implementation in the case when we do path interpolation (1st enhancement). But, it should a give general idea of the behavior.

What do you think? Please do correct me if I am wrong in this.

SteveMacenski, 2025-04-15:
What is the `0.05` for? I think that's the acceleration term which is very small (?). I think that should be parameterized and use accelerations similar to other nodes (2.0-4.0 m/s^2 range)

The `EgocentricPolarCoordinates` does provide the L2 distance in the [pose constructor](https://github.com/ros-navigation/navigation2/blob/20b6fe58c83e5d260855a412b331d45ddf5923bd/nav2_graceful_controller/include/nav2_graceful_controller/ego_polar_coords.hpp#L54-L69) that is correct. 

> with a different target pose.

The target pose is constant, only the `next_pose` changes based on the simulation for collision checking. The target pose is the goal pose / carrot.

SakshayMahna, 2025-08-17:
Ramping up the conversation again. The implementation looks like this:
```cpp
double v = v_linear_max_ / (1.0 + beta_ * std::pow(fabs(curvature), lambda_));

double acceleration = 3.0;
v = std::min(sqrt(2 * dist_to_target * acceleration), v);

v = std::clamp(v, v_linear_min_, v_linear_max_);
```

I tried two implementations, one with distance to target as L2 distance (calculated using `EgocentricPolarCoordinates`) and another path distance, carried over from graceful controller object.

Here are the graphs for almost same goal poses:

**General**

[image]

**Magnified**

[image]

Visually, as comparing slowdown radius and acceleration limit implementation, the robot seems to stop gradually in case of slowdown radius, and spontaneously for acceleration limit implementation.

SteveMacenski, 2025-08-26:
Great! Please open the PR :- )

mikeferguson, 2025-12-09:
Footprint scaling is mostly helpful when robots have accuracy issues at higher velocities - this can be caused by either poor response times/dynamics OR issues such as sliding from loss of traction. The idea in the graceful controller (and a number of the ROS 1 navigation controllers which also implemented this feature) is that when you are going faster, you want to leave a large a margin of error around the robot, so you increase the footprint during collision checking.

With a control law like the graceful controller, we also want to simulate the longest possible path because it tends to lead to smoother overall motions - so we actually desire to simulate the longer path, but with a slower velocity (and less scaled footprint), rather than relying on the velocity scaling that comes from short lookahead distance or tight curvature (which is really more intended for slowing down to the final goal point, or slowing down when going around tight curves).

The computation is really quite lightweight, so if your robot needs/wants more space at higher speeds, it is totally worth it. Also, since the loop is greedy and a single collision will restart the loop with a lower velocity, you tend to not actually do that many extra collision checks.

Tacha-S, 2025-12-09:
I completely agree that simulating the longest possible trajectory tends to lead to smoother motions, and therefore it is desirable to use as long a path as possible.
I also agree that, since the loop is greedy, increasing the number of iterations does not significantly increase the number of collision checks.
As everyone has mentioned, when control accuracy degrades at higher speeds, increasing the footprint margin according to velocity to assess the possibility of collisions when the robot slightly deviates from the intended path makes perfect sense.

However, stepping one level deeper, I believe there is an additional issue:
as shown in the image I shared above, is it really desirable for the robot to take a shortcut path that barely avoids the inscribed circle radius, as long as it simply slows down?
If the inflation layer were not applied, I could understand why such trajectories would occur.
But when an inflation layer is applied and the global planner produces a path that is already inflated to a certain extent, I believe the local planner should respect that to some degree.
After all, rather than moving extremely slowly along a path that passes very close to obstacles, it is sometimes smoother to follow a slightly more inflated trajectory at a moderate speed.

This is why I suggested taking the local cost conditions into account, inspired by RPP.
Coexistence with footprint scaling is perfectly fine if that turns out to be beneficial, and if there are other cost-aware approaches that could provide smoother motion than RPP, I would be happy to consider those as well.

Tacha-S, 2025-12-16:
I have created a pull request for the footprint expansion.
Please review it when you have time.
https://github.com/ros-navigation/navigation2/pull/5779

Tacha-S, 2026-01-13:
Yes, it has been resolved.
Sorry for leaving the question unanswered.

The deceleration near obstacles by expanding the footprint was working well, so I believe I implemented it as suggested by @mikeferguson .

As for the shortcut occurring despite tuning `k_phi` and `k_delta`, the cause was that near the goal, `prefer_final_rotation` forcibly changed the robot’s orientation to the straight-line direction from the current position to the goal.
To address this, similar to how `initial_rotation` has `initial_rotation_tolerance`, I added a `final_rotation_tolerance` to `prefer_final_rotation` so that unintended shortcuts can be reduced.

ajtudela, 2026-10-10:
Hi,

I'd like to keep working on improving the graceful controller. I've been catching up with all the changes from the last months (lookahead interpolation, deceleration limit, footprint scaling, the new approach search...) and it's looking really good!

While reviewing the code I noticed that `min_lookahead` isn't used anymore. It was used in `validateTargetPose` to discard targets too close to the robot, but the check disappeared in #5947 when that logic was moved to `validateTargetPoseOnApproach`. The parameter is still declared and documented, so I'm not sure if it was intentional or it just got lost in the refactor.

@Tacha-S @SteveMacenski was this on purpose? If not, I can open a PR to add it back (with a test) and backport it to Lyrical.

SteveMacenski, 2026-10-12:
That does appear to me to have been dropped in error, from looking at it right now. @Tacha-S I'd love to get your thoughts to make sure @ajtudela and I didn't miss something subtle on a re-read, but I think yes you should open a PR to readd @ajtudela :-) 

Edit:

A second #5947 bug I found while reading. `validateTargetPoseOnApproach` takes target_pose by reference and overwrites its heading with `atan2(y, x)`. When that attempt fails, the or condition calls `validateTargetPose` with the already-mutated pose. So, The fallback never tries the path’s real heading; it just repeats the shortcut attempt. With allow_backward and a target behind the robot, the first call flips the heading by PI and the fallback flips it by PI again. The second simulation therefore runs with an un-reversed heading while reversing = true. The fix is to pass a copy into the approach check and only write it back to target_pose on success

ajtudela, 2026-10-13:
Thanks @SteveMacenski, I'll open the PR for `min_lookahead` and I'll fix the second bug you found as well (passing a copy to the approach check and writing it back only on success).

While writing the tests I found two more problems, both reproduced in simulation with default parameters (0.05 m costmap, 0.4 m square footprint):

1. `findBestApproachTrajectory` doesn't select what it is meant to:
   - The ETA is effectively the number of simulation steps, so every candidate gets the same ETA (the simulations all stop at max_iter), and the angle that ends up selected is effectively arbitrary by rounding errors.
   - `best_eta` is updated before the safety check, so an unsafe candidate can block the safe ones.
   - `safe_approach_angle_` is rewritten during the sweep while it is also the base of the angle, so some angles are tested twice and others never (only 42 distinct angles out of 63 evaluations).
2. `simulateTrajectory` stops at `max_iter` and still returns success, so the last 5–15 cm before the target are never collision checked. In a test, a target at 0.5 m is accepted with a wall at 0.4 m, because the simulation is cut at 0.35 m.

[image]

The fixes are small and each one has its own test, so I can pack the four bugs in a single PR.

SteveMacenski, 2026-10-13:
Maybe open a few smaller PRs? These are somewhat subtle changes I'd like to understand in the code and review in isolation. It'll make this go much faster, I promise