ros-navigation/navigation2 #4872: -mtune=generic on RISC‑V is not recognized

ahmadSum1, 2025-01-23:
## Bug Report

**Required Info:**

- **Operating System:**
  - Ubuntu 24.04 (VisionFive 2)
- **Computer:**
  - VisionFive 2 (RISC-V architecture)
- **ROS2 Version:**
  - Jazzy, installed from source
- **Version or commit hash:**
  - `87aae5dc0f078c44ed17f252252b8f629bbcc7e4`
- **DDS implementation:**
  - `rmw_zenoh_cpp`

---

### Steps to reproduce issue

1. Clone the `navigation2` repository:
   ```bash
   mkdir -p src
   rosinstall_generator --rosdistro jazzy --deps navigation2 --exclude RPP | vcs import src
   ```
2. Navigate to the workspace, source ros2 and attempt to build the `nav2_mppi_controller` package:
   ```bash
   colcon build --packages-up-to nav2_mppi_controller --cmake-args -DCMAKE_BUILD_TYPE=Release
   ```
3. Observe the following errors related to `mtune`, `xtensor` and `xsimd`:
   - ```bash
     --- stderr: nav2_mppi_controller
     cc1plus: error: unknown cpu ‘generic’ for ‘-mtune’
     gmake[2]: *** [gtest/CMakeFiles/gtest_main.dir/build.make:76: gtest/CMakeFiles/gtest_main.dir/src/gtest_main.cc.o] Error 1
     gmake[1]: *** [CMakeFiles/Makefile2:611: gtest/CMakeFiles/gtest_main.dir/all] Error 2
     gmake[1]: *** Waiting for unfinished jobs....
     cc1plus: error: unknown cpu ‘generic’ for ‘-mtune’
     ```
   - `aligned_mode in namespace xsimd does not name a type`
   - `xt::xfunction operator<<` issues with output formatting.

> tried with both `riscv64-linux-gnu-gcc-11` and `riscv64-linux-gnu-gcc-13`
---

### Expected Behavior

The package should build successfully without requiring manual intervention to disable SIMD optimizations.

---

### Actual Behavior

The build fails due to compatibility issues with SIMD and `xsimd` on the RISC-V platform.

---

### fix that worked

To resolve the issue, I applied a workaround:
- Disabled `XSIMD` in `xtensor` by adding `-DXTENSOR_DISABLE_XSIMD` and `-DXSIMD_NO_SUPPORTED_ARCH=1` to the `CMakeLists.txt`.

Patch of my changes is attached:
```diff
From 7b39af406bd8a6462e8de95b6c4a92de2f0a1966 Mon Sep 17 00:00:00 2001
From: Sakib Ahmed <sakib.bracueee@gmail.com>
Date: Thu, 23 Jan 2025 19:04:55 +0000
Subject: [PATCH] riscv64-visionfive2 patch

---
 CMakeLists.txt | 64 ++++++++++++++++++++------------------------------
 1 file changed, 25 insertions(+), 39 deletions(-)

diff --git a/CMakeLists.txt b/CMakeLists.txt
index 95eabf6c4..cc9ac79ee 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -1,24 +1,21 @@
 cmake_minimum_required(VERSION 3.5)
 project(nav2_mppi_controller)
 
-add_definitions(-DXTENSOR_ENABLE_XSIMD)
-add_definitions(-DXTENSOR_USE_XSIMD)
+# Definitions to disable SIMD optimizations in xtensor
+add_definitions(-DXTENSOR_DISABLE_XSIMD)
+add_definitions(-DXSIMD_NO_SUPPORTED_ARCH=1)
 
+# Set threading support flags to off
 set(XTENSOR_USE_TBB 0)
 set(XTENSOR_USE_OPENMP 0)
-set(XTENSOR_USE_XSIMD 1)
-
-# set(XTENSOR_DEFAULT_LAYOUT column_major)  # row_major, column_major
-# set(XTENSOR_DEFAULT_TRAVERSAL row_major)  # row_major, column_major
+set(XTENSOR_USE_XSIMD 0)
 
 find_package(ament_cmake REQUIRED)
-find_package(xsimd REQUIRED)
 find_package(xtensor REQUIRED)
 
-include_directories(
-  include
-)
+include_directories(include)
 
+# List of dependencies
 set(dependencies_pkgs
   rclcpp
   nav2_common
@@ -35,38 +32,24 @@ set(dependencies_pkgs
   tf2_ros
 )
 
+# Find each dependency
 foreach(pkg IN LISTS dependencies_pkgs)
   find_package(${pkg} REQUIRED)
 endforeach()
 
+# Declare this package
 nav2_package()
 
-include(CheckCXXCompilerFlag)
-
-check_cxx_compiler_flag("-mno-avx512f" COMPILER_SUPPORTS_AVX512)
-check_cxx_compiler_flag("-msse4.2" COMPILER_SUPPORTS_SSE4)
-check_cxx_compiler_flag("-mavx2" COMPILER_SUPPORTS_AVX2)
-check_cxx_compiler_flag("-mfma" COMPILER_SUPPORTS_FMA)
-
-if(COMPILER_SUPPORTS_AVX512)
-  add_compile_options(-mno-avx512f)
-endif()
-
-if(COMPILER_SUPPORTS_SSE4)
-  add_compile_options(-msse4.2)
-endif()
-
-if(COMPILER_SUPPORTS_AVX2)
-  add_compile_options(-mavx2)
-endif()
-
-if(COMPILER_SUPPORTS_FMA)
-  add_compile_options(-mfma)
-endif()
-
-# If building one the same hardware to be deployed on, try `-march=native`!
-add_compile_options(-O3 -finline-limit=10000000 -ffp-contract=fast -ffast-math -mtune=generic)
+# Compilation flags optimized for RISC-V
+add_compile_options(
+  -O3                   # Optimization level
+  -ffp-contract=fast    # Enable fused multiply-add
+  -ffast-math           # Fast math optimizations
+  -march=rv64gc         # Target RISC-V rv64gc architecture
+  -mtune=sifive-7-series # Tune for SiFive 7-series processors or `rocket`
+)
 
+# Define libraries and their sources
 add_library(mppi_controller SHARED
   src/controller.cpp
   src/optimizer.cpp
@@ -93,13 +76,15 @@ add_library(mppi_critics SHARED
 
 set(libraries mppi_controller mppi_critics)
 
+# Apply compile options and dependencies for each library
 foreach(lib IN LISTS libraries)
   target_compile_options(${lib} PUBLIC -fconcepts)
-  target_include_directories(${lib} PUBLIC ${xsimd_INCLUDE_DIRS})  #  ${OpenMP_INCLUDE_DIRS}
-  target_link_libraries(${lib} xtensor xtensor::optimize xtensor::use_xsimd)
+  target_include_directories(${lib} PUBLIC ${xtensor_INCLUDE_DIRS})
+  target_link_libraries(${lib} xtensor)
   ament_target_dependencies(${lib} ${dependencies_pkgs})
 endforeach()
 
+# Install the libraries and include files
 install(TARGETS mppi_controller mppi_critics
   ARCHIVE DESTINATION lib
   LIBRARY DESTINATION lib
@@ -110,19 +95,20 @@ install(DIRECTORY include/
   DESTINATION include/
 )
 
+# Testing configuration (if enabled)
 if(BUILD_TESTING)
   find_package(ament_lint_auto REQUIRED)
   find_package(ament_cmake_gtest REQUIRED)
-  set(ament_cmake_copyright_FOUND TRUE)
   ament_lint_auto_find_test_dependencies()
   add_subdirectory(test)
-  # add_subdirectory(benchmark)
 endif()
 
+# Export libraries and dependencies
 ament_export_libraries(${libraries})
 ament_export_dependencies(${dependencies_pkgs})
 ament_export_include_directories(include)
 pluginlib_export_plugin_description_file(nav2_core mppic.xml)
 pluginlib_export_plugin_description_file(nav2_mppi_controller critics.xml)
 
+# Finalize the package
 ament_package()
-- 
2.43.0

...
```

---

### Additional Information

- The package is built and executed successfully on the RISC-V platform after applying the workaround.

SteveMacenski, 2025-01-24:
>     VisionFive 2 (RISC-V architecture)

Ooo neat - I haven't used / had a user mention RISC-V before! This is not an officially supported architecture in ROS-land, but happy to review/merge PRs that resolve this issue so that you can keep going. 

Do see that MPPI's preformance is good without simd optimizations enabled? That is a pretty critical part of how the system can function real-time on CPU-only. I don't think its valuable to support MPPI without simd since I'm not sure it can realistically run at 30+hz with a realistic sampling size otherwise. 

> -mtune=generic

This should work for any processor, that seems like a gap in RISC-V support in CMake. We could check the processor architecture and have different flags set depending on the processor, that should be an easy-enough pull request to submit to get around that. 

---

I'd recommend looking at the `main` branch of Nav2 where we did a recent ground-up rewrite of MPPI to be based on Eigen rather than xtensor, that may resolve your issue.

SteveMacenski, 2025-02-08:
@ahmadSum1 any followup?

ahmadSum1, 2025-02-27:
Sorry about the super late reply. 
If I understand correctly, all the compiler optimization flags should [ideally] be handled by the gcc/cmake?!, which unfortunately is "not there" yet for the visionfive2 or riscv64 in general. I didnt have any opportunity to look any further since MPPI isnt a priority for the control group at the moment. 
Perhaps someone with better understanding of c compilers will fix this upstream while we wait.

SteveMacenski, 2025-03-01:
I will close this then if this is not a priority for your group since RISC-V is not a standard or ROS Tier 1 supported architecture. 

I’m happy to review PRs needed to support it, but without the ability to reproduce, there’s not much actionable here for the community. 

Should this become a priority, please feel free to submit a PR or reopen this ticket for discussion! Maybe down the line this architecture will be more widely used to provide support 🙂

closed 2025-03-01 by SteveMacenski