ros-navigation/navigation2 #4874: nav2_costmap_2d/footprint.hpp `calculateMinAndMaxDistances` for non-symmetric footprints

smkyle90, 2025-01-28:
I am looking to get further understanding on the 
```cpp
void calculateMinAndMaxDistances(
  const std::vector<geometry_msgs::msg::Point> & footprint,
  double & min_dist, double & max_dist);
```
function in nav2_costmap_2d/footprint.hpp.

I believe there is an issue of what "min" means in this functions context and its subsequent contribution to the costmap's inflation layer.

Specifically, in cases where the footprint is asymmetric (not a square or circle) and not "centred" at zero, i.e., `base_link` is not at the centre of the longitudinal axis of the frame. `base_link` can be outside the inflation layer, yet the footprint can be in collision (in particular when using the `CostmapTopicCollisionChecker`).

In investigating this behaviour, I do notice that `calculateMinAndMaxDistance()` only has 1 test, and it simply covers the case for a footprint of length 2 or less making me question if this has slipped through the cracks, I am misunderstanding the point of the inflation layer, or if everyone uses symmetric footprints!

Below is the base "square and centred case", with two extreme cases to demonstrate the current implementation of `calculateMinAndMaxDistances()` function:
1. Square (side lengths of 2) centred footprint.
```md
x------------x
|            |
|      o     |    min_dist: Min distance to edge (half the distance from o to an edge)
|            |    max_dist: Max distance to vertex
x------------x
```

```cpp
TEST(collision_footprint, square_footprint_centred)
{
  geometry_msgs::msg::Point p1;
  p1.x = 1.0;
  p1.y = 1.0;

  geometry_msgs::msg::Point p2;
  p2.x = 1.0;
  p2.y = -1.0;

  geometry_msgs::msg::Point p3;
  p3.x = -1.0;
  p3.y = -1.0;

  geometry_msgs::msg::Point p4;
  p4.x = -1.0;
  p4.y = 1.0;

  std::vector<geometry_msgs::msg::Point> footprint = {p1, p2, p3, p4};
  double min_dist = 0.0;
  double max_dist = 0.0;

  nav2_costmap_2d::calculateMinAndMaxDistances(footprint, min_dist, max_dist);
  EXPECT_EQ(min_dist, 1.0);
  EXPECT_EQ(max_dist, sqrt(2.0));
}
```

2. Square (side lengths of 2) centre of footprint at front edge
```md
x------o-----x
|            |
|            |    `min_dist`: 0.0
|            |    `max_dist`: Max distance to vertex
x------------x
```
```cpp
TEST(collision_footprint, square_footprint_front)
{
  geometry_msgs::msg::Point p1;
  p1.x = 1.0;
  p1.y = 0.0;

  geometry_msgs::msg::Point p2;
  p2.x = 1.0;
  p2.y = -2.0;

  geometry_msgs::msg::Point p3;
  p3.x = -1.0;
  p3.y = -2.0;

  geometry_msgs::msg::Point p4;
  p4.x = -1.0;
  p4.y = 0.0;

  std::vector<geometry_msgs::msg::Point> footprint = {p1, p2, p3, p4};
  double min_dist = 0.0;
  double max_dist = 0.0;

  nav2_costmap_2d::calculateMinAndMaxDistances(footprint, min_dist, max_dist);
  EXPECT_EQ(min_dist, 0.0);
  EXPECT_EQ(max_dist, sqrt(5.0));
}
```
3. Square (side lengths of 2) centre of footprint at rear. Without loss of generality, 3 is the same as 2.

Case 1 here is **not** the issue. For cases 2 and 3, `min_dist` should **not** be 0.0, as this leads to an inflation layer of 0.0, which immediately impacts local collision checking capabilities. 

To account for both forward and reverse travel, I believe `min_dist` should be `2.0`, i.e., the _maximum_ distance from the footprint origin to the edge of the footprint. 

As such, the tests for cases 2 and 3 should be

```cpp
  nav2_costmap_2d::calculateMinAndMaxDistances(footprint, min_dist, max_dist);
  EXPECT_EQ(min_dist, 2.0);
  EXPECT_EQ(max_dist, sqrt(5.0));
```
and "min" in the context of this function is really "max distance to an edge". There is no change to the `max_dist`, or that this is the "max distance to a vertex".

Similar logic can be applied to rectangular footprints.
- For footprints that are wider than they are long: the costmap can promote perhaps non-collision free travel in narrower areas.
- For footprints that are longer than they are wider: the costmap can promote travelling too close to an area of high cost.

If my understanding of this intended behaviour is off, please let me know.

Either way, if there an appetite to wrap some tests around the `calculateMinAndMaxDistances()` function to better demonstrate its use, let me know.

SteveMacenski, 2025-01-28:
`calculateMinAndMaxDistances`: This finds the min/max distances from the robot's footprint origin. The `0,0` of the footprint necessarily needs to be `base_link` by definition. It is not required to be centered with the robot and commonly isn't for Ackermann drives where it is the center of the rear axel or a differential drive robot where it doesn't pivot around the center).

> base_link can be outside the inflation layer, yet the footprint can be in collision (in particular when using the CostmapTopicCollisionChecker).

The robot should always be fully encapsulated within the costmaps for use of almost all planning/control algorithms. The inflation layer is simply the size of the costmap (i.e. static map, set parameterized size, or size of costmap generated by the sensor data sizing). If the robot's footprint is not within the size of the costmap, planning / control algorithms will fail to work properly; most will report an error or ignore poses where the full footprint used for collision checking isn't in bounds. 

The tests are pretty minimal since this is legacy code from ROS 1 and is ~10 years old (and hasn't been changed since). 

> Square (side lengths of 2) centre of footprint at front edge

This seems like an unrealistic configuration for a robot solution to have the base_link frame, in the standard conventions of how robots are built, to be on an external edge of the footprint. Certainly you could also configure the costmap to have it external to the footprint entirely, but that is also not really how robots are built nor the conventions set. That seems analogous to setting an AMCL particle count to `-1` and expecting it to work, with negative resources 😉 

is this a real issue that has come up based on an actual robot system or are you just finding edge cases? 

> To account for both forward and reverse travel, I believe min_dist should be 2.0, i.e., the maximum distance from the footprint origin to the edge of the footprint.

`calculateMinAndMaxDistances` is only used to compute `inscribed_radius_` and `circumscribed_radius_` which seems proper to me. There's no relationship with min / max distances or the radiuses for collision checking and the direction of travel. 

But to your comment, no, the min_dist in that situation would be `0.0` because that is the distance from `base_link` to the nearest edge. But that is also seemingly a critical misconfiguration / understanding of how to set your footprint up if you're actually running into that problem :-) 

---

But lets take a step back and ask: what is the actual problem you're trying to solve?

SteveMacenski, 2025-03-14:
@smkyle90 any update?

padhupradheep, 2025-05-09:
Hey @smkyle90, I think @SteveMacenski has given an elaborate answer here. Is there something we could try and help you figure out, if not we can close this ticket.

smkyle90, 2025-05-09:
My issue is (was?) related to using the `CostmapTopicCollisionChecker` for collision prediction. It's not built for this, and as I am using it in a non-standard way, there are non-standard issues that come up. 

The issue I describe above exists, and presents itself in extreme cases where the predictive path has a large discretization (say 1m) and the costmap is sparse (say 1 grid's worth of cost). As such any footprint collision checking "misses" the collision as the line rasterization does not intersect the lethal cost AND the base pose is outside of the inflation radius (due to how the radius is computed).

All to say, for our actual application, where paths are discretized at ~0.2m and our lidar returns generate a dense costmap we significantly decrease the probability of the future collision being missed.

So much ado about nothing, and can be closed.