ros-navigation/navigation2 #4876: Voronoi boundary based global Path planning?

robotics-qc, 2025-01-29:
Hello Guys!

So I have been using Nav2 to projects for a while and use different local planners (such as DWB or Regulated Pure pursuit etc).

But for situations where the there are narrow corridors, cubicles etc (like an office environment) i thought having a global path that plans with maximum distance from all obstacles might be useful

Wondering if there is a planner or plugin that could be used to generate a global path that is more akin to a voronoi boundary - 

[image]

A smoothed version of the above for example?

If not, is there a use case to add this feature?

Thanks!

SteveMacenski, 2025-01-29:
Right now, there is not! 

There was a[ Voronoi planner in ROS 1](http://wiki.ros.org/voronoi_planner), but my understanding is that it was pretty slow and it was not really maintained. If you'd be open to developing an efficient / performant version, that could be something we'd consider merging into Nav2 (or at least nav2 auxiliary or its own repository under the `ros-navigation` org). 

I think the struggle was to efficiently generate the voronoi diagram each iteration since it should consider the cost changes due to obstacles inserted in the costmap from sensor data. I'm not sure if it was actually all that difficult, if scaling with larger maps is difficult given the size of the map, or this was a case where research code wasn't particularly well optimized. 

I think if you were open to developing and refining something like this, that would be a great contribution. It might be worth looking at more modern literature on Voronoi planners / using voronoi diagrams for planning & control to see if there's any tips / tricks we should consider in implementation to make it more powerful. 

For example, just a quick search came up with https://arxiv.org/html/2201.12981v4 from 2018 (which I only vaguely scanned through, not sure if there's anything particular in _this one_ but we should look at several).

For another example, maybe we generate a really nice voronoi library and we can use it for multiple things. For example: Do grad descent on the distance field it generates for an A* or kinematically feasible search primitives. Or do a more traditional Voronoi graph connection like your image. Or even use as part of the cost function for MPC or optimization-based techniques. Or do multi-resolution voronoi so that we can scale better for larger spaces. Or a number of other potential things.

robotics-qc, 2025-01-29:
That is interesting, let me think about that, and research some more!

Regarding compute efficiency - yes I see having a global planner that needs to constantly re-plan on a new map will need some good optimization to make it efficient.

Just thinking aloud here,
There are many applications where the constantly updating global costmap is not required - ( I have often disabled the obstacle layer myself in very dynamic environments to stop noise/dynamic objects from leaving artifacts on the global costmap myself)

I have had discussions with my peers and colleagues who also use Nav2 - we felt if there was a means to use a 'roadmap' based planner that can compute a graph/roadmap based on requirements (PRM, Voronoi boundaries, cost or custom defined potential fields) - this makes the path planning very deterministic and compute efficient!

On or before launch we can provide time to compute (either online or apriori) a path network in the provided map and then runtime becomes fast, easy on the compute, smoother trajectories etc.

The local planner will take care of sudden obstacles anyway, and the global planner will only need to compute paths to join/exit the roadmap graph.

In many AGV or AMR ground robots - especially for warehouses, parking lot, factories, office corridors etc I believe this may be useful

Thoughts?

SteveMacenski, 2025-01-29:
>  There are many applications where the constantly updating global costmap is not required - ( I have often disabled the obstacle layer myself in very dynamic environments to stop noise/dynamic objects from leaving artifacts on the global costmap)

I think that's a useful special case to support, but I would like things in Nav2-main source to support dynamically updating environmental models since those are the most common use-case with autonomous navigation (PS sounds like you might need different obstacle layers!). I'd still want it usable with just the cost map, but we could add in a feature parameter to use only the static map (TBD how to obtain it). Either way, the functions used to process the static map vs the costmap would be identical. 

>  we discussed how if there was a means to use a 'roadmap' based planner that can compute a graph/roadmap

There's an open ticket and some early draft work for that in Nav2: Nav2 Route Server. This is something I'm actually working on again starting next month (if I can ever get ahead of my emails and github reviews 😆 ). But this is another topic :-)

robotics-qc, 2025-02-04:
Okay so I have been reading more of the Nav2 documentation, and playing with the different controllers -

so I see if I set a super high inflation radius and a cost scaling then I largely get global paths that are close enough to voronoi boundaries on the whole for a few maps - it is just that when we have narrow spaces of differing widths then the cost scaling ends up affecting route feasibility. 

So I guess if we just adjust the inflation radius i.e if we max the inflation radius to be as big as the widest gap, then the global path planner basically does end up somewhat following an apporximate voronoi boundary (as long as the map  has corridors and hallways without too much clutter)

Thinking of ways we can generalize that for maps even with many objects -> essentially inflation with dynamic cost scaling...

Just a train of thought, will think more about it

SteveMacenski, 2025-02-05:
That would be a benefit of voronoi - it would be a replacement for inflation. Either as a costmap layer itself using the core library and/or as a planner using the voronoi field and/or ...

patham9, 2025-02-09:
Such a Voronoi-boundary path planner would be cool! Hm however it seems to be the case that the path suggested by Voronoi can be arbitrarily small so inflation might still be good to have in addition:
[image]
Maybe these ideas cannot replace each other but could potentially complement each other.
What do you think?

SteveMacenski, 2025-02-10:
I think definitely the value of the distance in the voronoi diagram needs to be looked at with respect to the footprint's size. The way we do that I think depends on the application (voronoi planner, potential field controller, just diagram for costmap `inflation_layer` alternative, etc). If its too narrow, we definitely need to reject its uses, but I don't think that's much different than any other planning / control algorithm: as we find the 'best path' or 'right direction', we check its validity and back out and go down a different branch if its not. 

That could be done with a basic convolution around obstacles like the inflation layer does with inscribed costs (just without the later decay functions). That is probably very sensible :-)

robotics-qc, 2025-02-13:
I was reading up that using SDFs for several primitives is common for 3D ray marching etc and is optimized for real time rendering. 

I am wondering if we can leverage any ideas from that and model our obstacles/sensor inputs as SDF primitives and borrow libraries/implementations from graphics use cases (which I would assume to be pretty efficient?)

Potentially opens the possibility to expand path planning in 3D voxel spaces in the future too!