ros-navigation/navigation2 #4888: Audit QoS settings in Nav2

SteveMacenski, 2025-01-30:
It would be good to do some auditing on QoS settings in Nav2 and make sure that they are sensible. In particular:
* Subscriber depths should be as minimal as possible to keep the data fresh and only process what is live and needed. Older data probably doesn't need to be kept if the system isn't keeping up. However: 
* Publisher depths should be larger, such that they can keep a queue (when using async publication) to process to make sure messages aren't lost, its up to the subscriptions to decide they're not interested 
* Remove or reduce use of best effort QoS that has the unfortunate side effect of blasting data at full rate, even if the network is having trouble keeping up, without any ACKs. Really, this should be used sparingly when on the same CPU. best effort's CPU performance is pretty not-great and can make network issues worse due to its ignorance about what else is going on and its own communication status
* Transient local for any static published data states 

Consider also adding in support for the newer features like:
* Deadline to notify when issues arise and handle them
* Lifespan to disregard sufficiently old data
* Liveliness to possibly replace bond in the lifecycle manager (but probably not?) 

It might also be good to have launch files that launch and process a discovery server to reduce network traffic.

DDS configs like transport layer, sync vs async, local host only, buffer / fragment sizes, unicast, etc should be left to the user's setup

---

## Considerations

I think we could have a set of Nav2 specific QoS policies for "publishers", "subscribers", and "latched" so that these are portable and consistent across the code base. During the audit, rather than fix each one, we can move each to use our default profiles, unless there is a compelling reason for some to be differentiated. 

We could even wrap the `create_XYZ()` factories in a Nav2 version that also does things under the hood with respect to QoS override acceptance, deadline/liveliness callbacks (`nav2_utils::LifecycleNode` to handle), perhaps even lifecycle for subscription, etc

ewak, 2025-01-30:
I commend the idea.

When not using Best Effort.  I *think* I of run into the problem of slow subscribers effectively live locking the publishers and other subscribers.  I try to mitigate that somewhat, especially when networking is involved, say for example costmap visualization on wifi, using some features of zenoh-ros2dds-bridge  (on humble).  https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/blob/b662a95730f8daeaa71669406d0e00bef7898bd5/DEFAULT_CONFIG.json5#L95

Is it just me and my misguided superstition? 

Do you have goto tools/techniques to discover/measure/pinpoint slow subscribers?

SteveMacenski, 2025-01-31:
>  I think I of run into the problem of slow subscribers effectively live locking the publishers and other subscribers. 

I believe that essentially is the behavior with using DDS' synchronous publishing (default) since the subscriber sends ACKs to the publisher, it does flow control on publication rate depending on packet loss and network bandwidth. There's some relationship going on there under the hood - which is good when you need to reliably guarantee delivery. I've recently been auditing the DDS implementations' documentation and tuning guides and this is the reason they recommend Sync publications for robotics / critical systems (where async where it is published in a separate thread is better for high rate streaming data/lower overhead; but comes at the downside of not being as reliable and as fancy tools for critical data guarantees). 

So no, I don't think that's probably superstition & reliable transport over WiFi is probably bad for that (and likely other) reasons. This is a benefit of the Best Effort publisher where there is no ack, flow control (but instead sends at full rate without regard for subscriber processing state to keep up or networking load). 

> Do you have goto tools/techniques to discover/measure/pinpoint slow subscribers?

I don't have any, but I know of some. [ROS 2 Tracing](https://github.com/ros2/ros2_tracing) was built to put tracepoints across the software so you can have timing information while things are processing. I haven't used, but I know it is well loved. There are also DDS logs that come out, but those are a _firehose_ and I've never personally successfully debugged a problem from them given how much is going on in Nav2.

SteveMacenski, 2025-05-27:
Oh! Also subscription, publisher, service, and action options object population!

SteveMacenski, 2025-05-28:
Start was completed in https://github.com/ros-navigation/navigation2/tree/4888

Remaining items:
* Migrate all uses of create_service, create_client, create_subscription, create_publisher, create_action to use this implementation
* Migrate all uses of previous constructors for ServiceClient, ServiceServer, SimpleActionServer to use the new lifecycle node `create_*` factories to mimic `rclcpp::create_*` but now returning Nav2 types (except for Action Clients, those are good as is)
* Determine distribution strategy: Should these files move into a new package (or perhaps `nav2_common`?) since they're foundational to all Nav2 nodes and no longer simply 'utilities'? Moving Simple Action Server, Service Client, Service Server, Lifecycle Node, and these new QoS and factories seems to make sense to bin together (maybe others?) as a `nav2_ros2`, `nav2_common`, `nav2_foundation` or so on. Make sure to make file naming, namespace naming, clear and aligned with this as well. 
* Finally, analyze the use of QoS profiles. Minimize best-effort sensor data QoS to where its needed only & increase the number of transient local topics for any information that's largely static.

mini-1235, 2025-06-07:
> Migrate all uses of previous constructors for ServiceClient, ServiceServer, SimpleActionServer to use the new lifecycle node create_* factories to mimic rclcpp::create_* but now returning Nav2 types (except for Action Clients, those are good as is)

We might actually want to also create a class for action clients(?) I noticed that action introspection is available since Kilted
https://docs.ros.org/en/kilted/Tutorials/Demos/Action-Introspection.html, this need to be configured like
```cpp
this->action_server_->configure_introspection(this->get_clock(), rclcpp::SystemDefaultsQoS(), introspection_state);
```

SteveMacenski, 2025-06-09:
Noted and added to my expanding TODO list 😆

SteveMacenski, 2025-06-11:
@mini-1235 that's now done in my branch including action server introspection. Still a bunch of cleanup and work to go, but I'm over the biggest hurtles now

mini-1235, 2025-06-30:
> [@mini-1235](https://github.com/mini-1235) take a look at [#5288](https://github.com/ros-navigation/navigation2/pull/5288)

@SteveMacenski Sorry I was on a vacation, I got a few questions regarding the PR:

- If I am not mistaken, the action introspection can be configured by setting `service_introspection_mode` to true now, in that case, I think that we can rename it to `introspection_mode` to avoid ambiguity(?) Or we can also separate them into two parameters( `service_introspection_mode`  + `action_introspection_mode` 

- I saw different qos settings for message type like `sensor_msgs/Image` and `sensor_msgs/Pointcloud`, can you explain a little bit about those settings

https://github.com/ros-navigation/navigation2/blob/97e25ec4dfc5f2d4ec7ef141219c54e54e8edd27/nav2_collision_monitor/src/pointcloud.cpp#L63-L67

https://github.com/ros-navigation/navigation2/blob/97e25ec4dfc5f2d4ec7ef141219c54e54e8edd27/nav2_costmap_2d/plugins/obstacle_layer.cpp#L232

SteveMacenski, 2025-07-01:
> I think that we can rename it to introspection_mode to avoid ambiguity

Sure! Want to open a PR? We just mostly need to update the configuration guide docs / the migration guide docs

They're both `nav2::qos::SensorDataQoS`, so they are the same. The obstacle layer has a larger depth set for some reason that I didn't care to argue with at the moment when doing the migration. That was just the behavior previously so I maintained it as to not change the behavior at the same time as the large migration.

mini-1235, 2025-07-02:
> They're both nav2::qos::SensorDataQoS, so they are the same. The obstacle layer has a larger depth set for some reason that I didn't care to argue with at the moment when doing the migration. That [...]

Ahh ok, I just didn't notice that it was originally depth 50