ros-navigation/navigation2 #4892: Keepout filter not being received correctly

Flamethr0wer, 2025-02-01:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Computer:
  - Raspberry Pi 4B 8GB-RAM
- ROS2 Version:
  - Humble binaries
- Version or commit hash:
  - ros-humble-navigation2	1.1.17-1jammy.20250114.205626
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
As far as I know, just launch Nav2 using a keepout filter mask. I have tried both my own launch files and the ones from nav2_bringup, and the yaml files are correct as indicated by the [docs](https://docs.nav2.org/tutorials/docs/navigation2_with_keepout_filter.html). However, here are the files I'm trying to use.
Launch file:
```
import os

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, GroupAction, SetEnvironmentVariable
from launch.conditions import IfCondition
from launch.substitutions import LaunchConfiguration, PythonExpression
from launch_ros.actions import LoadComposableNodes
from launch_ros.actions import Node
from launch_ros.descriptions import ComposableNode, ParameterFile
from nav2_common.launch import RewrittenYaml

def generate_launch_description():
    # Get the launch directory
    bringup_dir = get_package_share_directory('robot')

    namespace = LaunchConfiguration('namespace')
    use_sim_time = LaunchConfiguration('use_sim_time')
    autostart = LaunchConfiguration('autostart')
    params_file = LaunchConfiguration('params_file')
    map_yaml_file = LaunchConfiguration('map')
    use_composition = LaunchConfiguration('use_composition')
    container_name = LaunchConfiguration('container_name')
    container_name_full = (namespace, '/', container_name)
    use_respawn = LaunchConfiguration('use_respawn')
    log_level = LaunchConfiguration('log_level')

    lifecycle_nodes = ['map_server',
                       'amcl',
                       'controller_server',
                       'smoother_server',
                       'filter_mask_server', 
                       'costmap_filter_info_server',
                       'planner_server',
                       'behavior_server',
                       'bt_navigator',
                       'waypoint_follower',
                       'velocity_smoother']

    # Map fully qualified names to relative ones so the node's namespace can be prepended.
    # In case of the transforms (tf), currently, there doesn't seem to be a better alternative
    # https://github.com/ros/geometry2/issues/32
    # https://github.com/ros/robot_state_publisher/pull/30
    # TODO(orduno) Substitute with `PushNodeRemapping`
    #              https://github.com/ros2/launch_ros/issues/56
    remappings = [('/tf', 'tf'),
                  ('/tf_static', 'tf_static'),
                  ('/cmd_vel', '/diff_cont/cmd_vel_unstamped')]

    # Create our own temporary YAML files that include substitutions
[... 920 lines cut ...]
[component_container_isolated-1] [INFO] [1738414378.245165033] [behavior_server]: Activating drive_on_heading
[component_container_isolated-1] [INFO] [1738414378.246122212] [behavior_server]: Activating assisted_teleop
[component_container_isolated-1] [INFO] [1738414378.246539653] [behavior_server]: Activating wait
[component_container_isolated-1] [INFO] [1738414378.246928132] [behavior_server]: Creating bond (behavior_server) to lifecycle manager.
[component_container_isolated-1] [INFO] [1738414378.376508331] [lifecycle_manager_navigation]: Server behavior_server connected with bond.
[component_container_isolated-1] [INFO] [1738414378.376686441] [lifecycle_manager_navigation]: Activating bt_navigator
[component_container_isolated-1] [INFO] [1738414378.377595750] [bt_navigator]: Activating
[component_container_isolated-1] [INFO] [1738414378.914473107] [bt_navigator]: Creating bond (bt_navigator) to lifecycle manager.
[component_container_isolated-1] [INFO] [1738414379.035651529] [lifecycle_manager_navigation]: Server bt_navigator connected with bond.
[component_container_isolated-1] [INFO] [1738414379.035984212] [lifecycle_manager_navigation]: Activating waypoint_follower
[component_container_isolated-1] [INFO] [1738414379.036790781] [waypoint_follower]: Activating
[component_container_isolated-1] [INFO] [1738414379.036985520] [waypoint_follower]: Creating bond (waypoint_follower) to lifecycle manager.
[component_container_isolated-1] [INFO] [1738414379.175092421] [lifecycle_manager_navigation]: Server waypoint_follower connected with bond.
[component_container_isolated-1] [INFO] [1738414379.175322161] [lifecycle_manager_navigation]: Activating velocity_smoother
[component_container_isolated-1] [INFO] [1738414379.178127642] [velocity_smoother]: Activating
[component_container_isolated-1] [INFO] [1738414379.178575250] [velocity_smoother]: Creating bond (velocity_smoother) to lifecycle manager.
[component_container_isolated-1] [INFO] [1738414379.306648181] [lifecycle_manager_navigation]: Server velocity_smoother connected with bond.
[component_container_isolated-1] [INFO] [1738414379.306841532] [lifecycle_manager_navigation]: Managed nodes are active
[component_container_isolated-1] [INFO] [1738414379.306964883] [lifecycle_manager_navigation]: Creating bond timer...
```
If I try to visualize the data in /keepout_filter_mask on Rviz, I get "No map received":
[image]

#### Additional information
When launching the stack with the aforementioned Rviz view active, sometimes I see the raw data of my normal (not keepout) map on the /keepout_filter_mask map display for a moment, but this disappears after clicking on Reset and no data is published to the topic again.

SteveMacenski, 2025-02-03:
>  if I set the QoS settings on Rviz to transient_local and reliable, I can actually see something published to /keepout_filter_mask

👍 Good, then it working :-)

> but it is a kind of raw version (with no inflation visible, just the walls) of my NORMAL map

Then that's an issue with your mask map itself, you should follow the tutorial or adjust your filter metadata so that it translates into the cost you want with the `cost = base + value * multiplier` formulation. Or, maybe you're not giving it the mask and instead passing in your map's filepath to that server? 

We also have more examples in our system tests that show this in operation (and works!) that you may want to reference: https://github.com/ros-navigation/navigation2/tree/main/nav2_system_tests/src/costmap_filters

Flamethr0wer, 2025-02-03:
@SteveMacenski But I haven't modified the metadata, and either way it should be OK. And why would the map's filepath be passed to that server? My params file clearly gives the path to the filter file.
I have downloaded a filter mask from the demos to see if it was the file itself and it didn't work.

SteveMacenski, 2025-02-03:
Please check out the tutorial https://docs.nav2.org/tutorials/docs/navigation2_with_keepout_filter.html and associated configuration guide parameters https://docs.nav2.org/configuration/packages/costmap-plugins/keepout_filter.html. I think these explain it well :-)

Flamethr0wer, 2025-02-03:
@SteveMacenski I have read through those like 20 times already. I'm speaking of a case of the thing stopping to work with no changes to the files themselves. I think I've checked every single file involved for changes and found nothing. I've used defaults, tutorial files, my own files, but nothing.

SteveMacenski, 2025-02-03:
I think your best bet is to ask on Robotics Stack Exchange with a description of the issue. We have system tests and users of this in the wild, so I'm not 100% sure what your issue is, but I'm sure from the documentation and provided system integration test references you can find the error. 

It seems to me that things are working correctly, that your issue was an incorrect QoS in Rviz2. Now, you have the right visualizations, but your mask is not setup or configured correctly, which the documentation should help solve for you, with additional examples you reference in the system tests.  

I understand that can be frustrating, but we don't have the resources to individually debug every configuration / setup problem for all users as they're getting things setup for the first time. If we can't find a particular bug and we have tests and users using this feature, then this is something you're going to need to that the initiative to fix on your own system / hardware 😦

closed 2025-02-03 by SteveMacenski

Flamethr0wer, 2025-02-03:
@SteveMacenski No, but the issue isn't that I couldn't see it on Rviz, it's that using the global_costmap visualization worked before, showing the keepout filters, and now it doesn't, but mainly that my /keepout_filter_mask has mysteriously stopped publishing the mask and I have no idea why or how to debug it! I've been bumping the question for several days on Stack Exchange and have received no answer on guidance on how to even find out what's happening.

SteveMacenski, 2025-02-03:
I'm clearly seeing logging that this is working correctly:

> [controller_server-5] [INFO] [1738261723.566144942] [local_costmap.local_costmap]: KeepoutFilter: Received filter mask from /keepout_filter_mask topic.

Your issue is not with Nav2, I'm pretty certain its with your mask definition. See the tutorial and system tests examples please for assistance there and in configuring the mask metadata topic for the base / multiplier interpreter.

Flamethr0wer, 2025-02-03:
@SteveMacenski But that message is for the initial, wrong mask; I have confirmed the /keepout_filter_mask topic doesn't publish any data beyond that. Also, I am using the exact same mask and metadata from the last time it worked and have even tried the ones from the demos, so that's not it either.

Flamethr0wer, 2025-02-05:
It turns out the issue was because of a param substitution in the launch file; the parameter for the costmap filter and normal map yaml files has the same name (maybe that's something to consider changing). I don't understand why it didn't work with the demo files then, but whatever, I got it working by removing the parameter in the launch file altogether.