ros-navigation/navigation2 #4895: sam_bot_description, No module named 'ros_gz_bridge'

J-Streib, 2025-02-02:
- Operating System:
  -Ubuntu 22.04
- Computer:
  -  Nvidia Jetson Orin Nano
- ROS2 Version:
  - Humble binaries
- Version or commit hash:
  - from source: b7de5ceb2081083068e6e5c96850d001f0f1fd2b
  -  apt binaries: ros-humble-navigation2	1.1.17-1jammy.20250114.205626

- DDS implementation:
  - I don't know what this means
- Nav2 Package:
  - sam_bot_description
- Have you asked this on Robotics Stack Exchange?
  - No

#### Tuning / Configuration Goal
I am a beginner to ROS2 am trying to implement Nav2 on a custom robot for autonomous navigation. I am currently following the getting started tutorial to understand how Nav2 works, however when I run the launch file I get this output:

[INFO] [launch]: All log files can be found below /home/rovers/.ros/log/2025-02-02-14-26-40-980776-rovers-desktop-12024
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught multiple exceptions when trying to load file of format [py]:
 - ModuleNotFoundError: No module named 'ros_gz_bridge'
 - InvalidFrontendLaunchFileError: The launch file may have a syntax error, or its format is unknown

When I run ros2 pkg list I see that I should have all of the following  ros_gz packages installed:

ros_gz
ros_gz_bridge
ros_gz_image
ros_gz_interfaces
ros_gz_sim
ros_gz_sim_demos

#### What I've Tried
I came across this issue while using code that I've written using the tutorial. I added the following lines to my launch.py 

from ros_gz_bridge.actions import RosGzBridge
from ros_gz_sim.actions import GzServer

but still had no luck. It seems that python can't find the ros_gz_bridge and ros_gz_sim packages.

I then cloned the sam_bot_description package to see if that would work but am coming across the same error. I can't find any resources online that explain this issue.
 Any help would be greatly appreciated!

padhupradheep, 2025-02-02:
Hey, 

what exact launch are you trying to run? 

>from source: b7de5ceb2081083068e6e5c96850d001f0f1fd2b

This commit has points to the rolling branch, could you please switch to the humble branch and try once again? 

Checkout this nice [first time robot-setup guide](https://docs.nav2.org/setup_guides/index.html), will surely give you a lot of ideas.

SteveMacenski, 2025-02-02:
Please file the ticket on navigation2_tutorials where that code lives. Please also include the error and make sure you’re using the right branch for your distribution

closed 2025-02-02 by SteveMacenski