ros-navigation/navigation2 #4897: ‎ObservationBuffer::bufferCloud: also filter for points outside of min-max obstacle range

flabrosse, 2025-02-03:
## Feature request

Filter out points that are too far (outside of min-max range) from a pointcloud when buffering the observation.

#### Feature description

‎`ObservationBuffer::bufferCloud()` removes (or rather does not copy) points from the observation that are outside of specified min and max heights (`min_obstacle_height` and `max_obstacle_height`).

Why not do the same with points outside of [`obstacle_min_range`,`obstacle_max_range`]?  This could make all the further processing lighter, without the need to filter these points out in the costmap processing that comes after.

#### Implementation considerations

This could be implemented very simply by adding the check in the already existing loop (that omits the points outside of the vertical range).

SteveMacenski, 2025-02-03:
This is behavior from ROS 1 more than 10 years ago, so I don't honestly know why the trade offs were made like that. It was implemented that way since the first commit of ROS 1 Navigation back in 2009. 

If you wanted to experiment with moving that computation inside of `bufferCloud`, and you can show performance improvements, I'm happy to review + merge a PR implementing it :-)

mikeferguson, 2025-02-03:
So, if you remove all points beyond `obstacle_max_range` when adding to the observation buffer, you can't clear the ray towards them, right? (because they are gone long before the Obstacle/Voxel Layer actually gets to process them)

I actually had never noticed that the ObservationBuffer does this early filtering of points below/above the limits - but I wonder if this might account for why it is sometimes hard to properly clear obstacles when using 3d cameras that have a very large vertical field of view (if you see the ceiling, but the ceiling is above your max height, you can't use that to clear space from what I'm seeing here).

I could see two possible changes here (not sure which of them is actually correct right off the bat):

 * Add the min/max obstacle range check to the observation buffer as suggested - but can we also then remove the (possibly duplicated) checks for both vertical and horizontal limits in ObstacleLayer::updateBounds?
 * Get rid of the vertical check in the observation buffer (since it will allow us to clear with more available data, and there is already bounds checking in ObstacleLayer::updateBounds)

SteveMacenski, 2025-02-03:
> You can't clear the ray towards them, right

Ah, yes, indeed! 

>  but I wonder if this might account for why it is sometimes hard to properly clear obstacles when using 3d cameras that have a very large vertical field of view

I think this is what Davide brought up in this ROS 1 ticket I remember from way back when:  https://github.com/ros-planning/navigation/issues/662 (perhaps also https://github.com/ros-planning/navigation/issues/267 from searching).

---

On the changes: The min/max range in the observation buffer would need to make sure it used the right raytrace vs obstacle range values (since they can be different). Now that I think of it, until very recently, the raytrace didn't have a minimum. We added that in Nav2 within the last 2-3 years. That could explain why this wasn't in the observation buffer in ROS 1. 

> Get rid of the vertical check in the observation buffer (since it will allow us to clear with more available data, and there is already bounds checking in ObstacleLayer::updateBounds)

Given we're already looping over the pointcloud, I doubt that this really contributes much computation. 

But also we have the `min_obstacle_height_` for the layer, which is stored as `min_obstacle_height_` in the obstacle layer:

```
  node->get_parameter(name_ + "." + "min_obstacle_height", min_obstacle_height_);
```

Separately, we have the min_obstacle_height for each source, that is used for the value stored in the ObservationBuffer:

```
    node->get_parameter(name_ + "." + source + "." + "min_obstacle_height", min_obstacle_height);
```

These can be 2 separate values. Honestly, so much about this can be refactored / simplified.

mikeferguson, 2025-02-03:
Wow, I had completely forgotten about that 7 year old issue and the discussion - [this comment](https://github.com/ros-planning/navigation/issues/662#issuecomment-376565069) is actually the important part I think - although I'm not sure if Nav2 works the same way (basically, that the min/max in the observation buffer, at least in ROS1, is not the same parameter as the min/max for the layer - this allows you to crop individual sensors in the Z plane - which was a feature used in some ROS 1 robots, notably the PR2).

We had basically zero test coverage in ROS 1, so making deep changes was a lot harder there - in theory the test coverage in nav2 should support deciding if changes are actually breaking typical configurations.

SteveMacenski, 2025-02-04:
Our testing is improved, but also the 'legacy' packages from ROS 1 definitely have higher statistical gaps in test coverage (the coverage they mostly have are system-level smoke tests). I'm definitely open to making changes that are meaningful to fix bugs or make performance boosts, but I am a little weary about changes for architecture's sake in specifically AMCL and Costmap2D.  

Costmap2D in Nav2 works virtually identically at that level. We've added in a few new things like min raytracing range and costmap filters, but the standard functional logic is all the same beyond fixes made with threading and similar reported by users. The multiple min/max parameters can be still set in the same way. 

So with all this in mind, I think this answers @flabrosse's question. I don't think there's a necessary change to be made here unless a prototype was shown to make a meaningful performance improvement that was needed on a system. I don't think this brings up a new/old bug that we should address in Nav2 to be actioned upon, correct?  It looks like the patch from ROS 1 was applied here already in https://github.com/ros-navigation/navigation2/pull/1354 a long while back when I was on a sprint looking through all old ROS 1 navigation tickets / PRs to apply to Nav2.

SteveMacenski, 2025-02-04:
That observation buffer is part of the plugin itself, that is not performed by costmap 2D. If building a custom layer, you could choose not to use that and build your own buffer, process data in the callbacks, or otherwise process the data as you see fit for your applications. It might be that this utility isn't the right utility for you, or you can set the min/max heights when you construct the buffers to being sufficiently large that you don't do any data pruning. 

PS I have a non-persisting voxel layer plugin that you might be interested in https://github.com/SteveMacenski/nonpersistent_voxel_layer. It is the voxel layer except it doesn't do raytracing or persist measurements between updates so its only the current representation of the world. It sounds very similar to what you're looking for and is released into binaries for all major ROS distributions.

flabrosse, 2025-02-05:
Thanks @SteveMacenski.  I'll look at your nonpersistent voxel layer.

I had spotted that I could not use the observation buffer, and will probably get rid of it.   However, I only just now realised that the processing does not have to happen in updateBounds() but could indeed happen in the callback.  It all started as an exploration and I need to re-engineer all this.

I'm into off-road driving, so needed something more sophisticated than just marking things that are too high as obstacles.  My plugin populates the layer with a cost that is based on the gradient of the landscape and also checks for too tall steps.  As I said at the start, one of my use-cases is to run this on an RPi so I really try hard to make it as efficient as possible.  I'll make this public when I get something a bit more polished.

I'll close to issue as it was a non-issue.  Thanks for the pointers and suggestions.

SteveMacenski, 2025-02-05:
>  As I said at the start, one of my use-cases is to run this on an RPi so I really try hard to make it as efficient as possible. I'll make this public when I get something a bit more polished.

Please! I know folks have asked for something like this before. I have it in my TODO list, which as you may expect is long and growing, but its much nicer if its user contributed based on experience / testing to verify reliability :-)