ros-navigation/navigation2 #4898: Add FootprintCollisionChecker to Humble branch

ThomasHaley-neya, 2025-02-03:
## Feature request

#### Feature description
The FootprintCollisionChecker python API is not present in the Humble version of ROS2. After some messing around a bit on my local machine it seems pretty trivial to backport it, and I feel it would be helpful to have.

#### Implementation considerations
The only edit I had to make to the current FootPrintCollisionChecker on `main` to get it working on `humble`  was to change `return self.costmap_.worldToMapValidated(wx, wy)` to `return self.costmap_.worldToMap(wx, wy)` in the `worldToMapValidated()` function

ThomasHaley-neya, 2025-02-03:
Made a PR:

https://github.com/ros-navigation/navigation2/pull/4899

SteveMacenski, 2025-02-04:
Closing with PR open. This is pretty straight forward and I already followed up in the PR on a last question before merging.

closed 2025-02-04 by SteveMacenski