ros-navigation/navigation2 #4904: Voxel layer misbehaving.

kulkarni-raunak, 2025-02-05:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Computer:
  - Intel® Core™ i7-7700K CPU @ 4.20GHz × 8
- ROS2 Version:
  - ROS2 Jazzy (source build)
- DDS implementation:
  - Default

#### Expected behavior
Local_costmaps and Global_costmaps are visualised in rviz2 without any problems.

#### Actual behavior
1. Voxel layers are missing. 
2. Only Costmaps related to particular fixed_frames appears. (Global_costmap for map and local_costmap for odom)

#### Reproduction instructions

my_nav2_params.yaml
```yaml
bt_navigator:
  ros__parameters:
    use_sim_time: False
    global_frame: map
    robot_base_frame: base_link #TODO: 
    odom_topic: /ad/odometry #TODO: 
    bt_loop_duration: 10
    default_server_timeout: 20
    enable_groot_monitoring: True
    groot_zmq_publisher_port: 1666
    groot_zmq_server_port: 1667
    navigators: ["navigate_to_pose", "navigate_through_poses"]
    navigate_to_pose:
      plugin: "nav2_bt_navigator::NavigateToPoseNavigator"
    navigate_through_poses:
      plugin: "nav2_bt_navigator::NavigateThroughPosesNavigator"
    error_code_names:
      - compute_path_error_code
      - follow_path_error_code      

bt_navigator_rclcpp_node:
  ros__parameters:
    use_sim_time: False

controller_server:
  ros__parameters:
    use_sim_time: False
    controller_frequency: 5.0
    min_x_velocity_threshold: 0.1
    min_y_velocity_threshold: 0.1
    min_theta_velocity_threshold: 0.1
    failure_tolerance: 0.3
    progress_checker_plugins: ["progress_checker"]
    goal_checker_plugins: ["general_goal_checker"] # "precise_goal_checker"
    controller_plugins: ["FollowPath"]
    odom_topic: /ad/odometry #TODO: 
    use_realtime_priority: false
    # Progress checker parameters
    progress_checker:
      plugin: "nav2_controller::SimpleProgressChecker"
      required_movement_radius: 0.5
      movement_time_allowance: 10.0
    # Goal checker parameters
    general_goal_checker:
      stateful: True
      plugin: "nav2_controller::SimpleGoalChecker"
[... 261 lines cut ...]
      action_type: "approach"
      footprint_topic: "/local_costmap/published_footprint"
      time_before_collision: 1.2
      simulation_time_step: 0.1
      min_points: 6
      visualize: False
      enabled: True
    observation_sources: ["pointcloud"]
    pointcloud:
      type: "pointcloud"
      topic: "/ad/lidar/velodyne_points"
      min_height: 0.2
      max_height: 2.0
      enabled: True           
```
The subsequent costmaps appear in rviz2 only when selecting appropriate frames but no Voxel layer is added [see](https://drive.google.com/file/d/1QRrhExKwFv5pjOBCdu1QVue_5dnxmlhW/view?usp=sharing) 

Now changing 
- DDS implementation:
  - [rmw_zenoh_cpp](https://github.com/ros2/rmw_zenoh/tree/jazzy) 

The subsequent costmaps appear in rviz2 only when again selecting appropriate frames and this time with Voxel layers as well. [See](https://drive.google.com/file/d/11Tl4QiqAxEAbiWGTe7vSTIb8ul4RFhWA/view?usp=drive_link)

#### Additional information
Using a custom localization node.

SteveMacenski, 2025-02-05:
What is your expected behavior, I don't understand. Are you saying that you're missing the costmap occupancy grids of the 3D visualization of the voxel grid? I don't see that you have the voxel grid topic in your rviz display panel, nor is it published by default since that publication and processing method is expensive (but can be enabled with the cloud voxel grid publisher node in costmap 2d). 

I'd appreciate if you updated your title and description to include more specific detail about what it is that you're looking for. 

For the occupancy grid, that I see in the rviz config: Did you set the QoS correctly for the occupancy grids from local/global costmap?  Can you reproduce this with Nav2's included rviz configuration? 

Are you sure your topics are correct and frames exist? Any error logs?

kulkarni-raunak, 2025-02-06:
> What is your expected behavior, I don't understand. Are you saying that you're missing the costmap occupancy grids of the 3D visualization of the voxel grid? I don't see that you have the voxel grid [...]
I have added `voxel_layer` as a plugin to `local_costmap` and `global_costmap` and with `voxel_layer` I have set `enabled: True` and `publish_voxel_map: true`.  

1. With these one should always see updates from `voxel_layer` to `local_costmap` and `global_costmap` right? 
2. I see `voxel_layer` add to `global_costmap` and `local_costmap` when I change `RMW_IMPLEMENTATION=rmw_zenoh_cpp`. Is this normal?

> I'd appreciate if you updated your title and description to include more specific detail about what it is that you're looking for.
Done.
> For the occupancy grid, that I see in the rviz config: Did you set the QoS correctly for the occupancy grids from local/global costmap? Can you reproduce this with Nav2's included rviz configuration?
Does one need to set QoS in the `nav2_params.yaml` file? 
> Are you sure your topics are correct and frames exist? Any error logs?
Yes, topics exist and I can echo them as well. It's problem getting them to visualize in rviz2 using default DDS.

SteveMacenski, 2025-02-06:
Please ask on Robotics Stack Exchange, these are not problems with the stack and are Q&A style questions best suited for a Q&A platform and/or googling :-)