ros-navigation/navigation2 #4905: Costmap namespace prevents remapping relative topics

Martin-Oehler, 2025-02-06:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 24.04
- ROS2 Version:
  - Jazzy
- Version or commit hash: ros-jazzy-navigation2	1.3.4-1noble.20250115.215155
- DDS implementation: not relevant

#### Steps to reproduce issue

The costmap nodes are loaded as part of the planner/controller in a namespace, e.g. `local_costmap/local_costmap`. To ensure portability when pushing a new namespace (e.g. `/robot_a/local_costmap/local_costmap` ), I have specified sensor source topics as relative topics, e.g. `topic: lidar_scan`.

In practice, this results in the subscription `/robot_a/local_costmap/lidar_scan` because the cost map is hardcoded to be started in 
 the namespace `local_costmap`. It is not possible to remap the topic to the correct `/robot_a/lidar_scan` in any way. The only thing that works is a relay which introduces an additional copy.

With the current setup, it is not possible to create a portable configuration that works when pushed to a new namespace.

#### Expected behavior

It should be possible to create a nav2 configuration that can be pushed to a new namespace.

#### Actual behavior

It is not possible to create such a configuration without relay nodes that introduce additional copies.

#### Example

Launch file:

```yaml
- arg:
    name: namespace
    default: "robot_a"
- push_ros_namespace:
    namespace: "$(var namespace)"

- node:
    pkg: "nav2_controller"
    exec: "controller_server"
    name: "controller_server"
    output: "screen"
    respawn: "true"
    param:
        - from: $(find-pkg-share nav_launch)/config/nav2_config.yaml
```

Config:

```yaml
/**:
    local_costmap:
        local_costmap:
            ros__parameters:
                update_frequency: 1.0
                publish_frequency: 1.0
                global_frame: map
                robot_base_frame: base_link
                robot_radius: 0.22 
                resolution: 0.05
                height: 5
                width: 5
                track_unknown_space: true
                rolling_window: true
                plugins: ["voxel_layer", "inflation_layer"]
                voxel_layer:
                    plugin: "nav2_costmap_2d::VoxelLayer"
                    enabled: True
                    footprint_clearing_enabled: true
                    publish_voxel_map: True
                    origin_z: 0.0
                    z_resolution: 0.05
                    z_voxels: 16
                    unknown_threshold: 15
                    mark_threshold: 0
                    observation_sources: "lidar"
                    combination_method: 0
                    lidar: 
                        topic: lidar_scan
                        max_obstacle_height: 2.0
                        min_obstacle_height: 0.5
                        obstacle_max_range: 2.5
                        obstacle_min_range: 0.0
                        raytrace_max_range: 3.0
                        raytrace_min_range: 0.0
                        clearing: True
                        marking: True
```

SteveMacenski, 2025-02-06:
Just set the topic to `/lidar_scan`. This is documented in the configuration guide and even in the default parameters yaml file:

https://github.com/ros-navigation/navigation2/blob/main/nav2_bringup/params/nav2_params.yaml#L213-L220

Martin-Oehler, 2025-02-07:
Thank you for your pointer. I did check the default parameters, but the current jazzy release does not have this information. In fact, it seems like my exact issue has been fixed last December in this commit https://github.com/ros-navigation/navigation2/commit/3ee384df622ed964680452dcd53da6d0e9a15514. I was unable to verify this because I do not have a rolling installation and a lot of dependencies had updates. I guess, I will have to wait for a new release then.

closed 2025-02-07 by Martin-Oehler

SteveMacenski, 2025-02-07:
Note that this is not that new, while this is newly added to the yaml file for clarity

```
          # A relative topic will be appended to the parent of the local_costmap namespace.
          # For example:
          #   * User chosen namespace is `tb4`.
          #   * User chosen topic is `scan`.
          #   * Topic will be remapped to `/tb4/scan` without `local_costmap`.
          #   * Use global topic `/scan` if you do not wish the node namespace to apply
```

That PR does change behavior, but a note like this that would have appeared before would read as follows with the same advice

```
          # A relative topic will be appended to the parent of the local_costmap namespace.
          # For example:
          #   * User chosen namespace is `tb4`.
          #   * User chosen topic is `scan`.
          #   * Topic will be remapped to `/tb4/local_costmap/scan`.
          #   * Use global topic `/scan` if you do not wish the node namespace to apply
```

Most people use the absolute topics and this is not a new behavior to Nav2 :-)

Martin-Oehler, 2025-02-10:
Yes, however in our multi-robot scenario I can not use absolute topics so launch files stay reusable for other robot instances. The referenced commit specifically addresses this problem with nav2:

Configured topic: `scan`
Behavior before commit: Node subscribes to `/tb4/local_costmap/scan`
Behavior before commit: Node subscribes to `/tb4/scan`

david-wb, 2025-03-07:
I just ran into this issue today. Any idea when the fix will become available in ros2 jazzy? I need `scan` to be remapped to `<robot_namespace>/scan`, not `<robot_namespace>/local_costmap/scan`.

Martin-Oehler, 2025-03-07:
Seemingly, the nav2 release missed the last jazzy sync by just a few days. The next one should happen in March, but there has not been an announcement.

david-wb, 2025-03-07:
Thanks. The release process is still a mystery to me. In the meantime, I worked around it with a topic relay:

```python
        scan_relay = Node(
            package="topic_tools",
            executable="relay",
            name="scan_relay",
            namespace=f"/{agent.uid}",
            arguments=[f"scan", f"local_costmap/scan"],
            output="screen",
        )
```

Martin-Oehler, 2025-03-11:
The new release is out, but the changes are not in there. For some reason, the commit is only on the main branch and not jazzy, so it wasn't part of the release.

SteveMacenski, 2025-03-11:
It is a breaking change, so it is not backported to currently released distributions. It will be released into Kilted and newer