ros-navigation/navigation2 #4906: Docker install navigation2 compared to building from source github jazzy branch

w-kok, 2025-02-06:
Title: ROS 2 Jazzy/Iron Docker Install - Codebase and Update Frequency

Required Info:

Operating System:
Ubuntu 24.04 (running inside Docker)
Computer:

ROS2 Version:
ROS 2 Iron / Jazzy (depending on the project, using Docker)
Version or commit hash:

DDS implementation:

Nav2 Package:
Nav2 Controller (e.g., MPPI, DWB, Smac)
Have you asked this on Robotics Stack Exchange?

Question:
I am using ROS 2 Iron and Jazzy within a Docker-based installation and would like to understand which codebase is used in these Docker images. Specifically:

Does the Docker installation use the jazzy or iron branch from the source repositories on github?
How far behind is the Docker-based installation compared to the latest commits in the official source branches?
What is the typical update frequency for these Docker images?
This information is crucial for ensuring I am working with the latest features and bug fixes. Any insights would be appreciated. Thanks!

SteveMacenski, 2025-02-06:
The docker images use the source from the branch of the distribution for our nightly builds [1]. When we cut releases (bump Humble from 1.1.17 to 1.1.18 for example), then we run it again once on that tag number [2] and that docker image is uploaded as well. The tagged versions are never rebuilt so that they are stable. They are automatically done day-of when distribution releases are made for Nav2 binaries. The nightlys are rebuilt nightly. 

[1] https://github.com/ros-navigation/nav2_docker/blob/main/Dockerfile#L17
[2] https://github.com/ros-navigation/nav2_docker/blob/main/Dockerfile#L19

w-kok, 2025-02-06:
Thanks for your effort and input!

To clarify, I'm working with my own Docker setup, not the Nav2 Docker. I want to understand what gets installed when setting up on a machine or within a Docker container.

For example, when installing on a machine or building within another Dockerfile using:
RUN sudo apt-get install ros-$(ROS_distro)-navigation2
or, more generally, on a machine with:
sudo apt-get install ros-jazzy-navigation2

What versions are installed in these cases?
What is the release structure for these packages?

Running dpkg -l | grep navigation2 on an Iron distribution, for example, shows 1.2.10-1jammy. Would this version correspond to a tag in the correct branch—in this case, Iron—where version 1.3.5 has already been released?
This applies to different ROS distributions as well. Since many valuable features and bug fixes have been introduced.

I'm also asking because I’ve encountered issues building Nav2 from source.

Thank you for effort, hopefully you can shine some light on the versions installed and the deployment of NAV2.

SteveMacenski, 2025-02-07:
If you're building / managing your own docker image, then I can't tell you what your release, update, or install method policy is 🙃 

>  RUN sudo apt-get install ros-$(ROS_distro)-navigation2

You can use `dpkg` or `apt` to find the package's information. These are not specific to docker and are the general binaries that are released when any ROS package runs a release. Nav2's tags can be found in the `tags` part of the UI or as a branch you can check out if you want a particular tag.

closed 2025-02-07 by w-kok