ros-navigation/navigation2 #4907: Redesign dynamic parameters patterns

SteveMacenski, 2025-02-06:
It is brought to my attention dynamic parameter handling matured in Humble that allows us to use the `add_post_set_parameters_callback` instead of `add_on_set_parameters_callback` for processing parameter updates into member functions. This was brought up from a conversation started by @pepisg and myself on MPPI dynamic parameter handling issues and explained in https://github.com/ros2/rclcpp/pull/2735 by Chris.

The tl;dr minimum of this ticket is to: Update globally all of the `add_on_set_parameters_callback` to use `add_post_set_parameters_callback` instead and update the dynamic parameter callback if the signature changes

This is a really nice first contribution that touches alot of code!

Secondarily, many nodes are not as well organized with larger sets of parameters like MPPI and RPP with parameter handler objects. Along with the migration, we should consider making that also a global pattern of having all objects have parameter handlers to manage the parameter's state and dynamic callbacks to simplify the application code and have consistency architecturally.

Also a really nice first-contribution that gets folks into the architecture a little!

Nils-ChristianIseke, 2025-02-08:
I will work on this.

Sahas-Ananth, 2025-02-12:
Can I also help with this?

Nils-ChristianIseke, 2025-03-13:
As suggested by @SteveMacenski in https://github.com/ros-navigation/navigation2/pull/4971#pullrequestreview-2676035908.

_Preliminary list of all nodes that need to be adjusted (searched repo for on_set_parameters_callback)
I will adapt this list_

1. - [ ] [nav2_mppi_controller/src/parameters_handler.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_mppi_controller/src/parameters_handler.cpp)
2. - [ ] [nav2_controller/plugins/pose_progress_checker.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_controller/plugins/pose_progress_checker.cpp)
3. - [ ] [nav2_costmap_2d/src/costmap_2d_ros.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_costmap_2d/src/costmap_2d_ros.cpp)
4. - [ ] [nav2_costmap_2d/plugins/plugin_container_layer.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_costmap_2d/plugins/plugin_container_layer.cpp)
5. - [ ] [nav2_docking/opennav_docking/src/docking_server.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_docking/opennav_docking/src/docking_server.cpp)
6. - [ ] [nav2_dwb_controller/dwb_plugins/src/kinematic_parameters.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_dwb_controller/dwb_plugins/src/kinematic_parameters.cpp)
7. - [ ] [nav2_controller/src/controller_server.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_controller/src/controller_server.cpp)
8. - [ ] [nav2_costmap_2d/plugins/static_layer.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_costmap_2d/plugins/static_layer.cpp)
9. - [ ] [nav2_controller/plugins/simple_progress_checker.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_controller/plugins/simple_progress_checker.cpp)
10. - [ ] [nav2_rotation_shim_controller/src/nav2_rotation_shim_controller.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_rotation_shim_controller/src/nav2_rotation_shim_controller.cpp)
11. - [ ] [nav2_smac_planner/src/smac_planner_hybrid.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_smac_planner/src/smac_planner_hybrid.cpp)
12. - [ ] [nav2_graceful_controller/src/parameter_handler.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_graceful_controller/src/parameter_handler.cpp)
13. - [ ] [nav2_controller/plugins/simple_goal_checker.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_controller/plugins/simple_goal_checker.cpp`)
14. 1. [ ] [nav2_costmap_2d/plugins/inflation_layer.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_costmap_2d/plugins/inflation_layer.cpp)
15. - [ ] [nav2_smac_planner/src/smac_planner_2d.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_smac_planner/src/smac_planner_2d.cpp)
16. - [ ] [nav2_mppi_controller/tools/parameters_handler.hpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_mppi_controller/include/nav2_mppi_controller/tools/parameters_handler.hpp)
17. - [ ] [nav2_docking/opennav_docking/src/controller.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_docking/opennav_docking/src/controller.cpp)
18. - [ ] [nav2_collision_monitor/src/polygon.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_collision_monitor/src/polygon.cpp)
19. - [ ] [nav2_amcl/src/amcl_node.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_amcl/src/amcl_node.cpp)
20. - [ ] [nav2_costmap_2d/plugins/obstacle_layer.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_costmap_2d/plugins/obstacle_layer.cpp)
21. - [ ] [nav2_regulated_pure_pursuit_controller/src/parameter_handler.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_regulated_pure_pursuit_controller/src/parameter_handler.cpp)
22. - [ ] [nav2_controller/plugins/stopped_goal_checker.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_controller/plugins/stopped_goal_checker.cpp)
23. - [ ] [nav2_costmap_2d/plugins/voxel_layer.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_costmap_2d/plugins/voxel_layer.cpp)
24. - [ ] [nav2_planner/src/planner_server.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_planner/src/planner_server.cpp)
25. - [ ] [nav2_velocity_smoother/src/velocity_smoother.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_velocity_smoother/src/velocity_smoother.cpp)
26. - [ ] [nav2_planner/test/test_dynamic_parameters.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_planner/test/test_dynamic_parameters.cpp)
[... cut ...]
30. - [ ] [nav2_collision_monitor/src/source.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_collision_monitor/src/source.cpp)
31. - [ ] [nav2_smac_planner/src/smac_planner_lattice.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_smac_planner/src/smac_planner_lattice.cpp)
32. - [ ] [nav2_navfn_planner/src/navfn_planner.cpp](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_navfn_planner/src/navfn_planner.cpp)

SteveMacenski, 2025-03-13:
It might be nice to open PRs in 3-4 node segments so that the reviews can go faster and not block progress on all nodes for 1 node's small tweaks!

Nils-ChristianIseke, 2025-03-24:
@SteveMacenski , what are your thoughts on generate_parameter_library?  I’d like to use this, but I’m not sure if your concern (_what_ if we want more custom parts of our parameter handler, like wanting to mutex lock parameter access for multi-threaded situations where the dynamic parameter executor is different from the running executor)_ 
prevents us from doing so.

SteveMacenski, 2025-03-27:
I think for now, we should update the stack with our gameplan and not use the generate parameter library. After doing a look over a few of our uses of dynamic parameters, there are times that updates to specific parameters (or classes of parameters) also need to trigger actions like reinitalizations of objects. 

These also need to occur under the same mutex lock as changing the parameter so that other operations don't happen between when the parameter is updated and a potential post-change event callback is triggered that we could use to do so. Plus, I'm not sure if that library allows us to override dynamic parameter callbacks to check on values to do that process. 

I think that framework works for some nodes but not all, so I'd rather keep all of Nav2 consistent in design than cherry pick some percentage to do in another way.

Nils-ChristianIseke, 2025-03-27:
Great! Then I will continue to do that :) Thanks for writing down your reasoning!

SteveMacenski, 2025-03-27:
Sorry for the delay, my github notifications tray has too many thought-inducing topics that I'm not quite able to get them all done every day :-) I appreciate the patience

Nils-ChristianIseke, 2025-04-20:
@Sahas-Ananth After the first PR was merged (and can be used as template for future modifications) I would appreciate your help :).  You can have a look [here](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_regulated_pure_pursuit_controller/src/parameter_handler.cpp) to see how it should look at the end :). 
I recently started to work on nav2_controller.

SteveMacenski, 2025-04-24:
Something brought up in https://github.com/ros-navigation/navigation2/issues/5100 would be to change all the plugins to skip looking at updates for parameters outside of their namespace. That way we only validity check parameters having directly to do with this algorithm rather than checking against all parameters in that node (of potentially many algorithms + the core server).

mini-1235, 2025-10-16:
For AMCL, it re-initialize many member variables when updating dynamic parameters https://github.com/ros-navigation/navigation2/blob/01580bda7a3d38fc1215547959580165e8d7175e/nav2_amcl/src/amcl_node.cpp#L1189-L1227

Does it make sense to just update the node to use `add_post_set_parameters_callback` without creating a new ParameterHandler object?

mini-1235, 2026-02-11:
The table is up to date. #5920 is mainly to get the stalled PR off the list. It would be great if someone is interested in continuing the work on the remaining packages, but otherwise I can take over since it has been sitting for > 1 year