ros-navigation/navigation2 #4910: Dagger

Nils-ChristianIseke, 2025-02-08:
## Feature request
Integrate [Dagger](https://github.com/dagger/dagger) into the project to create a fast, and portable CI pipeline.
#### Feature description
See the [benefits](https://github.com/dagger/dagger#benefits-to-app-teams).

If it is considered useful, i can take care of the feature

SteveMacenski, 2025-02-10:
@ruffsl is the unquestionable leader of our CI efforts, so I leave any tooling decisions to him -- but if he agrees I'm sure he'd like the help!

SteveMacenski, 2025-03-14:
I think from the general lack of interest from Ruffin and myself, I'd prefer to not add in more tooling - we have something that works that we can manage without dedicating alot of regular time to upkeep and I'd rather not rock the boat unless there's something critically missing and/or broken. 

@Nils-ChristianIseke can you tell me a bit about why you'd like this for Nav2 CI?

Nils-ChristianIseke, 2025-03-15:
Fine for me I am Closing this issue.

closed 2025-03-15 by Nils-ChristianIseke