ros-navigation/navigation2 #4916: Introduciont ament_mypy

Nils-ChristianIseke, 2025-02-11:
## Feature request
Introduce ament_mypy to the codebase, to enforce and check type hinting on python code.

#### Feature description
Increasing the Python code quality by introducing a static type checker.

#### Implementation considerations
See [how to use mypy with an existing codebase](https://mypy.readthedocs.io/en/stable/existing_code.html).
To generate type hints automatically one could use tools like [pyannotate](https://github.com/pyannote/pyannote-audio) or [monkeytype](https://github.com/Instagram/MonkeyType).

SteveMacenski, 2025-02-11:
Should this be in nav2 or in the ament linting: https://github.com/ament/ament_lint

PS it looks like mypy is already there 🙃

SteveMacenski, 2025-03-25:
Smac Planner state lattice generator is completed!

leander-dsouza, 2025-03-26:
Hey Steve, I have tried integrating this over at #5022.

SteveMacenski, 2025-03-26:
Lovely, thanks!

SteveMacenski, 2025-03-28:
Reopening, I think that was just for smac planner, yeah?

leander-dsouza, 2025-03-28:
> Reopening, I think that was just for smac planner, yeah?

Yes, all the python packages need to be added separately.

Nils-ChristianIseke, 2025-03-30:
After adding type hints  to a package we should run:  
`ruff check --ignore Q000 --select ALL <package_name>` to ensure type hinting is according to the latest python standard.
We can ignore all non-type related issues.

leander-dsouza, 2025-03-30:
> After adding type hints to a package we should run: `ruff check --select UP <package_name>`

We can add the ruff linter to the pre-commit once #5029 is merged.

SteveMacenski, 2025-04-04:
I don't think (tentatively and ignorantly) that we should be ignoring any nav2 packages in the mypy system, else it seems to defeat some of the purpose of enabling it in the first place. 

```

[[tool.mypy.overrides]]
module = [
    "matplotlib.*",
    "rtree.*",
    "launch.*",
    "ament_index_python.*",
    "nav2_common.*",
    "nav2_msgs.*",
    "nav2_simple_commander.*",
    "launch_testing.*",
    "action_msgs.*",
    "geometry_msgs.*",
    "sensor_msgs.*",
    "tf2_ros.*",
    "nav_msgs.*",
    "rosgraph_msgs.*",
    "tf_transformations.*",
    "ament_pep257.*",
    "ament_flake8.*",
    "ament_copyright.*",
]
```

Would `mypy_path = "."` work?

leander-dsouza, 2025-04-04:
> Would `mypy_path = "."` work or need to add a few more `__init__.py`'s? What all checks is this skipping?

I have tried integrating `mypy_path` locally, but it does not seem to affect the relative path. I think I have to investigate this further in the near future.

The check that it is skipping is whether the `mypy_packages` can import the following packages safely.

leander-dsouza, 2025-04-11:
> [@leander-dsouza](https://github.com/leander-dsouza) mind giving me a hand with [#5056](https://github.com/ros-navigation/navigation2/pull/5056)? 

Yes, sure, Steve, I am happy to help. I will make a PR to #5056 soon.

Sorry for the delayed response; I was travelling cross country at the time.

closed 2025-04-14 by SteveMacenski

tonynajjar, 2025-12-24:
Howdy, I cloned nav2 in a fresh ros:rolling + apt upgrade and ran pre-commit. `ament_mypy` seems to be failing (error log is much longer but I'll spare you the details); is this expected? How come it's not caught in CI?

I can open a separate issue if it's not expected and actually a bug.

```
root@ae125719ddf5:~/navigation2# pre-commit run --all
[INFO] Installing environment for https://github.com/python-jsonschema/check-jsonschema.
[INFO] Once installed this environment will be reused.
[INFO] This may take a few minutes...
check for added large files..............................................Passed
check python ast.........................................................Passed
check for case conflicts.................................................Passed
check for merge conflicts................................................Passed
check for broken symlinks............................(no files to check)Skipped
check xml................................................................Passed
check yaml...............................................................Passed
debug statements (python)................................................Passed
fix end of files.........................................................Passed
forbid submodules....................................(no files to check)Skipped
mixed line ending........................................................Passed
trim trailing whitespace.................................................Passed
fix utf-8 byte order marker..............................................Passed
isort (python)...........................................................Passed
codespell................................................................Passed
Validate GitHub Workflows................................................Passed
Validate GitHub Actions..............................(no files to check)Skipped
Validate Dependabot Config (v2)..........................................Passed
ament_lint_cmake.........................................................Passed
ament_cpplint............................................................Passed
ament_uncrustify.........................................................Passed
ament_xmllint............................................................Passed
ament_flake8.............................................................Passed
ament_pep257.............................................................Passed
ament_mypy...............................................................Failed
- hook id: ament_mypy
- exit code: 1

mypy error encountered
Warning: unused section(s) in tools/pyproject.toml: module = ['ament_copyright.*', 'ament_flake8.*', 'ament_pep257.*', 'geographic_msgs.*', 'geometry_msgs.*', 'geopandas.*', 'matplotlib.*', 'nav2_loopback_sim.*', 'nav2_msgs.*', 'nav_msgs.*', 'pandas.*', 'sensor_msgs.*', 'std_msgs.*', 'std_srvs.*', 'tf2_ros.*', 'tf_transformations.*', 'zmq.*']

Regular report continues:
Success: no issues found in 9 source files

mypy error encountered
Warning: unused section(s) in tools/pyproject.toml: module = ['ament_copyright.*', 'ament_flake8.*', 'ament_pep257.*', 'geopandas.*', 'matplotlib.*', 'nav2_loopback_sim.*', 'nav_msgs.*', 'pandas.*', 'rtree.*', 'sensor_msgs.*', 'std_srvs.*', 'tf2_ros.*', 'tf_transformations.*', 'zmq.*']
```