ros-navigation/navigation2 #4922: create a pause between planner server and controller

andriana2, 2025-02-16:
Hi everyone I am doing my final proyect using _rover robotics mini_ and I am traing to obtein the path and way to have a button 

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Computer:
  - NUK
- ROS2 Version:
  - Humbel
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Nav2 Package:
  - 
- Have you asked this on Robotics Stack Exchange?
  - 

#### Tuning / Configuration Goal

#### What I've Tried

BCKSELFDRIVEWORLD, 2025-02-17:
I'm not entirely sure what you're asking. If you want the planner to visualize the trajectory, you can echo the topic "planner_server/plan" in RViz. Additionally, if the robot pauses—which indicates that the goal has been aborted—you can add an mctrl so that when goal_aborted is true, it triggers a goal_reset.

mamariomiamo, 2025-02-17:
Is it possible to add a Wait BT action node in the BT before the controller server?
https://docs.nav2.org/configuration/packages/bt-plugins/actions/Wait.html

SteveMacenski, 2025-03-05:
Please ask on robotics stack exchange, but it sounds to me like you're trying to build the autonomy / application visualization layer on top of Nav2 -- which many have done before. This is going to require that you develop that application yourself and isn't something that is given to you by the Nav2 stack. 

it sounds like you just want to call the stack and then not pass on the `cmd_vel` to the base, which should be easy enough by remapping the topics. When you want the robot to move, then map it back (or just have a muxing gate in your application that decides when to actually pass on commands to execute).

For the pause, simply store your current goal and resubmit it once you want to resume. If you want to follow the exact same path and not recompute it, then you would have to store that in the Behavior Tree and implement your pause/resume behavior within your Nav2 BT XML (which should be easy enough with some blackboard variables and a few simple BT nodes). 

Hopefully that should be plenty to get you started :-)

closed 2025-03-05 by SteveMacenski