ros-navigation/navigation2 #4937: Regulated Pure Pursuit:  adaptive roateToGoalHeading speed, use integrated distance for lookahdead point, and prevent oscillation between rotateToGoalHeading and RotateToPathHeading

lyfkyle, 2025-02-21:
## Feature request
When using RPP for my project, I noticed several limitations and have to make the following changes. Raising this feature request to discuss whether they should be added to the main branch

### Adaptive RotateToGoalHeading speed
Currently RPP uses a fixed rotate_to_goal_heading speed, defined by parameter _rotate_to_heading_angular_vel_. As I want my robot to rotate fast, this causes overshoot. A simple P controller could alleviate this issue. 
`  angular_vel = rotate_to_heading_angular_vel_ * angle_to_path;   // P controller`

### Use integrated distance for lookahead point
Currently RPP uses euclidean distance to decide the lookahead point (https://github.com/ros-navigation/navigation2/blob/main/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp#L381). However, our path has a lot of sharp near in-place rotations and we have to set lookahead distance to extremely small value for it to work. I feel that using _first_after_integrated_distance_ is more proper in this case because by definition, we are looking ahead **on** the path. In extreme case, if the planned path tells the robot to keep rotating in place, RPP should command robot to rotate in place. In this case, even _first_after_integrated_distance_ is not enough as it internally uses euclidean distance, we need another distance metrics for SE2 pose difference

### Robot oscillates between rotateToGoalHeading and rotateToPathHeading
We set _use_rotate_to_heading_ to True and observe robot oscillates between rotating in opposite directions near goal. It turns out that RPP is oscillating between between rotating to goal heading and rotating to path headin due to our localisation errors. _shouldRotateToGoalHeading_ is purely decided by the distance to goal, compared against _goal_dist_tol__ (https://github.com/ros-navigation/navigation2/blob/main/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp#L320). And since our localisation is oscillating, the robot rotates to goal heading at one instant as it is near enough to goal and decide to rotate to path heading at next instant as it is now far from goal. In the end, I added logic that once the RPP starts rotating to actual global goal, it should not enter other modes. I am very unsure about this change. At the same time, localisation oscillation is expected.  

### Others
I find the naming _shouldRotateToGoalHeading_ very confusing as it actually applies to all carrot pose as well. To make it worse, the threshold is read from goal checker config which by design, only applies to global path goal. Should we move functionalities around such that _shouldRotateToGoalHeading_ only applies to global path goal and rotating to carrot pose is handled by _shouldRotateToPath_?

Love to get some feedback and suggestions. Happy to raise a PR if some changes are desired.

SteveMacenski, 2025-03-11:
> Adaptive RotateToGoalHeading speed

I agree that could be improved. I think though we should do something like in https://github.com/ros-navigation/navigation2/blob/ee96246985a9a85313da00e4488213ba352bdb58/nav2_rotation_shim_controller/src/nav2_rotation_shim_controller.cpp#L298-L326 where use the acceleration constraints and distance to the heading to accelerate and decelerate the angular velocity. That's the most smooth solution!

> Use integrated distance for lookahead point

I had some discussions with @tonynajjar before I went on vacation about this subject and I tentatively agree. Tony, what was your results from those tests?

tonynajjar, 2025-03-15:
> I had some discussions with @tonynajjar before I went on vacation about this subject and I tentatively agree. Tony, what was your results from those tests?

I unfortunately did not end up going all the way with it. My goal was to avoid shortcutting the path in sharp turns and using the integrated distance was supposed to help with that but still, with a moderately big lookahead distance, some shortcutting is still bound to happen. Another potential solution I was considering was the detection of high-curvature points along the path and limit the lookahead distance to these points (similar to the cusps feature) but I couldn't find a reliable and not too computational-intensive way to do it

lyfkyle, 2025-03-26:
Sorry for delay in response. Thanks a lot for the dicussion. 

> I agree that could be improved. I think though we should do something where use the acceleration constraints and distance to the heading to accelerate and decelerate the angular velocity. That's the most smooth solution!

This ensures that robot always stay in the velocity that allows the robot to stop when it reaches the goal heading. I will test more. 

> Use integrated distance for lookahead point

I agree that shortcutting is unavoidable. RPP by design shortcuts between current robot position and carrot pose. However, I think using integrated distance makes more sense as it fits the definition of carrot pose more. Furthermore, it allows for different distance measure (SE2 distance for example) so that RPP could even follow a path that contains in-place rotations. 

>  But Adding a parameter stateful to do this would be allowable we have a similar setting in DWB. You can submit a PR for that and that should be an easy review.

Thanks for the suggestion. Will do.

> Unless the goal tolerance >= lookahead distance or minimum lookahead distance, this should be fine. When we're not approaching the goal, the condition dist_to_goal < goal_dist_tol_ is never true. 

Yes. Unfortunately in my project this happens because I am trying to follow a very sharp turn, hence setting lookahead distance to be small. Taking a step back, I think RPP needs to do two things.

1. Rotate to path heading. This is currently riggered when the heading difference is too large or carrot pose is too near to the current robot position. I am seriously considering whether the second condition is actually needed. If the heading difference is not large, maybe it is fine for robot to follow an arc even if carrot pose is near. Maybe we could get rid of the second condition entirely. 

2. Rotate to global goal heading after reaching global goal. This is achieved by checking `dist_to_goal < goal_dist_tol_ ` and it makes sense here to get goal_dist_tol_ from goal checker. After this condition is satisfied, simply retrieves the final path heading and rotate.

If we do this, then 1 is purely handled by `shouldRotateToPathHeading` and 2 is purely handled by `shouldRotateToGoalHeading`, which I find to be quite neat. 

Let me know what you think. I will work on the first 3 points first.

SteveMacenski, 2025-04-11:
@lyfkyle what are your results from the testing of those changes? 😄 

For (1): I don't see that second condition in the code. Can you point it out? I think that feeds into your later question and "If we do this, then 1 is purely handled by shouldRotateToPathHeading and 2 is purely handled by shouldRotateToGoalHeading, which I find to be quite neat." which my quick glance makes me think is already the case.