ros-navigation/navigation2 #4946: Set BT action nodes' error_code_id in case request timeout in BtActionNode

DylanDeCoeyer-Quimesis, 2025-02-24:
## Feature request

#### Feature description
Currently, an action request timeout occurring in the BtActionNode parent class triggers the failure of the node without setting any code ID in the `error_code_id` output port. I think that an error code should be propagated.

#### Implementation considerations
I see at least two options:
- make BtActionNode aware of the `error_code_id` output port, and set a generic error code ID directly from BtActionNode,
- catch the failure at the child node level, and set the `error_code_id` directly from the child node. Typically, with `CustomNode` inheriting from `BtActionNode`:
```
BT::NodeStatus CustomNode::tick()
{
    BT::NodeStatus status = BtActionNode::tick();

    if (status == BT::NodeStatus::FAILURE)
    {
        setOutput("error_code_id", ERROR_CODE_REQUEST_TIMEOUT);
    }

    return status;
}
```
This latter implementation assumes that any failure triggered by `BtActionNode`'s `tick()` function is due to a request timeout.

SteveMacenski, 2025-03-10:
Are you referring to when the goal request is no ACKed by the server ([here](https://github.com/ros-navigation/navigation2/blob/e3c41beb08cd1939d8eed378768d2744bf47904c/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L257) and [here](https://github.com/ros-navigation/navigation2/blob/e3c41beb08cd1939d8eed378768d2744bf47904c/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L227))? 

I think it would make sense to setup an `on_timeout()`  callback in the child nodes for them to implement what they want to do if the action server times out. In that, set the error code ID to a timeout code (if one is not already present, add one). 

Would you be open to contributing this?

DylanDeCoeyer-Quimesis, 2025-03-11:
Hi @SteveMacenski , yes that's what I'm referring to.

Good idea. I will try to provide a PR in the following days/weeks.

padhupradheep, 2025-05-08:
@DylanDeCoeyer-Quimesis any updates?

SteveMacenski, 2025-05-20:
Addressed in https://github.com/ros-navigation/navigation2/pull/5184

closed 2025-05-20 by SteveMacenski