ros-navigation/navigation2 #4951: Change nav2 params on the fly

GeraltRivskiy, 2025-02-26:
Hello! I'm new to nav2, I need to perform some use cases to my robot, for example, keep constant orientation while moving (omni platform). I see it could be done by changing min-max speed in the parameters, but I cant find the way to change it after I launched nav2. Is there any way to do so or can anyone suggest another way to do my task?

mini-1235, 2025-02-26:
I think you are looking for ros2 run rqt_reconfigure rqt_reconfigure

SteveMacenski, 2025-03-01:
You may use the dynamic parameters to change values at runtime. Rqt reconfigure is one such simple GUI tool to do so, or by using the parameter clients in ROS 2 to do so programmatically from other applications. 

I’d suggest looking at the dynamic parameters documentation on docs.ros.org  to learn more!

closed 2025-03-01 by SteveMacenski