ros-navigation/navigation2 #4952: AMCL crashes while setting dynamic parameters

GoesM, 2025-02-28:
## Bug report

**Required Info:**

- Operating System:
  -  ubuntu2404
- Computer:
  -  NUC
- ROS2 Version:
  -  foxy
- Version or commit hash:
  -  the latest
- DDS implementation:
  -  Fast-RTPS

#### Steps to reproduce issue

AMCL crashed when I change its dynamic parameters like following:
```
ros2 param set /amcl max_particles 123456
```

#### Expected behavior
No crash happened.

#### Actual behavior
AMCL crashed and with AddressSanitizer, we can get the bug report as following: 
```
=================================================================
==188828==ERROR: AddressSanitizer: out of memory: allocator is trying to allocate 0x80000f800 bytes
    #0 0x619f2cd1050d in calloc (/home/ROS-ws/install/nav2_amcl/lib/nav2_amcl/amcl+0xcc50d) (BuildId: c5cc9a39ce92f2efd962c3469235d816ea420a66)
    #1 0x7d8c72ef0b7b in pf_alloc (/home/ROS-ws/install/nav2_amcl/lib/libpf_lib.so+0x4b7b) (BuildId: 5843a8b760696056c62d50e87c8c83e5ba392b52)
    #2 0x7d8c747fd8eb in nav2_amcl::AmclNode::dynamicParametersCallback(std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter>>) (/home/ROS-ws/install/nav2_amcl/lib/libamcl_core.so+0x1fd8eb) (BuildId: 44bd6d35d74e7627fbbd10ab2d376c6af1bb8b69)
    #3 0x7d8c748bdeb3 in rcl_interfaces::msg::SetParametersResult_<std::allocator<void>> std::_Bind<rcl_interfaces::msg::SetParametersResult_<std::allocator<void>> (nav2_amcl::AmclNode::* (nav2_amcl::AmclNode*, std::_Placeholder<1>))(std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter>>)>::__call<rcl_interfaces::msg::SetParametersResult_<std::allocator<void>>, std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter>> const&, 0ul, 1ul>(std::tuple<std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter>> const&>&&, std::_Index_tuple<0ul, 1ul>) (/home/ROS-ws/install/nav2_amcl/lib/libamcl_core.so+0x2bdeb3) (BuildId: 44bd6d35d74e7627fbbd10ab2d376c6af1bb8b69)
    #4 0x7d8c748bdbb3 in std::_Function_handler<rcl_interfaces::msg::SetParametersResult_<std::allocator<void>> (std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter>> const&), std::_Bind<rcl_interfaces::msg::SetParametersResult_<std::allocator<void>> (nav2_amcl::AmclNode::* (nav2_amcl::AmclNode*, std::_Placeholder<1>))(std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter>>)>>::_M_invoke(std::_Any_data const&, std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter>> const&) (/home/ROS-ws/install/nav2_amcl/lib/libamcl_core.so+0x2bdbb3) (BuildId: 44bd6d35d74e7627fbbd10ab2d376c6af1bb8b69)
    #5 0x7d8c73f48473  (/opt/ros/jazzy/lib/librclcpp.so+0x148473) (BuildId: a6e4def86fc9f569833617ff077616d6d3344142)
    #6 0x7d8c73f4a7ee in rclcpp::node_interfaces::NodeParameters::set_parameters_atomically(std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter>> const&) (/opt/ros/jazzy/lib/librclcpp.so+0x14a7ee) (BuildId: a6e4def86fc9f569833617ff077616d6d3344142)
    #7 0x7d8c73fdb918  (/opt/ros/jazzy/lib/librclcpp.so+0x1db918) (BuildId: a6e4def86fc9f569833617ff077616d6d3344142)
    #8 0x7d8c73f8b937  (/opt/ros/jazzy/lib/librclcpp.so+0x18b937) (BuildId: a6e4def86fc9f569833617ff077616d6d3344142)
    #9 0x7d8c73f993f4  (/opt/ros/jazzy/lib/librclcpp.so+0x1993f4) (BuildId: a6e4def86fc9f569833617ff077616d6d3344142)
    #10 0x7d8c73fdafc9  (/opt/ros/jazzy/lib/librclcpp.so+0x1dafc9) (BuildId: a6e4def86fc9f569833617ff077616d6d3344142)
    #11 0x7d8c73f03a53 in rclcpp::Executor::execute_service(std::shared_ptr<rclcpp::ServiceBase>) (/opt/ros/jazzy/lib/librclcpp.so+0x103a53) (BuildId: a6e4def86fc9f569833617ff077616d6d3344142)
    #12 0x7d8c73f07c69 in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&) (/opt/ros/jazzy/lib/librclcpp.so+0x107c69) (BuildId: a6e4def86fc9f569833617ff077616d6d3344142)
    #13 0x7d8c73f18253 in rclcpp::executors::SingleThreadedExecutor::spin() (/opt/ros/jazzy/lib/librclcpp.so+0x118253) (BuildId: a6e4def86fc9f569833617ff077616d6d3344142)
    #14 0x7d8c73f105c8 in rclcpp::spin(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>) (/opt/ros/jazzy/lib/librclcpp.so+0x1105c8) (BuildId: a6e4def86fc9f569833617ff077616d6d3344142)
    #15 0x619f2cd50eb8 in main (/home/ROS-ws/install/nav2_amcl/lib/nav2_amcl/amcl+0x10ceb8) (BuildId: c5cc9a39ce92f2efd962c3469235d816ea420a66)
    #16 0x7d8c7262a1c9 in __libc_start_call_main csu/../sysdeps/nptl/libc_start_call_main.h:58:16
    #17 0x7d8c7262a28a in __libc_start_main csu/../csu/libc-start.c:360:3
    #18 0x619f2cc754d4 in _start (/home/ROS-ws/install/nav2_amcl/lib/nav2_amcl/amcl+0x314d4) (BuildId: c5cc9a39ce92f2efd962c3469235d816ea420a66)
```

#### Additional information

This may seem like increasing the value of 'max_marticle' will increase AMCL's memory requirements, but how do users know what the reasonable range of 'max_marticle' should be? 

Perhaps we can have a formula/code to calculate this reasonable range and then remind the user? 

After all, crashing directly during the process of modifying dynamic parameters may not be an elegant approach.

SteveMacenski, 2025-03-04:
But I think in this particular case:

> ==188828==ERROR: AddressSanitizer: out of memory: allocator is trying to allocate 0x80000f800 bytes

Unless I'm mis-parsing this, it seems to be written as plain language that your computer (or process, or docker container) simply lacks the memory to make the allocation of this size. I don't think this is a bug with AMCL, but asking the program do so something unrealistic. I know you've uncovered a few issues like this before with fuzzing resulting in  https://github.com/ros-navigation/navigation2/issues/4005 - I think this is adjacent to that? 

> Perhaps we can have a formula/code to calculate this reasonable range and then remind the user?

I don't think / know if that's possible. Do you have thoughts on what a formula like that would look like? I'm not sure AMCL knows enough about an environment, robot, or application to make an accurate judgement call. 

At the end of the day, folks are reasonable for tuning and configuring their own platforms using this framework. I don't know that I could create a formula for putting bounds on what is 'reasonable' for all possible applications of AMCL. I can do some to protect users from making bad choices, but not all things can I know what is a good range for all possible robot, sensor, and environment types (also considering that the industry is changing rapidly in the types of robots and environments). 

As to what is a reasonable range, there are a variety of papers analyzing particle filter localizers, their performance with different particle cloud sizes, and techniques. I think that documentation exists in literature and I believe Probablistic Robotics sets out a range of back of the envelope values as well.

GoesM, 2025-03-07:
Thank you for your patient reply!

> I know you've uncovered a few issues like this before with fuzzing resulting in #4005 - I think this is adjacent to that?

Yes, while submitting this issue, I also wanted to explore this type of problem more deeply. I don't want my work to be limited to just testing, as solving the problems discovered through testing is what gives the tests real-world significance. Therefore, my intention is to discuss and explore solutions with everyone, hoping to find a good approach or method to prevent such crashes from occurring.

I was thinking that, from a user's perspective, it's possible that they might want to push their hardware to its limits by trying boundary values for configuration parameters. So, I hope to explore some methods to calculate this range, thereby avoiding crashes or reducing attempts at extreme parameter values.

> I don't think / know if that's possible. Do you have thoughts on what a formula like that would look like? I'm not sure AMCL knows enough about an environment, robot, or application to make an accurate judgement call.

But as you mentioned, I also noticed that such range calculations would require a lot of environmental information. It seems that this approach isn't very feasible at the moment.

I might explore some other ideas to address this issue 🤔.

SteveMacenski, 2025-03-10:
OK! Let me know what you think and/or if there's something actionable to do that would help 👍

SteveMacenski, 2025-04-17:
@GoesM any updates?

padhupradheep, 2025-05-08:
@GoesM did you have any ideas that could be a possible solution for this issue?