ros-navigation/navigation2 #4957: Bringing back live Groot 2 monitoring

ajtudela, 2025-03-04:
## Feature request

I'm opening this issue to discuss the possibility of bringing back live monitoring of the behaviour tree using the new Groot 2.

I know there have been an endless number of tickets regarding live monitoring, especially since the introduction of multiple navigator types. But with the introduction of BT4 and Groot 2, things have changed, or so I think...

Last week, I did some tests and recorded some videos. 

- In the first video I opened two instances of Groot 2. One for NavToPose and one for NavThroughPoses, both connected to different ports. I send a goal, the robots move, then I place some goals and the robot moves through them. You can see in the video how both Groot2 instances capture the execution of the behaviour tree.

https://www.youtube.com/watch?v=juqoR-RyT6Q

- In the second video, I switch the xml of the BT.

https://youtu.be/xHFrtNq8P5w

With the PRO version of Groot 2, the content of the blackboard can be visualized, allowing the introspection of the BT nodes.

[image]

I have a branch [here](https://github.com/grupo-avispa/navigation2/tree/groot2) if anyone wants to try it.

SteveMacenski, 2025-03-11:
I'm not opposed. Is the monitoring not using ZeroMQ anymore causing issues with multiple clients in the same process?

```

  // Enable monitoring with Groot2
  if (enable_groot_monitoring_) {
    try {
      bt_->addGrootMonitoring(&tree_, groot_publisher_port_);
      RCLCPP_INFO(
        logger_, "Enabling Groot2 monitoring for %s: %d",
        action_name_.c_str(), groot_publisher_port_);
    } catch (const std::logic_error & e) {
      RCLCPP_ERROR(logger_, "Groot2 already enabled, Error: %s", e.what());
    }
  }
```

In this block above, does that error get logged? Can't we just re-enable it with the new parameters? I also don't see where this could throw, other than `RegisterJsonDefinition` where I'm not sure why that would throw if called multiple times. Please explain a bit. 

```
  // Register JSON definitions
  BT::RegisterJsonDefinition<builtin_interfaces::msg::Time>();
  BT::RegisterJsonDefinition<std_msgs::msg::Header>();
  BT::RegisterJsonDefinition<geometry_msgs::msg::Point>();
  BT::RegisterJsonDefinition<geometry_msgs::msg::Quaternion>();
  BT::RegisterJsonDefinition<geometry_msgs::msg::Pose>();
  BT::RegisterJsonDefinition<geometry_msgs::msg::PoseStamped>();
  BT::RegisterJsonDefinition<std::chrono::milliseconds>();
```

is this needed for all types in the BT? That doesn't seem very scalable given there's alot of ports in the BT nodes that are not just those.

---

Overall, it seems good to me in concept, but details to be worked out. I'd also want @facontidavide to review a PR enabling this since its his product and I assume he would want an integration to be as good and fully featured as possible for his customers

ajtudela, 2025-03-11:
> I'm not opposed. Is the monitoring not using ZeroMQ anymore causing issues with multiple clients in the same process?

I tested with two instances of Groot2: one for NavToPose (port 1667) and the other one for NavThroughPoses (port 1669) and the whole Nav2 running in a single container.

> is this needed for all types in the BT? That doesn't seem very scalable given there's alot of ports in the BT nodes that are not just those.

For custom types only. Standard types: int, std:string, etc are already converted to JSON. I have created conversions for all the custom types we use in the Nav2 BTs. Although I am having problems converting std::vector<geometry_msgs::PoseStamped>. I don't know if @facontidavide could help with this.

SteveMacenski, 2025-03-11:
What about `nav2_msgs`, `std_msgs`, or other `geometry_msgs` used? Is the list in your PR a complete list of all the input/output ports in `nav2_behavior_tree`? The set of BT string conversion utils that already exist for some ports might be useful to reference for other gaps. 

Thanks Davide for taking a look. 

> Let me look into it, maybe I can avoid the throwing.

Looking at the Nav2 software, I don't see why creating a new unique ptr to the groot monitor would cause an issue, that's just a new allocation. So, unless the registration calls throw, then I'm not sure why that block is necessary or the log "Groot2 already enabled". I assumed this was an issue with the branch, not BT.CPP/Groot, but let me know :-) 

On the registrations, is there a way to have the BT nodes themselves do this part? This isn't scalable if we have to statically register every type any BT node might use. Folks' custom BT nodes may well use custom types that we can't know about. The most scalable way I would think about doing that is to have the nodes themselves register the types they use and in the addGrootMonitoring method we call all the BT nodes in the tree's new `registerJsonPortDefinitions()`. We can provide a set of common used ones like in  `nav2_msgs`, `nav_msgs`, `std_msgs`, or other `geometry_msgs` built in, but gives the scalability for customization that this currently does not enable.

I think that's the only blocking "uh oh" that would prevent this from being mergable (+/- minor details to work out). Users need to be able to work with custom nodes with custom types, so we need a solution that enables that

facontidavide, 2025-03-11:
> I am having problems converting std::vector<geometry_msgs::PoseStamped>. I don't know if @facontidavide could help with this.

Weird, it should work automatically, in theory. I can help with that.

@ajtudela If you can submit a PR with a failing unit test to BehaviorTree.CPP, I will spend some time fixing it

facontidavide, 2025-03-11:
> Add that into the BT node tutorial in the docs with an example, and I think we're golden.

Pointing to this may help?

https://www.behaviortree.dev/docs/tutorial-basics/tutorial_11_groot2#visualize-custom-types-in-the-blackboard

SteveMacenski, 2025-03-11:
Absolutely, happy to do that instead while also pointing to some of the Nav2 implementations for more examples. Personally, I learn best by reviewing examples

ajtudela, 2025-04-04:
> What about `nav2_msgs`, `std_msgs`, or other `geometry_msgs` used? Is the list in your PR a complete list of all the input/output ports in `nav2_behavior_tree`? The set of BT string conversion utils that already exist for some ports might be useful to reference for other gaps.

Yes, all input/output ports used in `nav2_behavior_tree` are handled through conversions in the [`json_util`](https://github.com/grupo-avispa/navigation2/blob/groot2/nav2_behavior_tree/include/nav2_behavior_tree/json_utils.hpp).

> The most scalable way I would think about doing that is to have the nodes themselves register the types they use and in the addGrootMonitoring method we call all the BT nodes in the tree's new [...]

That’s exactly the approach I took:

The common types are defined directly in `BehaviorTreeEngine::addGrootMonitoring`:  
https://github.com/grupo-avispa/navigation2/blob/0d39c9b6f6e9022527d62361a8e754f417ff39eb/nav2_behavior_tree/src/behavior_tree_engine.cpp#L105-L115

And each BT node registers its own types individually: 
https://github.com/grupo-avispa/navigation2/blob/0d39c9b6f6e9022527d62361a8e754f417ff39eb/nav2_behavior_tree/include/nav2_behavior_tree/plugins/action/compute_path_to_pose_action.hpp#L76-L96

SteveMacenski, 2025-04-04:
Great! I'd be happy to review a PR to readd this then! 😄

closed 2025-04-10 by SteveMacenski