ros-navigation/navigation2 #4963: Feature Request: Add "plugins" Parameter to BT XML for Selective Clearing of Costmap Layers

BCKSELFDRIVEWORLD, 2025-03-05:
## Feature request

#### Feature description
Currently, the costmap clearing service clears all clearable layers. However, there are use cases where a user may want to clear only specific costmap layers (e.g., only the "keepout_filter" layer) instead of all available layers. This feature request proposes to add a "plugins" parameter in the BT XML configuration. This parameter would allow users to specify, via a comma-separated list, which plugin layers should be cleared.

#### Implementation considerations

`<ClearEntireCostmap
  name="ClearingActionsLocalCostmap"
  service_name="local_costmap/clear_entirely_local_costmap"
  plugins="keepout_filter"/>`

SteveMacenski, 2025-03-06:
Seems easy enough, the clearing service msg would need to be adjusted to contain a vector of strings (default empty; clear all) and then the service look at the pliugins' names to decide whether it should be cleared. 

I'd welcome a PR :-)

BCKSELFDRIVEWORLD, 2025-03-06:
Thanks for fast feedback.I will send pr ASAP.

SteveMacenski, 2025-03-11:
@BCKSELFDRIVEWORLD any update? 😄

BCKSELFDRIVEWORLD, 2025-03-11:
i tested on humble didn't see any issues. I plan to test it on Docker and open a PR before the week is over.

BCKSELFDRIVEWORLD, 2025-03-11:
Sorry for the delay, and please don't add any labels until the end of the week. :)

SteveMacenski, 2025-03-11:
Great, thanks!

SteveMacenski, 2025-05-01:
@BCKSELFDRIVEWORLD any update on that work? :-)

padhupradheep, 2025-05-08:
@BCKSELFDRIVEWORLD one more nudge? we would love to have this feature in!