ros-navigation/navigation2 #4967: Issue with keepout filter – Robot entering restricted areas

AnOrdinaryUsser, 2025-03-06:
**Required Info:**

- Operating System: V.M with Ubuntu 22.04
- ROS2 Version: Humble
- DDS implementation: FAST DDS
- Nav2 Package: DWB
- https://robotics.stackexchange.com/questions/114927/issue-with-keepout-filter-robot-entering-restricted-areas

#### Tuning / Configuration Goal

I am trying to configure the keepout filter and costmap parameters so that the robot consistently follows designated corridors without entering restricted areas. However, the robot sometimes gets too close to the inflation_radius of the global_costmap, causing it to stop moving when turning. I want to understand what might be misconfigured and how to properly tune these parameters.

#### What I've Tried

[image]

I am using the costmap filtering feature (as shown in the attached image). Based on my understanding:

The pink area represents the keepout zone, where the robot should not enter.
The blue area corresponds to the global_costmap radius.
The semi-purple area represents the local_costmap radius.
The red area is the intersection between both radii (global_costmap and local_costmap).
Additionally, the map is configured in trinary mode.
The planner seems to be working correctly because when I assign a plan, the robot generates a path that avoids these areas. However, as seen in the attached video, when the robot approaches the blue areas, it often gets too close. When turning, it sometimes slightly enters the inflation_radius of the global_costmap, causing it to stop moving. This issue mainly occurs when the robot needs to turn at curves.

📹 https://www.youtube.com/watch?v=Ml8G11srX_8

My question is: Shouldn’t the planner check for collisions based on the robot’s size and avoid these radii entirely?

### Parameter Configuration
Attached is the nav2_params.yaml file with my current configuration:
📄 https://www.transfernow.net/dl/202503060ZnIW6GE

I appreciate any insights or suggestions, as I am still learning ROS2 and may be making a beginner’s mistake.

BCKSELFDRIVEWORLD, 2025-03-06:
hi @AnOrdinaryUsser It sounds like the issue might be more about tuning. You might get some extra insights if you ask this on Robotics Stack Exchange.From what I gather, you’re working in a pretty tight space with a robot footprint that doesn’t allow much room for error. That means your controller really needs to be finely tuned 

i have 2 recommedition

First is try MPPI instead of DWA.MPPI gives you a lot more flexibility when it comes to tuning. For example, if you increase the cost values for MPPI’s Path Follow and Path Align Critic, you might end up with a much smoother and more stable path-following behavior.

Another is if you're a die-hard DWA fan If you really love DWA and want to stick with it, consider this: if your robot’s footprint occasionally grazes the keepout zone instead of coming to a full stop, adding a footprint_clearing parameter to your keepout settings could be just the tweak you need.

SteveMacenski, 2025-03-07:
Please do not cross post with stack exchange, I will follow up on your Q&A over there if the answer that's already on the question isn't already right

closed 2025-03-07 by SteveMacenski